libimobiledevice/idevicerestore #119: IOS 10.02 iphone 6 Libimobile update ( Still Error ) CSW

chacaaan, 2016-10-05:
C:\Users\muham\Desktop\iOS 10.0.2 Updated libimobiledevice Windows (Compiled by GeoSn0w)>idevicerestore -u b70fecbc9577d536e25196c1f6366e4304a9fe89 "C:\Users\muham\Desktop\edit done cfw shsh web\iPhone_4.7_10.0.2_14A456_Restore.ipsw"
NOTE: using cached version data
Found device in Recovery mode
Identified device as n61ap, iPhone7,2
Extracting BuildManifest from IPSW
Product Version: 10.0.2
Product Build: 14A456 Major: 14
INFO: device serial number is F73NG7P2G5N0
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without losing data.
Found ECID 5943093231869990
Getting ApNonce in recovery mode... 75 36 28 9e 0f 3d 3b 4e ba fe 14 20 f2 3a aa b4 2e e7 9b 27
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 48 73 0e 18 85 ff 06 41 6a a7 f2 18 0d 4a c6 84 1b 1f 76 ee
WARNING: Unable to find BbSkeyId node
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Using cached filesystem from 'C:\Users\muham\Desktop\edit done cfw shsh web\iPhone_4.7_10.0.2_14A456_Restore/058-41249-217.dmg'
Extracting iBEC.n61.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (632436 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-3406.1.79
iBoot build-style=RELEASE
Sending AppleLogo...
Extracting applelogo@2x~iphone.t7000.im4p...
Personalizing IMG4 component AppleLogo...
Sending AppleLogo (12116 bytes)...
ramdisk-size=0x10000000
Extracting 058-50460-076.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (41125363 bytes)...
Extracting DeviceTree.n61ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (122996 bytes)...
Extracting kernelcache.release.n61...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (12323612 bytes)...
About to restore device...
Waiting for device...
Device b70fecbc9577d536e25196c1f6366e4304a9fe89 is now connected in restore mode...
Connecting now...
Connected to com.apple.mobile.restored, version 14
Device b70fecbc9577d536e25196c1f6366e4304a9fe89 has successfully entered restore mode
Hardware Information:
BoardID: 6
ChipID: 28672
UniqueChipID: 5943093231869990
ProductionMode: true
Previous restore exit status: 0x100
Starting FDR listener thread
About to send NORData...
Found firmware path Firmware/all_flash/all_flash.n61ap.production
Getting firmware manifest Firmware/all_flash/all_flash.n61ap.production/manifest
Extracting LLB.n61.RELEASE.im4p...
Personalizing IMG4 component LLB...
Extracting iBoot.n61.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Extracting DeviceTree.n61ap.im4p...
Personalizing IMG4 component DeviceTree...
Extracting applelogo@2x~iphone.t7000.im4p...
Personalizing IMG4 component AppleLogo...
Extracting recoverymode@1334~iphone-lightning.t7000.im4p...
Personalizing IMG4 component RecoveryMode...
Extracting batterylow0@2x~iphone.t7000.im4p...
Personalizing IMG4 component BatteryLow0...
Extracting batterylow1@2x~iphone.t7000.im4p...
Personalizing IMG4 component BatteryLow1...
[... 814 lines cut ...]
ASR STATUS: start       391     multicast-client
ASR: Waiting for connection attempt from server
ASR STATUS: setup
ASR: Validating target...done
ASR STATUS: metadata
ASR: Validating source...
ASR: asr: Could not use image for multicast operation. Please use "asr imagescan"
ASR: Can't gather image metadata
ASR: Could not validate source - No such file or directory
ASR STATUS: fail
[19:52:50.0137-GMT]{4>18} CHECKPOINT FAILURE:(FAILURE:14) RESTORED:[0x0624] restore_images [0]D(failed to restore image)
executing /usr/sbin/nvram restore-step-results={0x11070624:{0:14}}
executing /usr/sbin/nvram restore-step-codes={0x11070624:{0:14}}
executing /usr/sbin/nvram restore-step-domains={0x11070624:{0:"AMRestoreErrorDomain"}}
executing /usr/sbin/nvram -s restore-step-error={0x11070624:"[0]D(failed to restore image)"}
[19:52:50.0292-GMT]{4>18} CHECKPOINT NOTICE: (NVRAM set) restore-step-error={0x11070624:"[0]D(failed to restore image)"} [sync=true] (first failure)
[19:52:50.0293-GMT]{4>18} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
executing /usr/sbin/nvram restore-step-ids={0x11030624:50;0x11030647:51}
executing /usr/sbin/nvram restore-step-names={0x11030624:restore_images;0x11030647:cleanup_check_result}
[19:52:50.0348-GMT]{4>18} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
executing /usr/sbin/nvram restore-step-ids={0x11030624:50}
executing /usr/sbin/nvram restore-step-names={0x11030624:restore_images}
[19:52:50.0395-GMT]{4>18} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
executing /usr/sbin/nvram restore-step-ids={0x11030624:50;0x11030648:52}
executing /usr/sbin/nvram restore-step-names={0x11030624:restore_images;0x11030648:cleanup_send_final_status}

chacaaan, 2016-10-05:
what should i do? can  someone help me for fix this problem ? thanks

nikias, 2016-10-05:
@chacaaan assuming from the file name it seems like you try to flash a custom firmware. That doesn't work though with recent devices.

chacaaan, 2016-10-06:
yaa itrying for bypass icloud because 2 week ago i found a iphone one the street...

this libimobile is working for normal ipsw 9.3.5 , 10.0.1 and 10.02 and also good for downgrade ios ( already tested )

what should i do for successing bypass from using custum firmware ?

nikias, 2016-10-06:
No way for custom firmware for anything newer than iPhone 4. No way to bypass iCloud.
And FCE365's videos are all fake btw.

closed 2016-10-06 by nikias

Nad4, 2016-10-06:
my friend dont lose your time with this CFW , all stupid videos

bigrich0272, 2016-10-06:
Why don't u return the iPhone u found since its a lost/stolen iPhone instead of tying to bypass iCloud cause u wanna be a thief

chacaaan, 2016-10-07:
Yaa i was waiting owner but he never call this phone almost 2 week then suddenly this phone locked whithout notice . Thanks for feedback . That why i asking here .

nikias, 2016-10-07:
The owner locked it remotely. Call the provider and tell them the IMEI and the sim card number so they can return it to the rightful owner.

chacaaan, 2016-10-07:
Okii :)

AjayGhale, 2016-10-11:
this kid now a days, who can even think that FCE vids are real?

GeoSn0w, 2016-10-16:
If you learn to READ the F.A.Q. page and the description you will see that it's clearly stated "WORK IN PROGRESS" and that might require iBoot exploit. Of course, nobody read the F.A.Q. page and the description and expect everything to work on a press of a button.

AjayGhale, 2016-10-16:
why dont you say the truth to every one? all your videos are fake. you have an iBoot xploit? lets compare your iBoot xploit with mine what do you say?? all your are fake!! just say it, YOU ARE A FAKE dont waste your time kid...

kpwn, 2016-10-16:
@GeoSn0w pls delete your account

anodox, 2016-10-26:
Is that true #geosn0w,  cause i got a same problem here....

ASR: Validating target...done
ASR STATUS: metadata
ASR: Validating source...
ASR: asr: Could not use image for multicast operation. Please use "asr imagescan
"
ASR: Can't gather image metadata
ASR: Could not validate source - No such file or directory
ASR STATUS: fail
[03:19:17.0466-GMT]{5>18} CHECKPOINT FAILURE:(FAILURE:14) RESTORED:[0x0624] rest
ore_images [0]D(failed to restore image)
executing /usr/sbin/nvram restore-step-results={0x11070624:{0:14}}
executing /usr/sbin/nvram restore-step-codes={0x11070624:{0:14}}
executing /usr/sbin/nvram restore-step-domains={0x11070624:{0:"AMRestoreErrorDom
ain"}}
executing /usr/sbin/nvram -s restore-step-error={0x11070624:"[0]D(failed to rest
ore image)"}
[03:19:17.0574-GMT]{5>18} CHECKPOINT NOTICE: (NVRAM set) restore-step-error={0x1
1070624:"[0]D(failed to restore image)"} [sync=true] (first failure)
[03:19:17.0575-GMT]{5>18} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_resu
lt
executing /usr/sbin/nvram restore-step-ids={0x11030624:46;0x11030647:47}
executing /usr/sbin/nvram restore-step-names={0x11030624:restore_images;0x110306
47:cleanup_check_result}
[03:19:17.0620-GMT]{5>18} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result

executing /usr/sbin/nvram restore-step-ids={0x11030624:46}
executing /usr/sbin/nvram restore-step-names={0x11030624:restore_images}
[03:19:17.0657-GMT]{5>18} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final
_status
executing /usr/sbin/nvram restore-step-ids={0x11030624:46;0x11030648:48}
executing /usr/sbin/nvram restore-step-names={0x11030624:restore_images;0x110306
48:cleanup_send_final_status}

FDR 03328C88 timeout waiting for command
FDR 03328C88 terminating...
ERROR: Unable to restore device

StaticCraft, 2016-11-12:
Would there be a way to get passed the iPhone 6 iCloud activation lock

givemenismo, 2016-11-13:
This is true anodox and I also have the same problem