libimobiledevice/idevicerestore #120: Can not restore Multiple ios devices in simultaneously

vipinbeni, 2016-10-06:
i have pass two devices id on different terminal but  one device is restore  other is fail enable to connect ,error  is coming kindly help me how can i run it for two or more devices in one time .

nikias, 2016-10-06:
are you using the ECID (use -i command line switch)? That's the only way to distinguish multiple devices in the different modes.

vipinbeni, 2016-10-07:
yes i am using ECID  in command but one device iprocess is fail after a log waiting for device .....
Error: unable to connect to  device in restore mode 
Error: unable to oprn device in restore mode 
Error:Unable to restore device

vipinbeni, 2016-10-07:
stellar@stellar-desktop:~/build-Stellar-Eraser-Desktop_Qt_5_5_1_GCC_32bit-Debug$ sudo ./Restore0 -i 2DDDE115D76 -e -l
NOTE: using cached version data
Found device in Normal mode
GnuTLS error: A TLS fatal alert has been received.
Identified device as k48ap, iPad1,1
Latest firmware is iPad1,1_5.1.1_9B206_Restore.ipsw
Verifying 'iPad1,1_5.1.1_9B206_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 5.1.1
Product Build: 9B206 Major: 9
Device supports Image4: false
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Found ECID 3151936707958
Getting ApNonce in normal mode... d2 12 92 d6 3a aa 99 6f 47 1d a9 44 fd 68 ca d8 a2 c4 b2 c1 
Trying to fetch new SHSH blob
Unable to get SEPNonce
Getting SepNonce in normal mode... failed
WARNING: Unable to find BbChipID node
WARNING: Unable to find BbSkeyId node
ERROR: Unable to get FirmwarePreflightInfo
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Using cached filesystem from 'iPad1,1_5.1.1_9B206_Restore/038-4291-006.dmg'
Entering recovery mode...
INFO: device serial number is GB0371D1Z38
Sending APTicket (2764 bytes)
Extracting iBEC.k48ap.RELEASE.dfu...
Personalizing IMG3 component iBEC...
reconstructed size: 186905
Sending iBEC (186905 bytes)...
Getting ApNonce in recovery mode... d2 12 92 d6 3a aa 99 6f 47 1d a9 44 fd 68 ca d8 a2 c4 b2 c1 
Sending APTicket (2764 bytes)
Recovery Mode Environment:
iBoot build-version=iBoot-1219.62.15
iBoot build-style=RELEASE
radio-error=0x10
radio-error-string=Not present
Sending AppleLogo...
Extracting applelogo.s5l8930x.img3...
Personalizing IMG3 component AppleLogo...
reconstructed size: 7513
Sending AppleLogo (7513 bytes)...
ramdisk-size=Not present
Extracting 038-4361-021.dmg...
Personalizing IMG3 component RestoreRamDisk...
reconstructed size: 17283417
Sending RestoreRamDisk (17283417 bytes)...
Extracting DeviceTree.k48ap.img3...
Personalizing IMG3 component RestoreDeviceTree...
reconstructed size: 63577
Sending RestoreDeviceTree (63577 bytes)...
Extracting kernelcache.release.k48...
Personalizing IMG3 component RestoreKernelCache...
reconstructed size: 6344537
Sending RestoreKernelCache (6344537 bytes)...
About to restore device... 
Waiting for device...
ERROR: Unable to connect to device in restore mode
ERROR: Unable to open device in restore mode
ERROR: Unable to restore device

vipinbeni, 2016-10-07:
Dear  @nikias please help me how can i restore multiple device at same time ..

vipinbeni, 2016-10-07:
rohit@rohit-desktop:~/build-rohit-Eraser-

Desktop_Qt_5_5_1_GCC_32bit-Debug$ sudo ./Restore1 -i 

38D6614EDCA -e -l^C
stellar@stellar-desktop:~/build-Stellar-

Eraser-Desktop_Qt_5_5_1_GCC_32bit-Debug$ sudo ./Restore1 -i 

38D6614EDCA -e -l
[sudo] password for stellar: 
^[[4~Sorry, try 

again.
[sudo] password for stellar: 
NOTE: using cached version 

data
Found device in Normal mode
GnuTLS error: A TLS fatal 

alert has been received.
Identified device as n94ap, 

iPhone4,1
Latest firmware is 

iPhone4,1_9.3.5_13G36_Restore.ipsw
Verifying 

'iPhone4,1_9.3.5_13G36_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 9.3.5
Product Build: 13G36 Major: 13
Device supports Image4: false
Variant: Customer Erase Install (IPSW)
This restore will erase 

your device data.
Found ECID 3905837919690
Getting ApNonce in 

normal mode... f4 91 70 ff 45 46 2e 05 14 0e c9 da 85 91 5c 

01 dc f2 c7 a3 
Trying to fetch new SHSH blob
Unable to get 

SEPNonce
Getting SepNonce in normal mode... failed
WARNING: 

Unable to find BbSkeyId node
Request URL set to 

https://gs.apple.com/TSS/controller?action=2
Sending TSS 

request attempt 1... response successfully received
Received 

SHSH blobs
Using cached filesystem from 

'iPhone4,1_9.3.5_13G36_Restore/058-48374-036.dmg'
Entering 

recovery mode...
INFO: device serial number is C8PMPCF4FMLD
Sending APTicket (2732 bytes)
Extracting 

iBEC.n94.RELEASE.dfu...
Not personalizing component iBEC...
Sending iBEC (278924 bytes)...
Getting ApNonce in recovery 

mode... f4 91 70 ff 45 46 2e 05 14 0e c9 da 85 91 5c 01 dc f2 

c7 a3 
Sending APTicket (2732 bytes)
Recovery Mode 

Environment:
iBoot build-version=iBoot-2817.60.2
iBoot build-

style=RELEASE
Sending AppleLogo...
Extracting 

applelogo@2x~iphone.s5l8940x.img3...
Not personalizing 

component AppleLogo...
Sending AppleLogo (6796 bytes)...
ramdisk-size=0x4000000
Extracting 058-49032-036.dmg...
Not 

personalizing component RestoreRamDisk...
Sending 

RestoreRamDisk (20775180 bytes)...
Extracting 

DeviceTree.n94ap.img3...
Not personalizing component 

RestoreDeviceTree...
Sending RestoreDeviceTree (83532 

bytes)...
Extracting kernelcache.release.n94...
Not 

personalizing component RestoreKernelCache...
Sending 

RestoreKernelCache (10722636 bytes)...
About to restore 

device... 
Waiting for device...
Device 

e5f3c065a9717d7abbd145ae184c55ff40afc309 is now connected in 

restore mode...
Connecting now...

vipinbeni, 2016-10-07:
second command is successful but i have notice one thing in both command i have send here that

Device

e5f3c065a9717d7abbd145ae184c55ff40afc309 is now connected in

restore mode...   is same for the both while i they are different iPhone   so why this e5f3c065a9717d7abbd145ae184c55ff40afc309  is coming same for both

vipinbeni, 2016-10-07:
e5f3c065a9717d7abbd145ae184c55ff40afc309   is coming same on both device when going About to restore

device..

nikias, 2016-10-07:
According to your output only the iPhone restore prints the UDID

nikias, 2016-10-07:
Also it looks like you're using some custom build. Can you please run it with a freshly compiled idevicerestore? With latest code from git I mean.
If you can't do that run your tool with -d command line switch for both devices to get some debug output.

vipinbeni, 2016-10-07:
now i try  latest build but result same could you pls help me how can i do this

nikias, 2016-10-07:
Can you run both commands with -d please? Without debug output I can't really see if there's something wrong.

vipinbeni, 2016-10-08:
Hi kindly see the out put
[device1.txt](https://github.com/libimobiledevice/idevicerestore/files/517401/device1.txt)
[device2.txt](https://github.com/libimobiledevice/idevicerestore/files/517402/device2.txt)

vipinbeni, 2016-10-08:
[1.docx](https://github.com/libimobiledevice/idevicerestore/files/517404/1.docx)
[2.docx](https://github.com/libimobiledevice/idevicerestore/files/517405/2.docx)

nikias, 2016-10-08:
Both outputs are incomplete. You just copied the last part but the beginning is missing.

nikias, 2016-10-08:
Also: have you tried to restore the failing device alone? Does it work then?

vipinbeni, 2016-10-08:
yes separately both devices restore successfully

vipinbeni, 2016-10-08:
Hi @nikias  is this possible to run on multiple device at same time  ? or we have to run one devise at a time ?

nikias, 2016-10-08:
It works. Been able to restore 10 devices in parallel.

nikias, 2016-10-08:
Paste a complete log for the failing device running idevicerestore with -d

nikias, 2016-10-15:
New logs with -d ?

vipinbeni, 2016-10-25:
@nikias when i connect 5 device then it works but any device got stuck in **waiting for device**  and it is random then any  1-2 of 5 device may got this and other restore successfully i am using usbmuxd 1.1 version  kindly help where i am wrong .

vipinbeni, 2016-10-25:
@nikias Could you please tell me how can i run multiple(4-5) device using terminal as when i do this only 1-2 device restore other fails i  use all latest source of libimobiledevice and idevicerestor and dependency

kindly explain for me the process you  able to restore 10 devices in parallel. please  tell me details 

waiting for your kind reply...................

nikias, 2016-10-25:
I can't really see what might be wrong. I have been running the idevicerestore commands as separate processes, just as you do. According to the logs the iPad1 doesn't enter restore mode; however I can't really tell which device is which. Anyway, for the failing devices - or for all of them -  can you see on screen if it loads the apple logo, and a bit later the progress bar appears?
On another note, the restore fails before the devices seem to enter restore mode, so that's not usbmuxd. What version of libusb-1.0 are you using? This might be the issue here.

vipinbeni, 2016-10-26:
Yes i am using libusb-1.0

nikias, 2016-10-26:
That version is 4.5 years old. Please update.

vipinbeni, 2016-10-26:
Kindly provide the link  where i got the latest as i search i found only libusb-1.0

nikias, 2016-10-26:
It's on github: 
https://github.com/libusb/libusb
Latest release:
https://github.com/libusb/libusb/releases/tag/v1.0.21

vipinbeni, 2016-10-27:
@nikias  Thanks for your reply and yes you are right device is fail after logo of apple is appears and a progress bar is start i ahve install latest libusb but still fail on same position i have connect three device  and all the time on device is fail after on same place waiting for device could you please tell me some other thing that help me....