libimobiledevice/idevicerestore #121: idevicerestore: Problem with iBEC extraction

lukpre, 2016-10-08:
Hey everyone

I'm quite new here and trying to help a friend to restore his Phone (iphone 4s) with idevicerestore and the latest iOS 9.3.5.

I have compiled libirecovery & idevicerestore (as well as usbmux and libusbmux to try) directly from your GIT Repository. There where no problems during the Installation but i can't get the ipsw File to the iPhone. I always end up with this:

root@50396b49d864:~# idevicerestore --ecid 000000C64C06AD4B -e -c iPhone-ios/iPhone4,1_9.3.5_13G36_Restore.ipsw 

NOTE: using cached version data
Found device in Recovery mode
Identified device as n94ap, iPhone4,1
Extracting Restore.plist from IPSW
Product Version: 9.3.5
Product Build: 13G36 Major: 13
INFO: device serial number is C8PJT6WKDT9Y
Device supports Image4: false
Custom firmware requested. Disabled TSS request.
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Using cached filesystem from 'iPhone-ios/iPhone4,1_9.3.5_13G36_Restore/058-48374-036.dmg'
Extracting iBEC.n94ap.RELEASE.dfu...
ERROR: zip_name_locate: Firmware/dfu/iBEC.n94ap.RELEASE.dfu
ERROR: Unable to extract iBEC.n94ap.RELEASE.dfu from iPhone-ios/iPhone4,1_9.3.5_13G36_Restore.ipsw
ERROR: Unable to extract component: iBEC
ERROR: Unable to send iBEC to device.
ERROR: Unable to send iBEC

The Phone is in restore Mode and i realy have not enough skills to get this solved. I guess something goes wrong when idevicerestore tries to extract the ipsw File. 

Has anyone an Idea why this is not working or what i'm doing wrong ?

I realy don't have a fast enough Internet to try the -l Option instead so i hope someone could help me.

Would be really appreciated (the guys phone does not work anymore at the moment).

Thank you:
Lukas

nikias, 2016-10-08:
Hi!

First, don't pass the -c parameter. Custom 
firmware doesn't work for anything newer than iPhone 4. If you have the ipsw downloaded, you're good to go.

Second, make sure the ipsw is not corrupt:
```
$ sha1sum iPhone4,1_9.3.5_13G36_Restore.ipsw 6f8e28e55662b35514a18fe010442464f3c596e6 *iPhone4,1_9.3.5_13G36_Restore.ipsw
```

Third, what's the libzlib version you have?

lukpre, 2016-10-10:
Hey 

Thank you very much for your help nikias. I guess the Problem was, the missing libzlib. It was not installed on the system and now the extraction works.

But ... i am unfortunately still not able to Flash the ipsw to the Device. Now i'm stuck with this:

`Sending iBSS (80410 bytes)...
[==================================================] 100.0%
ERROR: Unable to connect to device in DFU mode
ERROR: Unable to get ApNonce from device!
ERROR: Unable to place device into recovery mode from DFU mode
`

Strangely enough idevicerestore first sees the Phone in DFU Mode:

`root@24138393ca1b:~# idevicerestore -e iPhone-ios/iPhone4,1_9.3.5_13G36_Restore.ipsw 
NOTE: Updated version data.
Found device in DFU mode
...
`

Any more suggestions on what is going wrong ?  I've checked about putting the Phone to DFU Mode and also Recovery Mode. So before i started idevicerestore the Phone was in Recovery (then i put it to DFU Mode) and started the script.

When the Phone was not in DFU Mode it was not even able to Upload the Ibec. And as the error is saying it somehow seems to be unable to switch from DFU to Recovery Mode right ?

Realy would appreciate another hint.

Artemka27, 2017-01-07:
Hello!
I am in the same boat:
ERROR: zip_name_locate: Firmware/dfu/iBEC.p107ap.RELEASE.dfu
ERROR: Unable to extract iBEC.p107ap.RELEASE.dfu from .../iPad2,7_9.3.3_13G34_Restore.ipsw
ERROR: Unable to extract component: iBEC
ERROR: Unable to send iBEC to device.
ERROR: Unable to send iBEC

I have the zlib1g-dev and it is already the newest version (1:1.2.8.dfsg-2ubuntu4)
I see that I can extract (just with clicking the mouse) the *.ipsw file, and I can see the iBEC.p107ap.RELEASE.dfu file in there. Even if I exctract it to the Firmware/dfu folder, still the idevice does not pick it up and send it to the device.
Could I kindly ask you for help? I want to use my iPad with the not newest iOS 9.3.3 and can not do it through iTunes (as it even does not work on Ubuntu)

nikias, 2017-01-08:
@Artemka27 idevicerestore can't help you with firmware that's not signed by Apple anymore. The only chance you have is if you stored ApTickets with https://github.com/tihmstar/tsschecker when 9.3.3 was still signed you might be able to upgrade to 9.3.3 with https://github.com/tihmstar/futurerestore
If you didn't save tickets, there's nothing you can do.

@lukpre did you make it work somehow?

Artemka27, 2017-01-08:
@nikias Thank you for fast reply!
Sorry, did not knew about this. (was suspicions, but not sure)
Does this means I still can upgrade to the 9.3.5 (which is signed by Apple now) with your tool?

I  can not get iTunes working in Ubuntu still. But I can download the ipsw (and feed idevicerestore with it.) 

I ask you this question as I have tried with another computer that runs kUbuntu and with different ipsw's but each time I stuck at the ERROR: Unable to extract component: iBEC Looks like the process stuck on the process of extraction.
 
(or should I use futurerestore tool? - it looks like it is quite a new tool and not so many information about the usage of this tool in the Internet)

Thank you again for your help! Happy holidays!
Artem

nikias, 2017-01-08:
Upgrading to 9.3.5 should work, however something seems broken with libzip

Artemka27, 2017-01-08:
Thank you!
I tried with option downloading 9.3.5
It looks like I have some progress, but I see that now it is
Failed to resize system partition option

Artem

nikias, 2017-01-08:
Are you sure you're using the latest code from git master?

Artemka27, 2017-01-08:
Sure. I just did it today from this repo.
Artem

nikias, 2017-01-09:
Paste a full output of the restore command please.

Artemka27, 2017-01-09:
I left the device at night at my friend's place who has windows with iTunes installed. It completed the update, and now I connected it again to my Ubuntu computer. The sudo idevicerestore -l command did everything correctly for this time.
I am sorry I can not reproduce the error I had before. (If there is a bash output history, and you could tell me where to look - please let me know, I will put it here)

Kind Regards
Artem

nikias, 2017-01-09:
bash history only stores commands, not their output.

closed 2017-01-09 by nikias

vicmans, 2017-09-05:
My device is in DFU mode, i can't out from this mode, I am trying restore mi device:

```
idevicerestore -u fabaaff9-2e3b-4086-9d93-304d305303b0 -c "iPhone4,1_9.3.5_13G36_Restore.ipsw"
NOTE: using cached version data
Found device in DFU mode
Identified device as iPhone4,1
Extracting Restore.plist from IPSW
Product Version: 9.3.5
Product Build: 13G36 Major: 13
Device supports Image4: false
Custom firmware requested. Disabled TSS request.
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without losing data.
Using cached filesystem from 'C:\Users\VIT P2413\Downloads\iPhone4,1_9.3.5_13G36_Restore/058-48374-036.dmg'
Extracting iBSS.n94ap.RELEASE.dfu...
ERROR: zip_name_locate: Firmware/dfu/iBSS.n94ap.RELEASE.dfu
ERROR: Unable to extract iBSS.n94ap.RELEASE.dfu from C:\Users\VIT P2413\Downloads\iPhone4,1_9.3.5_13G36_Restore.ipsw
ERROR: Unable to extract component: iBSS
ERROR: Unable to send iBSS to device
ERROR: Unable to place device into recovery mode from DFU mode
```
In iTunes I can't restore my device, this show error 14.

vicmans, 2017-09-06:
I run:
`idevicerestore --ecid 0x32F7402BD29 iPhone4,1_9.3.5_13G36_Restore.ipsw`
This message appear
```
NOTE: using cached version data
Found device in Recovery mode
Identified device as iPhone4,1
Extracting BuildManifest from IPSW
Product Version: 9.3.5
Product Build: 13G36 Major: 13
INFO: device serial number is C3KJR5UJDTC0
Device supports Image4: false
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without losing data.
Found ECID 3502344682793
Getting ApNonce in recovery mode... 30 86 5f d2 79 82 97 b4 1a 2b 68 d9 47 fb b8 9a 56 b4 05 f6
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode...
WARNING: Unable to find BbSkeyId node
...
...
...

SYSLOG: Sep  6 15:00:16 localhost asr[29] <Error>: bootstrap_look_up2 failed with 0x44e
ASR: asr: Unable to disable idle sleep - -536870199
ASR STATUS: start       242.6   multicast-client
ASR: Waiting for connection attempt from server
SYSLOG: Sep  6 15:00:16 localhost asr[29] <Error>: __net_helper_get_connection_block_invoke_3 could not connect to networkd
ASR STATUS: setup
ASR: Validating target...
ASR: done
ASR STATUS: metadata
ASR: Validating source...
ASR: Can't gather image metadata
ASR: Could not validate source - Authentication error
ASR STATUS: fail

ERROR: Unable to restore device
```

I can't restore the iPhone System

killamoddingteam, 2017-11-10:
C:\Users\sabby\Desktop\libi>idevicerestore.exe -u --71F75ED15C64ECDCE7C795BFADCD0576C7BB6247 -c iphone.ipsw
NOTE: using cached version data
Found device in Recovery mode
Identified device as n71map, iPhone8,1
Extracting Restore.plist from IPSW
Product Version: 10.3.2
Product Build: 14F89 Major: 14
INFO: device serial number is DNPR5267GRY7
Device supports Image4: true
Custom firmware requested. Disabled TSS request.
ERROR: zip_name_locate: Firmware/all_flash/all_flash.n71map.production/manifest
ERROR: could not extract Firmware/all_flash/all_flash.n71map.production/manifest from IPSW

I NEED HELP

EthanHunt002, 2020-03-16:
C:\imobiledevice>idevicerestore.exe -c iPad2,1_9.3.5_13G36_Restore.ipsw -c
NOTE: using cached version data
Found device in Recovery mode
Identified device as iPad2,1
Extracting Restore.plist from IPSW
Product Version: 9.3.5
Product Build: 13G36 Major: 13
INFO: device serial number is DLXFJYYDDKPH
Device supports Image4: false
Custom firmware requested. Disabled TSS request.
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without losing data.
Using cached filesystem from 'iPad2,1_9.3.5_13G36_Restore/058-49123-036.dmg'
Extracting iBEC.k93ap.RELEASE.dfu...
ERROR: zip_name_locate: Firmware/dfu/iBEC.k93ap.RELEASE.dfu
ERROR: Unable to extract iBEC.k93ap.RELEASE.dfu from iPad2,1_9.3.5_13G36_Restore.ipsw
ERROR: Unable to extract component: iBEC
ERROR: Unable to send iBEC to device.
ERROR: Unable to send iBEC

i too have this issue..... a lil bit of help would be greatly appreciated❤😊