libimobiledevice/idevicerestore #124: Device never fully restores

madskonradsen, 2016-10-24:
Using idevicerestore -i blahblah -x ./pathto_ios_9.4.5.ipsw

I get the following beautiful output(this is just the last part of it as I assume it's the most important part):

```
...
Resizing system partition (51)
Unmounting filesystems (29)
Unmounting filesystems (29)
About to send filesystem...
Connected to ASR
Validating the filesystem
Filesystem validated
Sending filesystem now...
Done sending filesystem
Verifying restore (14)
Mounting filesystems (16)
Mounting filesystems (16)
About to send KernelCache...
Extracting kernelcache.release.j2a...
Not personalizing component KernelCache...
Sending KernelCache now...
Done sending KernelCache
Installing kernelcache (27)
Not sending NORData... Quitting...
Cleaning up...
DONE
```

But when it's done and the iDevice reboots it's still stuck with the "connect to iTunes"-sign indicating that the restore process didn't go as smoothly as expected.

nikias, 2016-10-25:
After the restore when the device shows the connect to itunes screen, can you run
```
irecovery -s
setenv auto-boot true
saveenv
reboot
```
and see if it helps?

madskonradsen, 2016-10-26:
Hey again! Just tried it - the device shows the Apple logo and then shuts off. Hmm... Could be a problem with the bootloader maybe? Does irecovery have any means of extracting logs from the boot?

nikias, 2017-06-15:
If you use ```-x``` it won't update the boot loader and other important components so your device won't be able to boot. This was a feature for really old iOS versions (3.x) to skip the baseband update.

closed 2017-06-15 by nikias