libimobiledevice/idevicerestore #128: Cannot restore iphone 7 to iOS 10.1.1

kxj0o, 2016-11-03:
NOTE: using cached version data
Found device in Normal mode
Identified device as d10ap, iPhone9,1
Extracting BuildManifest from IPSW
Product Version: 10.0.2
Product Build: 14A456 Major: 14
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without losing data.
Found ECID 2825273046270010
Getting ApNonce in normal mode... 7c 52 81 0e 7f 0e 6e 2c cf af d3 e7 b5 bb e7 f7 e8 54 fb e0 41 0c 13 c1 a4 98 02 d4 1a d9 8f 1a
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... af c8 83 e8 43 88 77 a2 9d aa df ad 54 9b 2772 6b 26 51 bc
WARNING: Unable to find BbSkeyId node
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting filesystem from IPSW
Entering recovery mode...
INFO: device serial number is C6KSCVTDHG70
Extracting iBEC.d10.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (399668 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-3406.1.79
iBoot build-style=RELEASE
Sending AppleLogo...
Extracting applelogo@2x~iphone.t8010.im4p...
Personalizing IMG4 component AppleLogo...
Sending AppleLogo (12820 bytes)...
ramdisk-size=0x10000000
Extracting 058-50460-076.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (41121971 bytes)...
Extracting DeviceTree.d10ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (149616 bytes)...
Extracting kernelcache.release.d10...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (10236870 bytes)...
About to restore device...
Waiting for device...
Device ffffffffffffffffffffffffffffffff0000002a is now connected in restore mode...
Connecting now...
Connected to com.apple.mobile.restored, version 14
Device ffffffffffffffffffffffffffffffff0000002a has successfully entered restore mode
Hardware Information:
BoardID: 8
ChipID: 32784
UniqueChipID: 2825273046270010
ProductionMode: true
Starting FDR listener thread
About to send NORData...
Found firmware path Firmware/all_flash/all_flash.d10ap.production
Getting firmware manifest Firmware/all_flash/all_flash.d10ap.production/manifest

Extracting LLB.d10.RELEASE.im4p...
Personalizing IMG4 component LLB...
Extracting iBoot.d10.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Extracting DeviceTree.d10ap.im4p...
Personalizing IMG4 component DeviceTree...
Extracting applelogo@2x~iphone.t8010.im4p...
Personalizing IMG4 component AppleLogo...
Extracting recoverymode@1334~iphone-lightning.t8010.im4p...
Personalizing IMG4 component RecoveryMode...
Extracting batterylow0@2x~iphone.t8010.im4p...
Personalizing IMG4 component BatteryLow0...
Extracting batterylow1@2x~iphone.t8010.im4p...
Personalizing IMG4 component BatteryLow1...
[... 1152 lines cut ...]
[09:41:16.0133-GMT]{4>6} CHECKPOINT FAILURE:(FAILURE:14) RESTORED:[0x0624] resto
re_images [0]D(failed to restore image)
executing /usr/sbin/nvram restore-step-results={0x11070624:{0:14}}
executing /usr/sbin/nvram restore-step-codes={0x11070624:{0:14}}
executing /usr/sbin/nvram restore-step-domains={0x11070624:{0:"AMRestoreErrorDom
ain"}}
executing /usr/sbin/nvram -s restore-step-error={0x11070624:"[0]D(failed to rest
ore image)"}
[09:41:16.0202-GMT]{4>6} CHECKPOINT NOTICE: (NVRAM set) restore-step-error={0x11
070624:"[0]D(failed to restore image)"} [sync=true] (first failure)
[09:41:16.0203-GMT]{4>6} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_resul
t
executing /usr/sbin/nvram restore-step-ids={0x11030624:50;0x11030647:51}
executing /usr/sbin/nvram restore-step-names={0x11030624:restore_images;0x110306
47:cleanup_check_result}
[09:41:16.0229-GMT]{4>6} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
executing /usr/sbin/nvram restore-step-ids={0x11030624:50}
executing /usr/sbin/nvram restore-step-names={0x11030624:restore_images}
[09:41:16.0254-GMT]{4>6} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_
status
executing /usr/sbin/nvram restore-step-ids={0x11030624:50;0x11030648:52}
executing /usr/sbin/nvram restore-step-names={0x11030624:restore_images;0x110306
48:cleanup_send_final_status}

ERROR: Unable to restore device

EdelMaks, 2017-06-12:
I have same error: "ERROR: Unable to read OOB data from filesystem offset" with iPhone 5s (trying to flash the latest image, which is downloaded correctly).

nikias, 2017-06-15:
Well it seems like fread() fails to read from the specified offset. It seems to me that for some reason idevicerestore wasn't configured with large file support so it might not be able to read the file properly. Check the autogen.sh (configure) output.