libimobiledevice/idevicerestore #129: Unable to install libimobiledevice on my mac (OS 10.12) with xcode 7.3.1

jeevan1524, 2016-11-03:
Hi, I am trying to install libimobiledevice on my mac which has OS 10.12 and xcode 7.3.1

These are the steps I am performing on the Terminal:
------------------------------------------------------
ebs-macbook-air:~ EB_QA_AUTO$ brew install libimobiledevice

Error: Your Xcode (7.3.1) is outdated.
Please update to Xcode 8.1 (or delete it).
Xcode can be updated from the App Store.
 
Please help.!!

jevinskie, 2016-11-09:
Did you try to upgrade your Xcode to version 8.1 as Homebrew suggested? I don't think Apple even supports Xcode 7.3 on 10.12 thought it might work for some definitions of "work".

applebeast, 2017-04-06:
Dude, same here. I dual booted linux and compiled there, and it workdd like magic. Tried homebrew, tried macports. Neither cuts the mustard, believe me.