libimobiledevice/idevicerestore #130: Unable to connect to device in Recovery Model

FranAngel, 2016-11-08:
Hi! I´m trying to install iOS on my iPhone 5c in Recovery Mode. I´m using latest Ubuntu v16.04 LTS and already installed libimobiledevice-dev -utils etc. 
Usbmuxd does not detect the device

Could you please help me?

Thanks!

nikias, 2016-11-08:
in recovery mode the device is not visible through usbmuxd. If you don't own the usb device node (which is pretty likely in ubuntu) you might want to run idevicerestore as root.

FranAngel, 2016-11-09:
Thank you! 
I´m new to Linux.  I´ve downloaded the idevicerestore zip file but I don´t know how to install. I´ve searched the install procedure but I don´t find it
Thank you again

FranAngel, 2016-11-11:
Hi! I finnally installed idevicerestore successfully. I´m running it, but I got an error message: "Unable to restore device"
I send you the full process:

root@UbuntuiPhone:/# idevicerestore -e /home/fran/Downloads/iPhone_4.0_32bit_10.1.1_14B100_Restore.ipsw
NOTE: using cached version data
Found device in Recovery mode
Identified device as n49ap, iPhone5,4
Extracting BuildManifest from IPSW
Product Version: 10.1.1
Product Build: 14B100 Major: 14
INFO: device serial number is F8WMT0R3FFT5
Device supports Image4: false
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Found ECID 801045767049
Getting ApNonce in recovery mode... 1c 2b c8 6b 0a 4e d9 5e 22 d8 59 80 4b 09 f7 1a 22 0a d9 11 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 
WARNING: Unable to find BbSkeyId node
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting filesystem from IPSW
[==================================================] 100.0%
Sending APTicket (2763 bytes)
Extracting iBEC.n49.RELEASE.dfu...
Not personalizing component iBEC...
Sending iBEC (295180 bytes)...
Getting ApNonce in recovery mode... 1c 2b c8 6b 0a 4e d9 5e 22 d8 59 80 4b 09 f7 1a 22 0a d9 11 
Sending APTicket (2763 bytes)
Recovery Mode Environment:
iBoot build-version=iBoot-3406.20.16
iBoot build-style=RELEASE
Sending AppleLogo...
Extracting applelogo@2x~iphone.s5l8950x.img3...
Not personalizing component AppleLogo...
Sending AppleLogo (6668 bytes)...
ramdisk-size=0x10000000
Extracting 058-52516-078.dmg...
Not personalizing component RestoreRamDisk...
Sending RestoreRamDisk (24223884 bytes)...
Extracting DeviceTree.n49ap.img3...
Not personalizing component RestoreDeviceTree...
Sending RestoreDeviceTree (88716 bytes)...
Extracting kernelcache.release.n49...
Not personalizing component RestoreKernelCache...
Sending RestoreKernelCache (11613452 bytes)...
About to restore device... 
Waiting for device...
Device 8dc958784520ed906f9adad26e1e79116a0687de is now connected in restore mode...
Connecting now...
Connected to com.apple.mobile.restored, version 14
Device 8dc958784520ed906f9adad26e1e79116a0687de has successfully entered restore mode
Hardware Information:
BoardID: 14
ChipID: 35152
UniqueChipID: 801045767049
ProductionMode: true
Previous restore exit status: 0x100
Starting FDR listener thread
ERROR: Unable to connect to FDR client (-2)
ERROR: Failed to start FDR Ctrl channel
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
Checking filesystems (15)
Previous Restore Log Received:
SYSLOG: Nov 11 10:36:33 localhost bootlog[0] <Notice>: BOOT_TIME 1478860593 27832
[10:36:55.0300-GMT]{5>20} CHECKPOINT NOTICE: Not image4 device: AP nonce not clearable
[... 2593 lines cut ...]
[10:48:21.0185-GMT]{5>20} CHECKPOINT FAILURE:(FAILURE:-1) RESTORED:[0x0635] update_device_firmware [0]D(failed to update device firmware)[1]D(ramrod firmware update failed)[2]D(update_baseband: failed querying baseband info)[3]D(Failed creating a communication transport to baseband.)
executing /usr/sbin/nvram restore-step-results={0x10070635:{0:-1};0x10071303:{0:-1}}
executing /usr/sbin/nvram restore-step-codes={0x10070635:{0:-1};0x10071303:{0:1004}}
executing /usr/sbin/nvram restore-step-domains={0x10070635:{0:"AMRestoreErrorDomain"};0x10071303:{0:"RamrodErrorDomain"}}
executing /usr/sbin/nvram restore-step-error={0x10070635:"[0]D(failed to update device firmware)[1]D(ramrod firmware update failed)[2]D(update_baseband: failed querying baseband info)[3]D(Failed creating a communication transport to baseband.)"}
executing /usr/sbin/nvram restore-step-uptime=221
executing /usr/sbin/nvram restore-step-user-progress=63
[10:48:21.0375-GMT]{5>20} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
executing /usr/sbin/nvram restore-step-ids={0x10030635:57;0x10031303:61;0x10030647:62}
executing /usr/sbin/nvram restore-step-names={0x10030635:update_device_firmware;0x10031303:update_baseband;0x10030647:cleanup_check_result}
executing /usr/sbin/nvram restore-step-uptime=221
executing /usr/sbin/nvram restore-step-user-progress=63
[10:48:21.0491-GMT]{5>20} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
executing /usr/sbin/nvram restore-step-ids={0x10030635:57;0x10031303:61}
executing /usr/sbin/nvram restore-step-names={0x10030635:update_device_firmware;0x10031303:update_baseband}
executing /usr/sbin/nvram restore-step-uptime=221
executing /usr/sbin/nvram restore-step-user-progress=63
[10:48:21.0608-GMT]{5>20} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
executing /usr/sbin/nvram restore-step-ids={0x10030635:57;0x10031303:61;0x10030648:63}
executing /usr/sbin/nvram restore-step-names={0x10030635:update_device_firmware;0x10031303:update_baseband;0x10030648:cleanup_send_final_status}
executing /usr/sbin/nvram restore-step-uptime=221
executing /usr/sbin/nvram restore-step-user-progress=63

ERROR: Unable to restore device
root@UbuntuiPhone:/#

nikias, 2017-06-15:
Seems like baseband update doesn't succeed. If it can't be restored with iTunes either, the device might be defective.

closed 2017-06-15 by nikias

sseen, 2017-07-13:
All error like

> ERROR: Unable to connect to recovery device
Unable to place device into recovery mode from DFU mode

is come from using the wrong usb cable（lighting to usb cable or Apple 30-pin to USB Cable）, u must use the MFI accessories. Apple recommends using only accessories that Apple has certified and that come with the "Made for iPod/iPhone/iPad" logo.

Once u change using the MFI usb cable, everything will be fine.