libimobiledevice/idevicerestore #132: Usb port disabled after error when restoring multiple devices

ccups, 2016-11-10:
Hi, I'm trying to get multiple instances of idevicerestore working at the same time and I'm having issues with a usb port being disabled after receiving an error from idevicerestore.

I'm running idevicerestore on 10 devices simultaneously with 7 plugged into a self powered usb hub and 3 directly into the computer.

My OS is 37~14.04.1-Ubuntu SMP Linux Mint 17.3 and I installed the latest files from this repository.

I start by killing any existing usbmuxd processes and start a new one with...
sudo usbmuxd -X
sudo usbmuxd -f

Then I open 10 terminal windows and initiate the restores with the following 
command...

sudo idevicerestore -u mydeviceudid -i mydeviceecidhex -e -l

Note: I'm using both the udid and the ecid in my command because I have had issues in the past when running this with multiple devices where I'll have a device try to connect to the wrong device regardless of the fact that I defined the udid in my command.  Since adding both the udid and the ecid I have not seen this issue again.

9 out of 10 finished without any errors, but I got the following error on one device...

idevicerestore -u ed31be2a3e261f38cc587deb5a9bc4cbfdb4348f -i 63788953650 -e -l
NOTE: using cached version data
Found device in Normal mode
Identified device as n51ap, iPhone6,1
Latest firmware is iPhone_4.0_64bit_10.1.1_14B150_Restore.ipsw
Verifying 'iPhone_4.0_64bit_10.1.1_14B150_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 10.1.1
Product Build: 14B150 Major: 14
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Found ECID 6835584448080
Getting ApNonce in normal mode... 7c 64 f2 8e 8d 04 2b d3 15 3b 6a 92 92 1a 3a 19 d7 14 d0 82 
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... dd b2 4c e3 c4 ab 40 d1 7f 4e c8 2d 95 79 5a 3f e5 6d c6 41 
WARNING: Unable to find BbSkeyId node
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Using cached filesystem from 'iPhone_4.0_64bit_10.1.1_14B150_Restore/058-50564-079.dmg'
Entering recovery mode...
INFO: device serial number is C39LQLVXFNJK
Extracting iBEC.n51.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (623768 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-3406.20.16
iBoot build-style=RELEASE
Sending AppleLogo...
Extracting applelogo@2x~iphone.s5l8960x.im4p...
Personalizing IMG4 component AppleLogo...
Sending AppleLogo (11639 bytes)...
ramdisk-size=0x10000000
Extracting 058-52511-082.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (40199190 bytes)...
Extracting DeviceTree.n51ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (107388 bytes)...
Extracting kernelcache.release.n51...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (11906828 bytes)...
About to restore device... 
Waiting for device...
restore_is_current_device: device df15a45017cf5911c04d56474ce3b16f68179037 is 
not in restore mode
restore_is_current_device: device d45b0477169563d0bed346d723dd08ca238a34ce is 
not in restore mode
restore_is_current_device: device cac7bf0e762f7e23e11e9dbca73f5d8259f2dc0b is 
not in restore mode
ERROR: Unable to connect to device in restore mode
ERROR: Unable to open device in restore mode
ERROR: Unable to restore device

After this error the light on the usb hub port failed to turn on again and did not recognize any device I plugged into it.  

In other trials of 10 devices at the same time all 10 worked, so I'm not sure what is causing this problem or where to look to find a solution.

Also, after turning the usb hub power off and reconnecting the disabled port works again.

Looking at my kernel log, the follow line was repeated hundreds of times with the same timestamp...
Nov 10 10:55:59 ccupsCPU kernel: [83373.378967] hub 1-13.1:1.0: hub_port_status failed (err = -71)

Does anybody have any ideas as to what could be causing this error and why it disables my usb port?

Any help with this would be greatly appreciated, thanks.

nikias, 2016-11-10:
USB hubs have been causing issues in the past too. Even on Mac computers. What type of USB hub do you have?

ccups, 2016-11-10:
That error was on an Anker 7 port usb 3.0 hub.

I just tried 10 on another hub we have (Startech ST1030USBM)  This hub is brand new.
https://www.startech.com/Cards-Adapters/USB-3.0/Hubs/10-port-industrial-usb-3-0-hub~ST1030USBM

The results were different on the Startech hub...  Only 6 worked and 4 failed with "Not data to read" getting spammed in debug mode in what seemed to be an infinite loop.  The phones got stuck on the itunes screen so I killed it with ctl c. 

Is there a limit to 6 devices per hub?  Because on the hub of 7 one failed and 4 failed on the 10.

nikias, 2016-11-10:
Hm Anker seems like a good one to me... Not sure why this happens

nikias, 2016-11-10:
There is no limit from the software side. Not sure what the kernel is doing though. It might be an issue with the usb 3 driver. What USB chipset does the computer have? I observed issues with certain USB chipsets (especially ATI/AMD).

ccups, 2016-11-10:
Where can I find which USB chipset I have on Linux?  I have a Dell XPS 8910 i5 originally installed with Windows, but it was formatted with Linux Mint as the current OS.

What would be the optimal setup to get idevicerestore running simultaneously with around 30 devices?

nikias, 2016-11-11:
Hm that one should have an Intel chipset. Those are usually doing fine from my experience. 30 devices produce a high load on the USB bus so I'd go with 10 in parallel. I have observed in similar setups that for some reason sometimes the restore just fails (for no obvious reason). However it wouldn't disconnect the usb hub...

ccups, 2016-11-11:
Do you know a good way to view details about the USB bus load so I can try to calculate how many devices one hub can handle? 

I just tried 2 hubs with 5 devices on each and 1 failed out of 10.  Same results in two trials.  Would getting a good pci express card help?

nikias, 2016-11-12:
No, I just made this observation in the field. Best practice seems to be having 10 device in parallel in a queue and repeat those that fail while continuing with the next bunch.
I think your hardware is probably fine, I don't think a pcie card will give different results.
Was the same device failing again or was it another device?

c0ry, 2016-11-14:
The devices that fail are different, but there seems to be a pattern on which hub slot fails.  For example, I've had 1 fail out of 5 on a single hub and it is usually on position number 4 of the hub.  I've had the same results with 2 different hubs using different devices on each.

When they fail, I'm sometimes seeing preflight errors on usbmuxd and idevicerestore usually locks up with the error... "ERROR: Unable to receive message from FDR"

ccups, 2016-11-14:
The devices that fail are different, but there seems to be a pattern on which hub slot fails. For example, I've had 1 fail out of 5 on a single hub and it is usually on position number 4 of the hub. I've had the same results with 2 different hubs using different devices on each.

When they fail, I'm sometimes seeing preflight errors on usbmuxd and idevicerestore usually locks up with the error... "ERROR: Unable to receive message from FDR"

daniel-bluesea, 2016-11-15:
@ccups  @nikias  is it an "official" recommendation to use both -u and -i flags while performing multi-device restoring?

nikias, 2016-11-15:
@daniel-bluesea the normal practice is to just use -i (ECID)

nikias, 2016-11-15:
@ccups this is so weird. And the hub disconnects when it fails with these hubs? I wonder what happens when you disable the xhci driver so you just have ehci (USB 2.0) enabled. Try this:
```
for i in /sys/bus/pci/drivers/xhci_hcd/*:*; do
  echo "${i##*/}" > "${i%/*}/unbind"
done
```
This disables the USB 3 support.

To renenable it:
```
for i in /sys/bus/pci/drivers/xhci_hcd/*:*; do
  echo "${i##*/}" > "${i%/*}/bind"
done
```
Not sure if this actually works, but it's worth a try.

nikias, 2016-11-15:
@ccups unplug all hubs before doing that

daniel-bluesea, 2016-11-16:
@nikias yes you are right, using the -i flag is sufficient (my problem was using the decimal form of the ecid.. after switching to hex, it ran smoothly)

thanks

ccups, 2016-11-16:
@nikias The hub does not disconnect when it fails, but sometimes a port on the hub will get completely disabled until I turn the power off and on again.  So for example, positions 1,2,3 and 5 stay connected on the hub while position number 4 is dead until I reboot the hub.

I've troubleshooted disabling the xhci driver before posting here.  I think I need to rebuild the kernel in order to disable or blacklist it.  I'll try again with your suggestions, but I doubt it will work. 

Also, I tried this on an older computer with no usb 3.0 ports a few weeks ago and I still had issues.

nikias, 2016-11-16:
@ccups I honestly have no clue what might be the reason. I suppose the older computer has different hardware and kernel version. If that's the case I wouldn't even suspect your current kernel (given that "still had issues" means usb hub port disconnected like with your computer in question).

taanders, 2018-09-06:
I have had some issues with idevicerestore and those Startech ST1030USBM hubs. Inside those hubs are three "voltage regulator/controller boards". It seems that when iOS devices are connected to every other port then it seems to work better but when sequential ports are connected then many of started idevicerestore processes are failed for some cannot read FDR message errors. Other tested hubs seems to work fine. This is maybe some kind of hardware issue on those Startech hubs. I have tested those hubs on Windows and Linux and both are failing when sequential ports are connected.

nikias, 2018-09-06:
@taanders thanks for mentioning this. This is by far not the first time I hear about hardware-related issues, and in most cases it was related to USB hubs.