libimobiledevice/idevicerestore #135: Restore multiple devices failed at [restore_is_current_device: device XXXXXis not in restore mode]

jackczhang, 2016-11-22:
I have run the idevicerestore for 10 devices, and part of them failed at below log, and the device screen is at Welcome page, did not restart to restore mode.

About to restore device... 
Waiting for device...
**restore_is_current_device: device 2e4cd66fc5f469cefbeb13d65f7bbfdbb471807f is not in restore mode**
restore_is_current_device: device 2c31df7633bfa0e61d2d50aa3a668eab9bbb237c is not in restore mode
restore_is_current_device: device fa2e0a24f2633cf47a842391ad9c021683c153f2 is not in restore mode
restore_is_current_device: device 2fb5fdc39ad6c31c91d45c2886995be89af5f35b is not in restore mode
restore_is_current_device: device 5233f633e53ca620cad42540c7c319eaa473ff4f is not in restore mode
restore_is_current_device: device 0a13786a638e6ad39b4425ff897513825371d07a is not in restore mode
restore_is_current_device: device 94b34773e628f1494c20059815c1c9de8a873997 is not in restore mode
restore_is_current_device: device cf7c198671d91552b56a099b9d3871957982b53d is not in restore mode

jackczhang, 2016-11-22:
Does this error happen at the multiple thread error of the libplist module?

jackczhang, 2016-11-22:
Just like Crash on multithread environment, xmlCleanup causes crash #68 which is just a workaround

nikias, 2017-06-15:
idevicerestore wasn't really designed to support multiple restores at the same time though it tries its best to detect the correct device in every single step...