libimobiledevice/idevicerestore #138: ERROR: device didn't accept BasebandData

daniel-bluesea, 2016-12-04:
it occurs every now and then, what could be the problem?

more relevant log lines:
About to send BasebandData...
Sending BasebandData now...
Done sending BasebandData
Updating baseband (19)
Updating baseband (19)
Updating baseband (19)
Updating baseband (19)
Got status message
Status: Verification Error

daniel-bluesea, 2016-12-04:
is it possible to add some kind of retry mechanism?

nikias, 2017-06-15:
The setting of UpdateBaseband to false doesn't work because all production devices ignore the restore options and only use defaults predefined in the restore ramdisk.
Why the baseband update might have failed I can't tell... adding a retry doesn't work because the device controls the restore process.

closed 2017-06-15 by nikias