libimobiledevice/idevicerestore #139: Error compilation: In function idevicerestore_start

MaximilienLIX, 2016-12-07:
Hi,
i try to build idevicerestore from an ArchLinux system and i've this error when i execute make.
`idevicerestore-idevicerestore.o: In function `idevicerestore_start':
/tmp/i/idevicerestore/src/idevicerestore.c:716: warning: the use of `tempnam' is dangerous, better use `mkstemp'
/usr/bin/ld: idevicerestore-thread.o: undefined reference to symbol 'pthread_once@@GLIBC_2.2.5'
/usr/lib/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:447: idevicerestore] Error 1
make[2] : on quitte le répertoire « /tmp/i/idevicerestore/src »
make[1]: *** [Makefile:424: all-recursive] Error 1
make[1] : on quitte le répertoire « /tmp/i/idevicerestore »
make: *** [Makefile:356: all] Error 2
`

I've installed librecovery before.

Thx for read this and sorry for my bad english.

eimann, 2017-12-18:
Try this patch:

`From 162a079123f81f14287dcd6d3e5b4488dc6981ee Mon Sep 17 00:00:00 2001
From: Ivan Shapovalov <intelfx100@gmail.com>
Date: Fri, 25 Sep 2015 23:37:27 +0300
Subject: [PATCH] configure.ac: check for pthreads

Fixes building at least on Arch.
---
 configure.ac | 1 +
 1 file changed, 1 insertion(+)

diff --git a/configure.ac b/configure.ac
index d36cc67..950ef02 100644
--- a/configure.ac
+++ b/configure.ac
@@ -35,6 +35,7 @@ AM_PROG_CC_C_O
 AC_PROG_LIBTOOL
 
 # Checks for libraries.
+AC_SEARCH_LIBS([pthread_create], [pthread])
 PKG_CHECK_MODULES(libirecovery, libirecovery >= $LIBIRECOVERY_VERSION)
 PKG_CHECK_MODULES(libimobiledevice, libimobiledevice-1.0 >= $LIBIMOBILEDEVICE_VERSION)
 PKG_CHECK_MODULES(libplist, libplist >= $LIBPLIST_VERSION)
-- 
2.6.2`

egyshell, 2018-05-03:
how i can use this patch