libimobiledevice/idevicerestore #140: Error when download firmware first time

ankitrwt, 2016-12-09:
when I download firmware first time after power on my pc it shows error

root:idevicerestore# idevicerestore -n -l -C /root/Desktop/test/
NOTE: Updated version data.
Entity: line 1: parser error : Space required after the Public Identifier
<!DOCTYPE html PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN">
                                                              ^
Entity: line 1: parser error : SystemLiteral " or ' expected
<!DOCTYPE html PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN">
                                                              ^
Entity: line 1: parser error : SYSTEM or PUBLIC, the URI is missing
<!DOCTYPE html PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN">
                                                              ^
Entity: line 26: parser error : Opening and ending tag mismatch: meta line 4 and head
</head>
       ^
Entity: line 50: parser error : EntityRef: expecting ';'
"http://url.fortinet.net/rate/submit.php?id=5E5F3673281A74213B2474723C7A262A&cat
                                                                               ^
Entity: line 50: parser error : EntityRef: expecting ';'
/url.fortinet.net/rate/submit.php?id=5E5F3673281A74213B2474723C7A262A&cat=19&loc
                                                                               ^
Entity: line 50: parser error : EntityRef: expecting ';'
4213B2474723C7A262A&cat=19&loc=http://itunes%2eapple%2ecom%2fcheck%2fversion&ver
                                                                               ^
Entity: line 54: parser error : Opening and ending tag mismatch: head line 3 and html
</html>
       ^
Entity: line 55: parser error : Premature end of data in tag html line 2

^
ERROR: Cannot parse plist data from '/root/Desktop/test//version.xml'.
Found device in Normal mode
Identified device as n94ap, iPhone4,1
ipsw_get_latest_fw: ERROR: Can't find MobileDeviceSoftwareVersionsByVersion dict in version data
ERROR: can't get URL for latest firmware

nikias, 2016-12-09:
It fails because it doesn't even get the version.xml file correctly. Instead it downloads a fortinet rate limit page. Can you open http://itunes.apple.com/check/version in a browser?

ankitrwt, 2016-12-09:
yes, but why it fails only first time after then it works correctly
do i need to download it manually first in the (-C)cache directory ?

nikias, 2016-12-09:
I don't know. So when you delete that file and run idevicerestore again it will download it correctly? You can manually download it. Put it in the cache directory when you specify a cache directory with -C or the current directory otherwise. However after 24 hours it will be downloaded again whenever you run idevicerestore again.

ankitrwt, 2016-12-09:
yes..

ankitrwt, 2016-12-09:
idevicerestore -n -l it fails to download version.xml file first time when I reboot my system or when I Log In to system after 1-2hrs..  Its not happens every time, but yes sometimes it fails

ankitrwt, 2016-12-09:
Entity: line 1: parser error : Space required after the Public Identifier
<!DOCTYPE html PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN">
                                                              ^
Entity: line 1: parser error : SystemLiteral " or ' expected
<!DOCTYPE html PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN">
                                                              ^
Entity: line 1: parser error : SYSTEM or PUBLIC, the URI is missing
<!DOCTYPE html PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN">
                                                              ^
Entity: line 26: parser error : Opening and ending tag mismatch: meta line 4 and head
</head>
       ^
Entity: line 50: parser error : EntityRef: expecting ';'
"http://url.fortinet.net/rate/submit.php?id=5E5F3673281A74213B2474723C7A262A&cat
                                                                               ^
Entity: line 50: parser error : EntityRef: expecting ';'
/url.fortinet.net/rate/submit.php?id=5E5F3673281A74213B2474723C7A262A&cat=19&loc
                                                                               ^
Entity: line 50: parser error : EntityRef: expecting ';'
4213B2474723C7A262A&cat=19&loc=http://itunes%2eapple%2ecom%2fcheck%2fversion&ver
                                                                               ^
Entity: line 54: parser error : Opening and ending tag mismatch: head line 3 and html
</html>
       ^
Entity: line 55: parser error : Premature end of data in tag html line 2

^
ERROR: Cannot parse plist data from '/home/firmware/ios/ios-master/tempData/iPhone4,1/version.xml'.
Unable to to load version data

ankitrwt, 2016-12-09:
//Entity: line 1: parser error : Space required after the Public Identifier
//<!DOCTYPE html PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN">
                                                              ^
//Entity: line 1: parser error : SystemLiteral " or ' expected
//<!DOCTYPE html PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN">
                                                              ^
//Entity: line 1: parser error : SYSTEM or PUBLIC, the URI is missing
//<!DOCTYPE html PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN">
                                                              ^
//Entity: line 26: parser error : Opening and ending tag mismatch: meta line 4 and head
//</head>
       ^
//Entity: line 50: parser error : EntityRef: expecting ';'
//"http://url.fortinet.net/rate/submit.php?id=5E5F3673281A74213B2474723C7A262A&cat
                                                                               ^
//Entity: line 50: parser error : EntityRef: expecting ';'
///url.fortinet.net/rate/submit.php?id=5E5F3673281A74213B2474723C7A262A&cat=19&loc
                                                                               ^
//Entity: line 50: parser error : EntityRef: expecting ';'
//4213B2474723C7A262A&cat=19&loc=http://itunes%2eapple%2ecom%2fcheck%2fversion&ver
                                                                               ^
//Entity: line 54: parser error : Opening and ending tag mismatch: head line 3 and html
//</html>
       ^
//Entity: line 55: parser error : Premature end of data in tag html line 2

^
//ERROR: Cannot parse plist data from '/home/firmware/ios/ios-master/tempData/iPhone4,1/version.xml'.
Unable to to load version data

ankitrwt, 2016-12-09:
[](Entity: line 1: parser error : Space required after the Public Identifier
<!DOCTYPE html PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN">
                                                              ^
Entity: line 1: parser error : SystemLiteral " or ' expected
<!DOCTYPE html PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN">
                                                              ^
Entity: line 1: parser error : SYSTEM or PUBLIC, the URI is missing
<!DOCTYPE html PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN">
                                                              ^
Entity: line 26: parser error : Opening and ending tag mismatch: meta line 4 and head
</head>
       ^
Entity: line 50: parser error : EntityRef: expecting ';'
"http://url.fortinet.net/rate/submit.php?id=5E5F3673281A74213B2474723C7A262A&cat
                                                                               ^
Entity: line 50: parser error : EntityRef: expecting ';'
/url.fortinet.net/rate/submit.php?id=5E5F3673281A74213B2474723C7A262A&cat=19&loc
                                                                               ^
Entity: line 50: parser error : EntityRef: expecting ';'
4213B2474723C7A262A&cat=19&loc=http://itunes%2eapple%2ecom%2fcheck%2fversion&ver
                                                                               ^
Entity: line 54: parser error : Opening and ending tag mismatch: head line 3 and html
</html>
       ^
Entity: line 55: parser error : Premature end of data in tag html line 2

^
ERROR: Cannot parse plist data from '/home/firmware/ios/ios-master/tempData/iPhone4,1/version.xml'.
Unable to to load version data
"Message: Downloading Fails return:-1")

ankitrwt, 2016-12-09:
Here is my idevicerestore -n -l -C /root/Desktop/test/ output
@nikias 
[image]

ankitrwt, 2016-12-12:
This is the error.. 
@nikias 
[image]

nikias, 2016-12-12:
Your network has some filtering that is messing with your connections. See the "fortinet" URLs in the output. This is not an issue with idevicerestore.

closed 2016-12-12 by nikias