libimobiledevice/idevicerestore #141: Error: No package 'libirecovery' found

ankitrwt, 2016-12-26:
I am using Arch Linux and I installed all dependencies as required one by one but when I install idevicerestore it says configure: error: Package requirements (libirecovery >= 0.2.0) were not met....
Any Help will be appreciated, Thanks
this is the output of ./autogen.sh 

[root@ankit-System idevicerestore-master]# ./autogen.sh 
libtoolize: putting auxiliary files in '.'.
libtoolize: linking file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
libtoolize: Consider adding '-I m4' to ACLOCAL_AMFLAGS in Makefile.am.
configure.ac:33: installing './compile'
configure.ac:8: installing './missing'
src/Makefile.am: installing './depcomp'
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking target system type... x86_64-unknown-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-unknown-linux-gnu file names to x86_64-unknown-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-unknown-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
checking if gcc static flag -static works... yes
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for libirecovery... no
configure: error: Package requirements (libirecovery >= 0.2.0) were not met:

No package 'libirecovery' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables libirecovery_CFLAGS
and libirecovery_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

closed 2016-12-28 by ankitrwt

mounaiban, 2019-01-25:
If anyone is still getting `autogen.sh` error messages that tell you things like
~~~~
checking for libirecovery... no
configure: error: Package requirements (libirecovery >= 0.2.0) not met
~~~~

*even after* you have verified that the latest or correct version of `libirecovery` has successfully compiled and installed, it may be due to `libirecovery` having been installed in the wrong path.

Repeating the **`libirecovery`** compilation and installation process from scratch with a slightly different script from what's written in the `README` file

~~~~
./autogen.sh -prefix=/usr
make
sudo make install
~~~~

has worked for me on Fedora 29 Workstation. However, I encourage you to try this regardless of which distro you are running.

### Explanation
The compilation process for `idevicerestore` expects to find `libirecovery` in the `lib` subdirectories under `/usr` (i.e. `/usr/lib` or `/usr/lib64`). However, on some systems (like, practically *every* Linux system I have used lately), the default installation path for `libirecovery` (and pretty much most software I have built from source) is `/usr/local`.

This is evidenced by the message that appears after the autogen stage of the `libirecovery` installation:
 ~~~~
$ ./autogen.sh

## hundreds of lines omitted ##

Configuration for libirecovery ...
----------------------------------------

  Install prefix ...........: /usr/local
  USB backend ..............: libusb...

~~~~
> Version numbers have been redacted to minimise confusion as they may change by the time you read this.

If you stuck with the defaults, `libirecovery` would have in fact been installed on your system, but in the wrong directories at `/usr/local/lib64` (or `/usr/local/lib`) instead! Setting the `-prefix=/usr` option when running `autogen.sh` during the `libirecovery` installation process overrides the path prefix so that everything goes in the expected directory at `/usr/lib*`

See also Issue 13 on the **libirecovery** repo: https://github.com/libimobiledevice/libirecovery/issues/13

kristenlc, 2022-09-09:
It is now September 8, 2022. I have gone through 6 hours of headaches trying to get idevicerestore to compile, trying every possible source on the internet. All leads back to this error message. macOS High Sierra 10.13.6, Xcode 9.4.1.

I finally got it to work after compiling and installing libirecovery from source, and creating symlinks manually by hand.

until now I have tried various different versions of irecovery, I have tried adding -prefix=/usr to ./autogen.sh before compiling and I have tried without that. I have tried installing from the homebrew formula, I have tried linking with homebrew after installing from source, I have tried installing libirecovery from homebrew then removing the files in /usr/local/lib for irecovery and installing in the default prefix from source and then relinking using the command line tools with homebrew. I have tried macports install of libirecovery. I have tried every single possible thing mentioned on the internet that I could find in 6 hours, and no luck.

I am going to attach what fixed this error for me and I hope it benefits someone, or two, or a hundred people.

> cd /usr/lib
> sudo ln -s libirecovery-1.0.la libirecovery-0.2.la # maybe not necessary 
> sudo ln -s libirecovery-1.0.a libirecovery-0.2.a # maybe not necessary 
> sudo ln -s libirecovery-1.0.dylib libirecovery-0.2.dylib # maybe not necessary 
> 
> sudo ln -s libirecovery-1.0.la libirecovery-0.2.0.la # maybe not necessary 
> sudo ln -s libirecovery-1.0.a libirecovery-0.2.0.a # maybe not necessary 
> sudo ln -s libirecovery-1.0.dylib libirecovery-0.2.0.dylib # maybe not necessary 
> 
> sudo ln -s libirecovery-1.0.la libirecovery.la 
> sudo ln -s libirecovery-1.0.a libirecovery.a
> sudo ln -s libirecovery-1.0.dylib libirecovery.dylib 
> 
> sudo ln -s libplist-2.0.la libplist.la
> sudo ln -s libplist-2.0.a libplist.a
> sudo ln -s libplist-2.0.dylib libplist.dylib
> 
> sudo ln -s libplist++-2.0.la libplist++.la # maybe not necessary 
> sudo ln -s libplist++-2.0.a libplist++.a  # maybe not necessary 
> sudo ln -s libplist++-2.0.dylib libplist++.dylib # maybe not necessary 
> 
> cd pkgconfig
> 
> sudo ln -s libirecovery-1.0.pc libirecovery-0.2.pc # maybe not necessary
> sudo ln -s libirecovery-1.0.pc libirecovery-0.2.0.pc # maybe not necessary
> sudo ln -s libirecovery-1.0.pc libirecovery.pc
> 
> sudo ln -s libplist-2.0.pc libplist.pc
> sudo ln -s libplist++-2.0.pc libplist++.pc # maybe not necessary
> 
> cd /usr/local/lib
> sudo ln -s libirecovery-1.0.la libirecovery-0.2.la # maybe not necessary 
> sudo ln -s libirecovery-1.0.a libirecovery-0.2.a # maybe not necessary 
> sudo ln -s libirecovery-1.0.dylib libirecovery-0.2.dylib # maybe not necessary 
> 
> sudo ln -s libirecovery-1.0.la libirecovery-0.2.0.la # maybe not necessary 
> sudo ln -s libirecovery-1.0.a libirecovery-0.2.0.a # maybe not necessary 
> sudo ln -s libirecovery-1.0.dylib libirecovery-0.2.0.dylib # maybe not necessary 
> 
> sudo ln -s libirecovery-1.0.la libirecovery.la 
> sudo ln -s libirecovery-1.0.a libirecovery.a
> sudo ln -s libirecovery-1.0.dylib libirecovery.dylib 
> 
> sudo ln -s libplist-2.0.la libplist.la
> sudo ln -s libplist-2.0.a libplist.a
> sudo ln -s libplist-2.0.dylib libplist.dylib
> 
> sudo ln -s libplist++-2.0.la libplist++.la # maybe not necessary 
> sudo ln -s libplist++-2.0.a libplist++.a  # maybe not necessary 
> sudo ln -s libplist++-2.0.dylib libplist++.dylib # maybe not necessary 
> 
> cd pkgconfig
> 
> sudo ln -s libirecovery-1.0.pc libirecovery-0.2.pc # maybe not necessary
> sudo ln -s libirecovery-1.0.pc libirecovery-0.2.0.pc # maybe not necessary
> sudo ln -s libirecovery-1.0.pc libirecovery.pc
> 
> sudo ln -s libplist-2.0.pc libplist.pc
> sudo ln -s libplist++-2.0.pc libplist++.pc # maybe not necessary

Explanation

It is looking for a package named libirecovery but the installed one on the system is libirecovery-1.0, so it is saying no package found with that name. Ok, creating a symlink so libirecovery is an alias for libirecovery-1.0 on the system, fixes that issue. Simple enough right? Well I wish the internet would Have let me know about that 6 hours ago. The same thing goes for libplist, which must be installed the latest version from source and not from homebrew as the formula on homebrew is outdated. That is all, k thx, bye.

kristenlc, 2022-09-09:
If you get this error when running `make` after the above steps

```
/usr/local/include/libgeneral/macros.h:55:13:` error: expected "FILENAME" or
      <FILENAME>
#   include CUSTOM_LOGGING
```

You can fix by 

> git clone --recursive https://github.com/tihmstar/libgeneral.git
> cd libgeneral
> git checkout 636786a13dd16dc75bbfd1c20557095e9a1c5bce
> ./autogen.sh
> sudo rm -rf /usr/local/lib/libgeneral*
> sudo rm -rf /usr/local/lib/pkgconfig/libgeneral*
> sudo rm -rf /usr/local/include/libgeneral*
> make clean
> make
> sudo make install
> ls /usr/local/lib/libgeneral*

But there is nothing you can do to fix the ____chkstk_darwin symbol(s) not found you get on high sierra 10.13
```
  "____chkstk_darwin", referenced from:
      _decompress_lzss in libxpwn.a(lzss.c.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [futurerestore] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```