libimobiledevice/idevicerestore #143: idevicerestore can't restore more than 10 device in ubuntu

walkman4321, 2016-12-27:
Hi,
I try to restore 14 device simultaneously using idevicerestore but some of them always failed restore 
If I  try to restore 10 device simultaneously using idevicerestore.All device restored successfully.
I can't understand what is cause of problem.Is there any limit related to device connected in ubuntu 14.04 OS or problem with idevicerestore

jackczhang, 2016-12-28:
I also get this issue in windows using USB 3.0, USB 2.0 is fine.

nikias, 2016-12-28:
It seems to be related to the USB bus being overloaded. All the packets have to go through the bus so there's always something that can go wrong. There is no proper solution except putting the devices in a queue and restore the failing ones until they succeed. If you have 10 devices working for you, don't use more at a time. Restore the remaining 4 afterwards.