libimobiledevice/idevicerestore #146: Unable to compile on macOS 10.12.3

ztytian, 2017-01-25:
When I tried to compile with make it threw me a bunch of errors and I has no idea how to get around it. Here're the errors.

$make
/Library/Developer/CommandLineTools/usr/bin/make  all-recursive
Making all in src
  CC       idevicerestore-idevicerestore.o
idevicerestore.c:716:17: warning: 'tempnam' is deprecated: This function is
      provided for compatibility reasons only. Due to security concerns inherent
      in the design of tempnam(3), it is highly recommended that you use
      mkstemp(3) instead. [-Wdeprecated-declarations]
                        filesystem = tempnam(NULL, "ipsw_");
                                     ^
/usr/include/stdio.h:397:7: note: 'tempnam' has been explicitly marked
      deprecated here
char    *tempnam(const char *__dir, const char *__prefix) __DARWIN_ALIAS...
         ^
1 warning generated.
  CC       idevicerestore-common.o
  CC       idevicerestore-tss.o
  CC       idevicerestore-fls.o
  CC       idevicerestore-mbn.o
  CC       idevicerestore-img3.o
  CC       idevicerestore-img4.o
img4.c:97:2: warning: implicit declaration of function 'info' is invalid in C99
      [-Wimplicit-function-declaration]
        info("Personalizing IMG4 component %s...\n", component_name);
        ^
img4.c:131:3: warning: implicit declaration of function 'error' is invalid in
      C99 [-Wimplicit-function-declaration]
                error("ERROR: out of memory when personalizing IMG4 comp...
                ^
2 warnings generated.
  CC       idevicerestore-ipsw.o
  CC       idevicerestore-normal.o
  CC       idevicerestore-dfu.o
dfu.c:236:48: warning: implicit declaration of function 'irecv_get_device_info'
      is invalid in C99 [-Wimplicit-function-declaration]
        const struct irecv_device_info *device_info = irecv_get_device_i...
                                                      ^
dfu.c:236:34: warning: incompatible integer to pointer conversion initializing
      'const struct irecv_device_info *' with an expression of type 'int'
      [-Wint-conversion]
  ...*device_info = irecv_get_device_info(client->dfu->client);
      ^             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
dfu.c:241:21: error: incomplete definition of type 'struct irecv_device_info'
        *cpid = device_info->cpid;
                ~~~~~~~~~~~^
dfu.c:236:15: note: forward declaration of 'struct irecv_device_info'
        const struct irecv_device_info *device_info = irecv_get_device_i...
                     ^
dfu.c:253:34: warning: incompatible integer to pointer conversion initializing
      'const struct irecv_device_info *' with an expression of type 'int'
      [-Wint-conversion]
  ...*device_info = irecv_get_device_info(client->dfu->client);
      ^             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
dfu.c:258:21: error: incomplete definition of type 'struct irecv_device_info'
        *ecid = device_info->ecid;
                ~~~~~~~~~~~^
dfu.c:253:15: note: forward declaration of 'struct irecv_device_info'
        const struct irecv_device_info *device_info = irecv_get_device_i...
                     ^
dfu.c:271:34: warning: incompatible integer to pointer conversion initializing
      'const struct irecv_device_info *' with an expression of type 'int'
      [-Wint-conversion]
  ...*device_info = irecv_get_device_info(client->dfu->client);
      ^             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
dfu.c:276:21: error: incomplete definition of type 'struct irecv_device_info'
        return (device_info->ibfl & IBOOT_FLAG_IMAGE4_AWARE);
                ~~~~~~~~~~~^
dfu.c:271:15: note: forward declaration of 'struct irecv_device_info'
        const struct irecv_device_info *device_info = irecv_get_device_i...
                     ^
dfu.c:286:34: warning: incompatible integer to pointer conversion initializing
      'const struct irecv_device_info *' with an expression of type 'int'
      [-Wint-conversion]
  ...*device_info = irecv_get_device_info(client->dfu->client);
      ^             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
dfu.c:291:17: error: incomplete definition of type 'struct irecv_device_info'
        if (device_info->ap_nonce && device_info->ap_nonce_size > 0) {
            ~~~~~~~~~~~^
dfu.c:286:15: note: forward declaration of 'struct irecv_device_info'
        const struct irecv_device_info *device_info = irecv_get_device_i...
                     ^
dfu.c:291:42: error: incomplete definition of type 'struct irecv_device_info'
        if (device_info->ap_nonce && device_info->ap_nonce_size > 0) {
                                     ~~~~~~~~~~~^
dfu.c:286:15: note: forward declaration of 'struct irecv_device_info'
        const struct irecv_device_info *device_info = irecv_get_device_i...
                     ^
dfu.c:292:46: error: incomplete definition of type 'struct irecv_device_info'
                *nonce = (unsigned char*)malloc(device_info->ap_nonce_size);
                                                ~~~~~~~~~~~^
dfu.c:286:15: note: forward declaration of 'struct irecv_device_info'
        const struct irecv_device_info *device_info = irecv_get_device_i...
                     ^
dfu.c:296:28: error: incomplete definition of type 'struct irecv_device_info'
                *nonce_size = device_info->ap_nonce_size;
                              ~~~~~~~~~~~^
dfu.c:286:15: note: forward declaration of 'struct irecv_device_info'
        const struct irecv_device_info *device_info = irecv_get_device_i...
                     ^
dfu.c:297:29: error: incomplete definition of type 'struct irecv_device_info'
                memcpy(*nonce, device_info->ap_nonce, *nonce_size);
                               ~~~~~~~~~~~^
/usr/include/secure/_string.h:65:33: note: expanded from macro 'memcpy'
  __builtin___memcpy_chk (dest, src, len, __darwin_obsz0 (dest))
                                ^~~
dfu.c:286:15: note: forward declaration of 'struct irecv_device_info'
        const struct irecv_device_info *device_info = irecv_get_device_i...
                     ^
dfu.c:310:34: warning: incompatible integer to pointer conversion initializing
      'const struct irecv_device_info *' with an expression of type 'int'
      [-Wint-conversion]
  ...*device_info = irecv_get_device_info(client->dfu->client);
      ^             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
dfu.c:315:17: error: incomplete definition of type 'struct irecv_device_info'
        if (device_info->sep_nonce && device_info->sep_nonce_size > 0) {
            ~~~~~~~~~~~^
dfu.c:310:15: note: forward declaration of 'struct irecv_device_info'
        const struct irecv_device_info *device_info = irecv_get_device_i...
                     ^
dfu.c:315:43: error: incomplete definition of type 'struct irecv_device_info'
        if (device_info->sep_nonce && device_info->sep_nonce_size > 0) {
                                      ~~~~~~~~~~~^
dfu.c:310:15: note: forward declaration of 'struct irecv_device_info'
        const struct irecv_device_info *device_info = irecv_get_device_i...
                     ^
dfu.c:316:46: error: incomplete definition of type 'struct irecv_device_info'
                *nonce = (unsigned char*)malloc(device_info->sep_nonce_size);
                                                ~~~~~~~~~~~^
dfu.c:310:15: note: forward declaration of 'struct irecv_device_info'
        const struct irecv_device_info *device_info = irecv_get_device_i...
                     ^
dfu.c:320:28: error: incomplete definition of type 'struct irecv_device_info'
                *nonce_size = device_info->sep_nonce_size;
                              ~~~~~~~~~~~^
dfu.c:310:15: note: forward declaration of 'struct irecv_device_info'
        const struct irecv_device_info *device_info = irecv_get_device_i...
                     ^
dfu.c:321:29: error: incomplete definition of type 'struct irecv_device_info'
                memcpy(*nonce, device_info->sep_nonce, *nonce_size);
                               ~~~~~~~~~~~^
/usr/include/secure/_string.h:65:33: note: expanded from macro 'memcpy'
  __builtin___memcpy_chk (dest, src, len, __darwin_obsz0 (dest))
                                ^~~
dfu.c:310:15: note: forward declaration of 'struct irecv_device_info'
        const struct irecv_device_info *device_info = irecv_get_device_i...
                     ^
6 warnings and 13 errors generated.
make[2]: *** [idevicerestore-dfu.o] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

Does anyone know how to solve it?

jacky0124, 2017-02-01:
same problem with me

ztytian, 2017-04-10:
Whoever can compile successfully please send a executable... Pls

braden-wright93, 2017-06-14:
I mean, do you have libimobiledevice, libplist, and libusbmuxd already installed?
You can get some of those off Homebrew, I think.

nikias, 2017-06-15:
You need libirecovery properly installed and discoverable by pkg-config (and all other dependencies too). This is not a bug.

closed 2017-06-15 by nikias