libimobiledevice/idevicerestore #148: idevicerestore iOS 10.3

JoseCarlosGarcia95, 2017-01-30:
I tried to restore to iOS 10.3 using idevicerestore, and I can't maybe this issues happen because the new format (APFS).

pd: Maybe my idevice is waiting some flag for changing the filesystem format?

pd1: iOS 10.3 < is working fine!

nikias, 2017-01-30:
Any output from idevicerestore would be helpful.

JoseCarlosGarcia95, 2017-01-30:
Sure, here you've got!

About to send NORData...
Found firmware path Firmware/all_flash
Getting firmware manifest Firmware/all_flash/manifest
ERROR: zip_name_locate: Firmware/all_flash/manifest
ERROR: Unable to extract firmware manifest from ipsw
ERROR: Unable to send NOR data
ERROR: Unable to successfully restore device
ERROR: Unable to restore device

JoseCarlosGarcia95, 2017-01-30:
In the other hand, when I try to restore my device (without entering in dfu mode, and my device is in iOS 10.3) syslog show the following error:

[19:14:23.0814-GMT]{4>8} CHECKPOINT NOTICE: Image4 device: AP nonce clearable
entering ramrod_clear_ap_nonce
[19:14:23.0827-GMT]{4>8} CHECKPOINT NOTICE: AP nonce consumed
No IOFlashController instance found
[19:14:23.0829-GMT]{4>8} CHECKPOINT NOTICE: Pre-existing NVRAM variable: auto-boot=false
[19:14:23.0829-GMT]{4>8} CHECKPOINT NOTICE: Pre-existing NVRAM variable: restore-outcome=initial_monitor_no_return
[19:14:23.0830-GMT]{4>8} CHECKPOINT ANOMALY: [check_collection]restore-step-monitor(does_not_exist)
[19:14:23.0831-GMT]{4>8} CHECKPOINT PROGRESS: START (unknown) -> (initial_engine_no_return)
[19:14:23.0831-GMT]{4>8} CHECKPOINT NOTICE: NVRAM access available on initial check
restore-anomalies = {0x00000000:[check_collection]restore-step-monitor(does_not_exist)}
restore-outcome = initial_engine_no_return
executing /usr/sbin/nvram restore-outcome=initial_engine_no_return
[19:14:23.0847-GMT]{4>8} CHECKPOINT BEGIN: MAIN:[0x0400] umask
restore-step-ids = {0x11030400:1}
restore-step-names = {0x11030400:umask}
restore-step-uptime = 0
restore-step-user-progress = -1
[19:14:23.0849-GMT]{4>8} CHECKPOINT END: MAIN:[0x0400] umask
restore-step-ids = {}
restore-step-names = {}
restore-step-uptime = 0
restore-step-user-progress = -1
[19:14:23.0850-GMT]{4>8} CHECKPOINT BEGIN: MAIN:[0x0402] setvbuf
restore-step-ids = {0x11030402:2}
restore-step-names = {0x11030402:setvbuf}
restore-step-uptime = 0
restore-step-user-progress = -1
[19:14:23.0852-GMT]{4>8} CHECKPOINT END: MAIN:[0x0402] setvbuf
restore-step-ids = {}
restore-step-names = {}
restore-step-uptime = 0
restore-step-user-progress = -1
[19:14:23.0854-GMT]{4>8} CHECKPOINT BEGIN: MAIN:[0x0403] kernel_logger_thread
restore-step-ids = {0x11030403:3}
restore-step-names = {0x11030403:kernel_logger_thread}
restore-step-uptime = 0
restore-step-user-progress = -1
[19:14:23.0855-GMT]{4>8} CHECKPOINT END: MAIN:[0x0403] kernel_logger_thread
unable to open /dev/klog: No such file or directory
restore-step-ids = {}
restore-step-names = {}
restore-step-uptime = 0
restore-step-user-progress = -1
[19:14:23.0857-GMT]{4>8} CHECKPOINT BEGIN: MAIN:[0x0406] set_progress_0
restore-step-ids = {0x11030406:4}
restore-step-names = {0x11030406:set_progress_0}
restore-step-uptime = 0
restore-step-user-progress = -1
display-scale = 3
display-rotation = 0
found applelogo at /usr/share/progressui/applelogo@3x.tga
found display: primary
display: 1080 x 1920
powering on display
ramrod_display_set_granular_progress: 0.000000
[19:14:23.0908-GMT]{4>8} CHECKPOINT END: MAIN:[0x0406] set_progress_0
restore-step-ids = {}
restore-step-names = {}
restore-step-uptime = 0
restore-step-user-progress = 0
[19:14:23.0909-GMT]{4>8} CHECKPOINT BEGIN: MAIN:[0x0407] start_gasgauge_thread
restore-step-ids = {0x11030407:5}
restore-step-names = {0x11030407:start_gasgauge_thread}
restore-step-uptime = 0
restore-step-user-progress = 0
[19:14:23.0912-GMT]{4>8} CHECKPOINT END: MAIN:[0x0407] start_gasgauge_thread
restore-step-ids = {}
restore-step-names = {}
[... 794 lines cut ...]
ASR: Could not repartition target device
ASR STATUS: fail
[19:15:31.0238-GMT]{4>8} CHECKPOINT FAILURE:(FAILURE:14) RESTORED:[0x0624] restore_images [0]D(failed to restore image)
restore-step-results = {0x11070624:{0:14}}
restore-step-codes = {0x11070624:{0:14}}
restore-step-domains = {0x11070624:{0:"AMRestoreErrorDomain"}}
restore-step-error = {0x11070624:"[0]D(failed to restore image)"}
restore-step-uptime = 68
restore-step-user-progress = 1
[19:15:31.0241-GMT]{4>8} CHECKPOINT NOTICE: (NVRAM set) restore-step-user-progress=1 [sync=true] (first failure)
[19:15:31.0241-GMT]{4>8} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x11030624:50;0x11030647:51}
restore-step-names = {0x11030624:restore_images;0x11030647:cleanup_check_result}
restore-step-uptime = 68
restore-step-user-progress = 1
[19:15:31.0243-GMT]{4>8} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x11030624:50}
restore-step-names = {0x11030624:restore_images}
restore-step-uptime = 68
restore-step-user-progress = 1
[19:15:31.0244-GMT]{4>8} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x11030624:50;0x11030648:52}
restore-step-names = {0x11030624:restore_images;0x11030648:cleanup_send_final_status}
restore-step-uptime = 68
restore-step-user-progress = 1

nikias, 2017-01-30:
Maybe the ipsw structure changed?
```
ERROR: zip_name_locate: Firmware/all_flash/manifest
```

JoseCarlosGarcia95, 2017-01-30:
Yes, apple has removed manifest.
I think that manifest could be generated using some device variables, for example, iOS 10.2.1 manifest:

LLB.n66.RELEASE.im4p
iBoot.n66.RELEASE.im4p
DeviceTree.n66ap.im4p
applelogo@3x~iphone.s8000.im4p
recoverymode@1920~iphone-lightning.s8000.im4p
batterylow0@3x~iphone.s8000.im4p
batterylow1@3x~iphone.s8000.im4p
batterycharging0@3x~iphone.s8000.im4p
batterycharging1@3x~iphone.s8000.im4p
glyphplugin@1920~iphone-lightning.s8000.im4p
batteryfull@3x~iphone.s8000.im4p
sep-firmware.n66.RELEASE.im4p

**Some variables means: **
s8000: is the processor class https://www.theiphonewiki.com/wiki/S8000
n66: iPhone 6s class
1920: Screen resolution for Plus mode

I have created a zip with the new structure (with *.dmg deleted): 
https://mega.nz/#!28RTUBZS!BH_Kw1pxjbL0uJ2MK4zLVbj_Trw0YpRokLLrho8EYzc

PD: If I've got time this week I will push an update with the fix.

nikias, 2017-02-01:
Can you try with this patch?
https://gist.github.com/nikias/8a9f4d7937d71bd2d375944d096cd781

nikias, 2017-02-01:
Also, if you can, try to restore with a pre-10.3 iOS version. Basically the manifest was removed because all the info has been in the BuildManifest.plist for a long time now.

JoseCarlosGarcia95, 2017-02-01:
Sure, I'll try this weekend

JoseCarlosGarcia95, 2017-02-02:
Stuck on Waiting for device now.
--
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (12543000 bytes)...
About to restore device... 
Waiting for device...

JoseCarlosGarcia95, 2017-02-02:
And working for iOS 10.2.1

JoseCarlosGarcia95, 2017-02-02:
I have tested a dirty trick and now is working:
- I have added the manifest to all_flash folder (just for testing) for my iPhone 6 Plus and it's working fine, so it's only a problem with the manifest file :-)

PD: I'm getting this error: ERROR: FDR 0x7f4d44002dd0 Unable to receive proxy payload (-2)
but it's look like that my iPhone has been restored successfully

nikias, 2017-02-02:
Can you test my patch and try without a modified ipsw? idevicerestore is supposed to work without any manual dirty tricks ;)

JoseCarlosGarcia95, 2017-02-02:
I have tried your patch and I'm getting stuck at "Waiting for device...."

nikias, 2017-02-02:
The changes of my patch are in a part that is way after that part, in fact after uploading the root filesystem. This must be something else.

JoseCarlosGarcia95, 2017-02-06:
I have found the bug, the sep firmware has never been sent because they're using 'IsSecondaryFirmwarePayload' and no 'IsFirmwarePayload'
In the other hand, Apple has deleted glyphplugin from manifest?

JoseCarlosGarcia95, 2017-02-11:
Confirmed that your patch is working, my wire was broken.

OothecaPickleGNUrmsTUXFSF, 2017-02-21:
Patch works fine for me as well. Thanks @nikias!!!!!

nikias, 2017-02-21:
See commit aaec66651aba5f617c5c2c22b019d115e2db8e8f.

closed 2017-02-21 by nikias

Reus2Tech, 2017-02-23:
@nikias @JoseCarlosGarcia95 I have iphone 6 on 10.3 beta 1 how can I do it?
plz help to restore to 10.2

Reus2Tech, 2017-02-23:
is there any tutorial 
if there gave me the link

Reus2Tech, 2017-02-23:
@OothecaPickleGNUrmsTUXFSF How you did it 
Can you help me to restore to 10.2 
I'm now on 10.3 beta 1

JoseCarlosGarcia95, 2017-02-23:
Hey!
Please don't spam this!
When I tried to restore to iOS 10.2.1 from iOS 10.3 I had some troubles because apfs.
How I solved this? I put my device in DFU mode and then I could restore without problems.

PD: I don't think that this is a bug.

Reus2Tech, 2017-03-02:
@JoseCarlosGarcia95 There is tutorial for that plz any thing

JoseCarlosGarcia95, 2017-03-02:
Take a look here: https://www.theiphonewiki.com/wiki/DFU_Mode

Reus2Tech, 2017-03-03:
@JoseCarlosGarcia95 Thank you Bro 
But I want tutorial for use libimobiledevice
I'm sorry for noise you

nikias, 2017-03-03:
The problem is you're causing noise for EVERYONE on this ticket (and the maintainers in any case) whenever you comment on something. This is a bug tracker and not a support forum. Please only report bugs here. Thanks. And please don't reply to this message.

nikias, 2017-04-03:
@JoseCarlosGarcia95 @OothecaPickleGNUrmsTUXFSF have you by any chance tried this on a newer 10.3 or 10.3.1 restore?

OothecaPickleGNUrmsTUXFSF, 2017-04-04:
@nikias what do you mean?

nikias, 2017-04-04:
@OothecaPickleGNUrmsTUXFSF have you tried to restore a device with the recent iOS 10.3 or 10.3.1 release?

OothecaPickleGNUrmsTUXFSF, 2017-04-04:
I'll do it right now and get back to you.

OothecaPickleGNUrmsTUXFSF, 2017-04-04:
@nikias [Successfully restored my iPhone5,3 to 10.3.1 and 10.3 with commit 8a75d31](https://ghostbin.com/paste/jonag).

OothecaPickleGNUrmsTUXFSF, 2017-04-04:
@nikias It seems that @ankitrwt attempted to restore a 64-bit device. Would you like me to try restoring my iPhone 5S?

nikias, 2017-04-04:
@OothecaPickleGNUrmsTUXFSF oh good observation. Maybe it's about the FDR part of the restore...
Only restore if you can sacrifice it, I don't want you to lose a specific firmware version just because of my curiosity :)

OothecaPickleGNUrmsTUXFSF, 2017-04-04:
@nikias No, it's okay; I actually have two jailbroken 5S's, so I can afford to lose one. :) Do you want me to try restoring?

nikias, 2017-04-04:
@OothecaPickleGNUrmsTUXFSF you don't have to.

ankitrwt, 2017-04-04:
hello @nikias @OothecaPickleGNUrmsTUXFSF 
i tried again but stil the results are 
for iPhone 6

root:Firmware_CacheDir# idevicerestore -e -l -C .
NOTE: using cached version data
Found device in Normal mode
Identified device as n53ap, iPhone6,2
Latest firmware is iPhone_4.0_64bit_10.3_14E277_Restore.ipsw
Verifying './iPhone_4.0_64bit_10.3_14E277_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 10.3
Product Build: 14E277 Major: 14
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Found ECID 2706612266844
Getting ApNonce in normal mode... 42 15 d8 b0 f3 18 90 fd f1 2b 42 c3 0f 12 f8 35 50 c3 d3 60 
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... 2e 59 0d 19 a1 a8 54 52 b5 20 74 f5 aa 5f 03 f1 7c e3 c5 dc 
WARNING: Unable to find BbSkeyId node
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Using cached filesystem from './iPhone_4.0_64bit_10.3_14E277_Restore/058-67581-067.dmg'
Entering recovery mode...
INFO: device serial number is DX4RTH7JFRC5
Extracting iBEC.iphone6.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (648345 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-3406.50.244
iBoot build-style=RELEASE
Sending AppleLogo...
Extracting applelogo@2x~iphone.s5l8960x.im4p...
Personalizing IMG4 component AppleLogo...
Sending AppleLogo (11640 bytes)...
ramdisk-size=0x10000000
Extracting 058-55512-165.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (41575446 bytes)...
Extracting DeviceTree.n53ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (107921 bytes)...
Extracting kernelcache.release.iphone6...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (12168008 bytes)...
About to restore device... 
Waiting for device...
Device b43439f38a77718f40dc28bea621309b9cca1353 is now connected in restore mode...
Connecting now...
Connected to com.apple.mobile.restored, version 14
Device b43439f38a77718f40dc28bea621309b9cca1353 has successfully entered restore mode
Hardware Information:
BoardID: 2
ChipID: 35168
UniqueChipID: 2706612266844
ProductionMode: true
Starting FDR listener thread
About to send NORData...
Found firmware path Firmware/all_flash
ERROR: zip_name_locate: Firmware/all_flash/manifest
Getting firmware manifest from build identity
Extracting LLB.iphone6.RELEASE.im4p...
Personalizing IMG4 component LLB...
Extracting applelogo@2x~iphone.s5l8960x.im4p...
Personalizing IMG4 component AppleLogo...
Extracting batterycharging0@2x~iphone.s5l8960x.im4p...
Personalizing IMG4 component BatteryCharging0...
[... 1711 lines cut ...]
Sending TSS request attempt 1... response successfully received
Received Baseband SHSH blobs
Sending BasebandData now...
Done sending BasebandData
Updating Baseband in progress...
About to send BasebandData...
WARNING: Unable to find BbSkeyId node
Sending Baseband TSS request...
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received Baseband SHSH blobs
Sending BasebandData now...
Done sending BasebandData
Updating Baseband completed.
Updating SE Firmware (59)
Fixing up /var (17)
Creating system key bag (50)
Modifying persistent boot-args (25)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Got status message
Status: Restore Finished
Cleaning up...
DONE

ankitrwt, 2017-04-04:
dmesg 

[ 2093.630107] usb 1-9: new high-speed USB device number 12 using xhci_hcd
[ 2093.758793] usb 1-9: New USB device found, idVendor=05ac, idProduct=1227
[ 2093.758800] usb 1-9: New USB device strings: Mfr=2, Product=3, SerialNumber=4
[ 2093.758805] usb 1-9: Product: Apple Mobile Device (DFU Mode)
[ 2093.758809] usb 1-9: Manufacturer: Apple Inc.
[ 2093.758813] usb 1-9: SerialNumber: CPID:7000 CPRV:11 CPFM:03 SCEP:01 BDID:06 ECID:0006690810798C26 IBFL:1D

i restored iPhone 5s and iPhone 6s back to ios 10.2.1 they restored successful, 
i dont know where i am doing wrong with ios 10.3

any help appreciated 
Thank you

OothecaPickleGNUrmsTUXFSF, 2017-04-04:
@ankitrwt I'll look into this later tonight.

nikias, 2017-04-05:
@OothecaPickleGNUrmsTUXFSF did you have a chance to try? I just got my baby yesterday so the last few days I was mostly in the hospital ;)

OothecaPickleGNUrmsTUXFSF, 2017-04-05:
@nikias no, not yet, but i'll definitely do it today. :) also, congratulations on your baby. :D

ankitrwt, 2017-04-06:
congratulations @nikias

vipinbeni, 2017-04-06:
congratulations @nikias

vipinbeni, 2017-04-06:
i also facing same issue iphone goes in dfu mode after restore 10.3.1.

mallon735, 2017-04-07:
I have found a bug in restore.c file. Free'd value is used later in the code.

restore.c:1016 filename = strrchr(comppath, '/');
restore.c:1040 free(comppath);
restore.c:1053 if (!strncmp("iBoot", filename, 4))

I will make more tests to see if this will fix the DFU issue. So far the same iPhone6 phone (re)flashing results were unpredictable.

nikias, 2017-04-07:
@mallon735 good catch, however my experience tells me that if this is the issue here then it would have caused the same for earlier iOS versions too.

mallon735, 2017-04-07:
I know that this code was added on 27.Feb.2017 aaec66 -  restore: Add support for 10.3+ firmware images (ipsw) 
Also I noticed, that this DFU effect was occurring randomly on S5 and 6 models. 6S and 7 upgrade was working always.

mallon735, 2017-04-07:
Patch file
[idevicerestore.txt](https://github.com/libimobiledevice/idevicerestore/files/905615/idevicerestore.txt)

nikias, 2017-04-07:
Oh heh. Already committed, which turns out to be the same, but I also fixed the strncmp string length :)

nikias, 2017-04-07:
Another thing I realized is that @OothecaPickleGNUrmsTUXFSF reported that restoring 32-bit devices worked, and the code in question is used for all devices :)

mallon735, 2017-04-07:
Firstly congratulations on your baby @nikias 
This iOS 10.3.1 DFU issue is solved. I have made 10+ successful restores without seeing DFU mode.

nikias, 2017-04-07:
@mallon735 what devices for reference?

mallon735, 2017-04-07:
iPhone 5S and 6.
I suggest that if iBoot is not the first entry in the array, phones will always fall into DFU. I have not tested this, but the filename value was freed before this section:
```
/* make sure iBoot is the first entry in the array */
if (!strncmp("iBoot", filename, 5)) {
  plist_array_insert_item(norimage_array, plist_new_data((char*)nor_data, (uint64_t)nor_size), 0);
} else {
  plist_array_append_item(norimage_array, plist_new_data((char*)nor_data, (uint64_t)nor_size));
}
```

nikias, 2017-04-07:
@mallon735 yeah I guess you are right. Depending on the operating system and the compiler (and maybe other things) this might indeed cause the restore to fail. So my latest commit should finally fix that :)

mathius1714, 2017-05-23:
Hello, I'm trying to instal a Custom IPSW with the "setup.app" patch to do an Icloud Bypass. I got may times "Error: 14" on iTunes, so now I'm trying to install with libimobiledevices, this is my error:

C:\Libi>idevicerestore.exe -u --2c3ac7bbcf63d0ca62120867cbef51d0c427c231 -c iPhone_4.7_10.3.1_14E304_Restore.ipsw NOTE: Updated version data. Found device in Recovery mode Identified device as n61ap, iPhone7,2 Extracting Restore.plist from IPSW Product Version: 10.3.1 Product Build: 14E304 Major: 14 INFO: device serial number is C39PP3X1G5MN Device supports Image4: true Custom firmware requested. Disabled TSS request. ERROR: zip_name_locate: Firmware/all_flash/all_flash.n61ap.production/manifest ERROR: could not extract Firmware/all_flash/all_flash.n61ap.production/manifest from IPSW

I'm noob programming and I'm using Windows. I need help!

Thank you so much!

mathius1714, 2017-05-24:
@JoseCarlosGarcia95 Hola, podrías ayudarme con el mensaje que he publicado arriba? Muchas gracias!

OothecaPickle, 2017-05-24:
@mathius1714 https://github.com/libimobiledevice/idevicerestore/issues/167#issuecomment-303704000

mathius1714, 2017-05-24:
@OothecaPickle Are you telling that it's impossible to do an Icloud Bypass on ios 10.3.1? I only need to know this, nobody wants to waste time. Thanks

OothecaPickle, 2017-05-24:
@mathius1714 yes, as of now that is correct

mathius1714, 2017-05-24:
@OothecaPickle So, will be possible to do? And now isn't posible because the new APFS? Thanks

nikias, 2017-05-24:
@OothecaPickle stop asking these questions. There will never be a software-modification iCloud bypass. DEAL WITH IT. The device can't be activated with Apple's servers until Apple unblocks it (which I never heard of they actually do).

[6 more replies]