libimobiledevice/idevicerestore #149: Restore failing after sending RestoreKernelCache

JWhy, 2017-02-08:
Hello,

I'm trying to restore a friend's iPhone 6. When booting it regularly it shows the Apple logo and after some second the display fades and it reboots. When using iTunes on Windows to restore it from restore mode (which appears sometimes) it will disconnect and reboot at some point, sometimes before the IPSW extraction finished.

When using idevicerecovery it does behave similiarly, both in restore and DFU mode. In DFU it even proceeds to the RestoreKernelCache. It reboots then and shows the restore mode screen again.

This is the debug log of idevicerestore:
http://pastebin.com/EyB6bLZX
I blacked half of the IDs looking sensitive out with X's.

I used the most recent official IPSW for the device.

Do you have any clue what is causing the problem?
Thank you in advance!

JoseCarlosGarcia95, 2017-02-11:
Could you try adding --erase at the end of your command? i.e idevicerestore -d /media/windata/iPhone_4.7_10.2.1_14D27_Restore.ipsw --restore

nikias, 2017-02-11:
According to the log it never enters restore mode. When you start the operation the recovery screen should go black, then it should show the Apple logo, then spinner should appear below the apple, and then a progress bar should appear. Are you able to see these four states?

nikias, 2017-06-15:
Closing due to inactivity. If restoring with iTunes also fails idevicerestore won't help either.

closed 2017-06-15 by nikias