libimobiledevice/idevicerestore #153: Unable to recieve message from FDR

OviOneKenoby, 2017-03-24:
Hello,

First post here.
So i`m trying to restore an iPhone 6 Plus.
Itunes returns error 9, 3utools just stops at sending NORData.
So i thought i`d give this library a shot. And because it`s so detailed, i`m going to use this one from now on (really like it).
So to get to the point, i get the following after idevicerestore -e -L (using windows).

........
About to restore device...
Waiting for device...
Device e4c807cfc760ce460641fe4a28c7c7c40f74c440 is now connected in restore mode...
Connecting now...
Connected to com.apple.mobile.restored, version 14
Device e4c807cfc760ce460641fe4a28c7c7c40f74c440 has successfully entered restore mode
Hardware Information:
BoardID: 4
ChipID: 28672
UniqueChipID: 2917220043251759
ProductionMode: true
Previous restore exit status: 0x100
Starting FDR listener thread
About to send NORData...
Found firmware path Firmware/all_flash/all_flash.n56ap.production
Getting firmware manifest Firmware/all_flash/all_flash.n56ap.production/manifest
Extracting LLB.n56.RELEASE.im4p...
Personalizing IMG4 component LLB...
Extracting iBoot.n56.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Extracting DeviceTree.n56ap.im4p...
Personalizing IMG4 component DeviceTree...
Extracting applelogo@3x~iphone.t7000.im4p...
Personalizing IMG4 component AppleLogo...
Extracting recoverymode@1920~iphone-lightning.t7000.im4p...
Personalizing IMG4 component RecoveryMode...
Extracting batterylow0@3x~iphone.t7000.im4p...
Personalizing IMG4 component BatteryLow0...
Extracting batterylow1@3x~iphone.t7000.im4p...
Personalizing IMG4 component BatteryLow1...
Extracting batterycharging0@3x~iphone.t7000.im4p...
Personalizing IMG4 component BatteryCharging0...
Extracting batterycharging1@3x~iphone.t7000.im4p...
Personalizing IMG4 component BatteryCharging1...
Extracting glyphplugin@1920~iphone-lightning.t7000.im4p...
Personalizing IMG4 component BatteryPlugin...
Extracting batteryfull@3x~iphone.t7000.im4p...
Personalizing IMG4 component BatteryFull...
Extracting sep-firmware.n56.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Extracting sep-firmware.n56.RELEASE.im4p...
Personalizing IMG4 component SEP...
Sending NORData now...
Done sending NORData
ERROR: Unable to receive message from FDR 03D1DF58 (-2). 0/2 bytes

Have to mention that at the sending NORData, it takes about 30 seconds (no idea how much time it should take).
Is this a HW problem? 
Any help would be appreciated.

Thank you!

braden-wright93, 2017-06-13:
Just wanted to bump this as I'm seeing the exact same issue on OSX with the same device - an iPhone 6 Plus. iTunes returned Error 9, so I've tried 3 different cables, etc. then turned to libimobiledevice.

Updated to the latest commit (b99782b) and still no luck.

Gonna look through the code and see if I can find anything out...

braden-wright93, 2017-06-15:
Ok, well still no luck.
I rebuilt all the software here with the latest commits (since I *was* using the libplist and usbmuxd from Homebrew) and still no luck.

It seems like:
`libimobiledevice/src/idevice.c:390 internal_connection_recv_timeout(...)  calls:`
    `libusbmuxd/src/libusbmuxd.c:1130 usbmuxd_recv_timeout(...)  which calls:`
        `???/socket.h?  socket_recv_timeout(...)  ?`

And usbmux_recv_timeout returns -35 at some point.
Still researching...

OviOneKenoby, 2017-06-15:
Hi.
I managed to fix it by resoldering the nand chip.
And a full board reflow after.
Now it works fine.

braden-wright93, 2017-06-15:
Gotcha! Appreciate the tip. Time to bust out my soldering equipment and see if I can fix this thing 😎

nikias, 2017-06-15:
If restore doesn't work with iTunes then idevicerestore (most likely) won't be able to restore the device too.

closed 2017-06-15 by nikias