libimobiledevice/idevicerestore #154: Hi guys,

truonggiang0710, 2017-03-30:


nikias, 2017-03-30:
Is it really DFU mode, i.e. black screen? Or does it show the connect to iTunes screen, which is recovery mode? If so, can you kick the device out of recovery mode with irecovery tool?
```
irecovery -s
```
(wait for it to connect)
```
setenv auto-boot true
saveenv
reboot
```
Press ctrl+c to exit.

truonggiang0710, 2017-03-30:
Yes, It really in DFU mode. It's in black screen.

applebeast, 2017-04-06:
Ok. 2 things to do. 

First, hold the home and power buttons down at the same time for 15 seconds, then wait 5 seconds and hit the power. If this doesn't work, you need to do 1 of 3 things: open the phone and dis/reconnect thd battery, or finally just put it in the freezer until the battery's chemical reactions stop, and then take it out and turn it on.

applebeast, 2017-04-06:
restore it again.

nikias, 2017-06-15:
Does it restore with iTunes?

truonggiang0710, 2017-06-15:
It's restored OK over iTunes !

nikias, 2017-06-15:
No idea then... if you have the option to restore the device again with idevicerestore we can have a closer look. If not we can close the issue.

closed 2017-09-12 by truonggiang0710