libimobiledevice/idevicerestore #155: Fails to restore iPhone with ios 10.3

ankitrwt, 2017-03-30:
hi guys,
i tried to restore iPhone with the latest ios 10.3 but it fails to restore...

root:Firmware_CacheDir# idevicerestore -e iPhone_4.7_10.3_14E277_Restore.ipsw -C ./
NOTE: using cached version data
Found device in Recovery mode
Identified device as n61ap, iPhone7,2
Extracting BuildManifest from IPSW
Product Version: 10.3
Product Build: 14E277 Major: 14
INFO: device serial number is FFMQW43JG5MV
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Found ECID 1804333217319974
Getting ApNonce in recovery mode... 9f 19 b0 f1 9f 02 98 5f 86 53 2f d4 78 bd 16 b1 d1 15 29 53 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... c5 d7 a3 97 b0 a4 55 7b e7 c6 8c 2d ad 08 82 82 82 85 a0 3c 
WARNING: Unable to find BbSkeyId node
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting filesystem from IPSW
[==================================================] 100.0%
Extracting iBEC.n61.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (657014 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-3406.50.244
iBoot build-style=RELEASE
Sending AppleLogo...
Extracting applelogo@2x~iphone.t7000.im4p...
Personalizing IMG4 component AppleLogo...
Sending AppleLogo (12117 bytes)...
ramdisk-size=0x10000000
Extracting 058-55512-165.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (41571827 bytes)...
Extracting DeviceTree.n61ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (123986 bytes)...
Extracting kernelcache.release.iphone7...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (12591071 bytes)...
About to restore device... 
Waiting for device...
Device 80bf669ed94eab01711cde24101d07f748991871 is now connected in restore mode...
Connecting now...
Connected to com.apple.mobile.restored, version 14
Device 80bf669ed94eab01711cde24101d07f748991871 has successfully entered restore mode
Hardware Information:
BoardID: 6
ChipID: 28672
UniqueChipID: 1804333217319974
ProductionMode: true
Previous restore exit status: 0x100
Starting FDR listener thread
About to send NORData...
Found firmware path Firmware/all_flash
Getting firmware manifest Firmware/all_flash/manifest
ERROR: zip_name_locate: Firmware/all_flash/manifest
ERROR: Unable to extract firmware manifest from ipsw
ERROR: Unable to send NOR data
ERROR: Unable to successfully restore device
ERROR: Unable to restore device

any help appreciated 
Thank you

truonggiang0710, 2017-03-30:
Hi @ankitrwt ,

Please read this and update last code.
"
Starting with iOS 10.3 the ipsw files don't contain any 'manifest' file
anymore which we used to determine the files required for the reply to the
'NorData' request. Since this info is also included in the BuildManifest
we us that instead, unless a 'manifest' file is present.
https://github.com/libimobiledevice/idevicerestore/commit/aaec66651aba5f617c5c2c22b019d115e2db8e8f
"

ankitrwt, 2017-03-30:
@truonggiang0710 
Thanks for the reply,
I restored iPhone from commit no aaec666 it restored successful but after restore iPhone doesnt got reboot itself, and now its not getting power on 
i tried it on iPhone 6 and iPhone 5s

nikias, 2017-03-30:
Do you have the same issues as mentioned here: #154 ?

ankitrwt, 2017-03-30:
@nikias
Thanks for reply, 
yes same issue, I am working on Ubuntu 16.04 , its not detected on my system from any of idevice commands.. but on dmesg its show 

[16377.859339] usb 1-9: New USB device found, idVendor=05ac, idProduct=1227
[16377.859346] usb 1-9: New USB device strings: Mfr=2, Product=3, SerialNumber=4
[16377.859351] usb 1-9: Product: Apple Mobile Device (DFU Mode)
[16377.859355] usb 1-9: Manufacturer: Apple Inc.
[16377.859358] usb 1-9: SerialNumber: CPID:7000 CPRV:11 CPFM:03 SCEP:01 BDID:06 ECID:0006690810798C26 IBFL:1D

nikias, 2017-03-30:
Can you post a full restore log? Preferrably add it as an attachment here or as a gist.

ankitrwt, 2017-03-31:
iPhone 5s restore Log, this one also ends up with dfu mode

root:Firmware_CacheDir# idevicerestore -d -e iPhone_4.0_64bit_10.3_14E277_Restore.ipsw -C .
Reading data from ./version.xml
NOTE: using cached version data
Found device in Normal mode
Identified device as n53ap, iPhone6,2
Extracting BuildManifest from IPSW
Product Version: 10.3
Product Build: 14E277 Major: 14
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Getting device's ECID for TSS request
Found ECID 5455326872088
Getting ApNonce in normal mode... cd 25 90 ad 64 5f 18 a7 2a 62 44 62 23 47 17 21 d9 55 a3 07 
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... 55 bb 68 a4 29 27 39 ab 20 c5 df 04 25 2b 5c 0a 86 4a 95 ea 
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbFDRSecurityKeyHash node
WARNING: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry AppleLogo
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging0
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging1
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryFull
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow0
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow1
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryPlugin
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry DeviceTree
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry KernelCache
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry LLB
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry OS
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry RecoveryMode
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry RestoreDeviceTree
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry RestoreKernelCache
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry RestoreLogo
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry RestoreRamDisk
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
[... 1556 lines cut ...]
Updating baseband (19)
Updating baseband (19)
Updating Baseband in progress...
About to send BasebandData...
Reading data from /tmp/8be73JKtVMF
Sending BasebandData now...
Done sending BasebandData
Updating baseband (19)
Updating baseband (19)
Updating baseband (19)
Updating baseband (19)
Updating baseband (19)
Updating Baseband completed.
Fixing up /var (17)
Creating system key bag (50)
Modifying persistent boot-args (25)
Unmounting filesystems (29)
Unmounting filesystems (29)
Got status message
Status: Restore Finished
FDR 0x263ea50 timeout waiting for command
FDR 0x263ea50 terminating...
Cleaning up...
DONE
root:Firmware_CacheDir#

ankitrwt, 2017-03-31:
iPhone 5s restore results

with ios 10.2.1
[restore_10.2.1.txt](https://github.com/libimobiledevice/idevicerestore/files/884456/restore_10.2.1.txt)

with ios 10.3
[restore_10.3.txt](https://github.com/libimobiledevice/idevicerestore/files/884457/restore_10.3.txt)

ankitrwt, 2017-04-03:
Hello guys,
i restored iPhone with ios 10.3 now they are stuck into dfu mode, Is there any way to start up iPhone into normal mode or recovery mode?

nikias, 2017-04-03:
I don't know why it doesn't work but with 10.3 beta 1 the restore worked (after my fix aaec66) as reported in #148.

nikias, 2017-04-07:
This issue seems fixed with my latest commit 1f3aff. Please verify.

ankitrwt, 2017-04-10:
Thanks @nikias 
I tried on iPhone 5s and 6, the restore results are successful and after restore idevice turns on in normal mode.. (y)

nikias, 2017-04-10:
Closing, issue resolved.

closed 2017-04-10 by nikias