libimobiledevice/idevicerestore #156: How to flash without tss

applebeast, 2017-04-06:
How can I modify this program to continue instead of erroring out if tss server cannot be reached?

braden-wright93, 2017-06-14:
Create your own branch and patch the code accordingly, in theory, then merge later if it works well.
There's not an 'option' for that as it stands now.
If the device *should* accept the firmware without the signature (via stitched aptickets or whathaveyou) then you could theoretically just comment out those parts of the code - or at least that would be a starting point.

GeoSn0w, 2017-06-18:
I don't think it is that simple. I've learned the hard way the device has a lot of security checks in place. Even if you patch the tool to "flash without TSS", the device would simply refuse the payload you're sending since it won't bootstrap any iBEC / iBSS / Ramdisk to begin the restore until the TSS request has returned a correct response.