libimobiledevice/idevicerestore #165: iPhone 5S - iOS 10.3.1 - Verification Stage - failed to invert APFS image

scoynet, 2017-05-10:
Hello All,

  I have an iPhone 5S that won't boot. It's stuck in recovery mode. I used irecovery to force auto-boot but it eventually reboots back into recovery mode. I used idevicerestore to try and update to 10.3.1 but it fails at the verification stage with an error about inverting the APFS image. I tried older firmware versions but I guess Apple doesn't sign them any more and they can't be used.

  I've pasted the tail end of the output below. Everything above seemed OK. Let me know if it would be beneficial to post the entire thing.

  Any ideas? Am I doing anything wrong? Is there a way to skip the verification stage? Any thoughts or suggestions are appreciated.

Thanks!

[root@va70 ~]# idevicerestore -i 0000040CF3D162B0 iPhone_4.0_64bit_10.3.1_14E304_Restore.ipsw
....
restore_apfs_image : finished restoring an APFS image
invert_apfs_image : inverting : /System/Library/Filesystems/apfs.fs/apfs_invert /dev/disk0s1 0 apfs_invert_asr_img
executing /System/Library/Filesystems/apfs.fs/apfs_invert -d /dev/disk0s1 -s 1 -n apfs_invert_asr_img
ASR: *** Mounting outer volume (/dev/disk0s1 s1)...
ASR: nx_mount:846: Container cleanly-unmounted flag set.
ASR: spaceman_metazone_init:326: spaceman metazone of size 262144 blocks (encrypted: 3644088-3775160 unencrypted: 3775160-3906232)
ASR: nx_mount:965: 
ASR: checkpoint search: largest xid 485, best xid 485 @ 33
ASR: nx_mount:1027: sanity checking all nx state... please be patient.
ASR: nx_check:335: 
ASR: cib 0 ci 22 free count 750 doesn't match free bits in bitmap block 714
ASR: nx_check:335: cib 0 ci 24 free count 4960 doesn't match free bits in bitmap block 4919
ASR: nx_check:335: cib 0 ci 25 free count 5396 doesn't match free bits in bitmap block 5368
ASR: nx_check:335: cib 0 ci 29 free count 4533 doesn't match free bits in bitmap block 4488
ASR: nx_check:335: cib 0 ci 30 free count 4671 doesn't match free bits in bitmap block 4655
ASR: nx_check:335: cib 0 ci 31 free count 5254 doesn't match free bits in bitmap block 5247
ASR: nx_check:335: cib 0 ci 32 free count 5732 doesn't match free bits in bitmap block 5713
ASR: nx_check:335: cib 0 ci 33 free count 6121 doesn't match free bits in bitmap block 6041
ASR: nx_check:335: cib 0 ci 34 free count 4300 doesn't match free bits in bitmap block 4278
ASR: nx_check:335: cib 0 ci 35 free count 5116 doesn't match free bits in bitmap block 5061
ASR: nx_check:335: cib 0 ci 37 free count 4437 doesn't match free bits in bitmap block 4390
ASR: nx_check:335: cib 0 ci 38 free count 4773 doesn't match free bits in bitmap block 4760
ASR: nx_check:335: cib 0 ci 39 free count 5625 doesn't match free bits in bitmap block 5613
ASR: nx_check:335: cib 0 ci 40 free count 5735 doesn't match free bits in bitmap block 5728
ASR: nx_check:335: cib 0 ci 41 free count 3415 doesn't match free bits in bitmap block 3393
ASR: nx_check:335: cib 0 ci 42 free count 5766 doesn't match free bits in bitmap block 5719
ASR: nx_check:335: cib 0 ci 43 free count 5041 doesn't match free bits in bitmap block 5017
ASR: nx_check:335: cib 0 ci 44 free count 5873 doesn't match free bits 
ASR: in bitmap block 5834
ASR: nx_check:335: cib 0 ci 45 free count 5764 doesn't match free bits in bitmap block 5705
ASR: nx_check:335: cib 0 ci 46 free count 7925 doesn't match free bits in bitmap block 7913
ASR: nx_check:335: cib 0 ci 47 free count 6755 doesn't match free bits in bitmap block 6744
ASR: nx_check:335: cib 0 ci 49 free count 4836 doesn't match free bits in bitmap block 4829
ASR: nx_check:335: cib 0 ci 50 free count 5568 doesn't match free bits in bitmap block 5558
ASR: nx_check:335: cib 0 ci 51 free count 5395 doesn't match free bits in bitmap block 5321
ASR: nx_check:335: cib 0 ci 52 free count 5337 doesn't match free bits in bitmap block 5324
ASR: nx_check:335: cib 0 ci 53 free count 6362 doesn't match free bits in bitmap block 6357
ASR: nx_check:335: cib 0 ci 54 free count 5797 doesn't match free bits in bitmap block 5777
ASR: nx_check:335: cib 0 ci 55 free count 6851 doesn't match free bits in bitmap block 6825
ASR: nx_check:335: cib 0 ci 56 free count 5583 doesn't match free bits in bitmap block 5546
ASR: nx_check:335: cib 0 ci 59 free count 6407 doe
ASR: sn't match free bits in bitmap block 6398
ASR: nx_check:335: cib 0 ci 60 free count 5809 doesn't match free bits in bitmap block 5790
ASR: nx_check:335: cib 0 ci 61 free count 7621 doesn't match free bits in bitmap block 7589
ASR: nx_check:335: cib 0 ci 62 free count 8475 doesn't match free bits in bitmap block 8411
ASR: nx_check:335: cib 0 ci 65 free count 4955 doesn't match free bits in bitmap block 4883
ASR: nx_check:335: cib 0 ci 66 free count 4341 doesn't match free bits in bitmap block 4315
ASR: nx_check:335: cib 0 ci 67 free count 5138 doesn't match free bits in bitmap block 5116
ASR: nx_check:335: cib 0 ci 68 free count 4455 doesn't match free bits in bitmap block 4445
ASR: nx_check:335: cib 0 ci 72 free count 5367 doesn't match free bits in bitmap block 5349
ASR: nx_check:335: cib 0 ci 73 free count 5187 doesn't match free bits in bitmap block 5142
ASR: nx_check:335: cib 0 ci 74 free count 4856 doesn't match free bits in bitmap block 4828
ASR: nx_check:335: cib 0 ci 75 free count 4952 doesn't match free bits in bitmap block 4939
ASR: nx_check:335: cib 0 ci 7
ASR: 6 free count 2027 doesn't match free bits in bitmap block 2013
ASR: nx_check:335: cib 0 ci 78 free count 6986 doesn't match free bits in bitmap block 6945
ASR: nx_check:335: cib 0 ci 79 free count 4938 doesn't match free bits in bitmap block 4869
ASR: nx_check:335: cib 0 ci 80 free count 5997 doesn't match free bits in bitmap block 5933
[... 43 lines cut ...]
[11:09:33.0339-GMT]{4>8} CHECKPOINT FAILURE:(FAILURE:14) RESTORED:[0x064F] check_apfs_format [0]D(failed to invert APFS image)
restore-step-results = {0x1107064F:{0:14};0x11070C05:{0:14}}
restore-step-codes = {0x1107064F:{0:14};0x11070C05:{0:14}}
restore-step-domains = {0x1107064F:{0:"AMRestoreErrorDomain"};0x11070C05:{0:"AMRestoreErrorDomain"}}
restore-step-error = {0x1107064F:"[0]D(failed to invert APFS image)"}
restore-step-uptime = 204
restore-step-user-progress = 63
[11:09:33.0351-GMT]{4>8} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103064F:48;0x11030C05:53;0x11030647:54}
restore-step-names = {0x1103064F:check_apfs_format;0x11030C05:restore_invert;0x11030647:cleanup_check_result}
restore-step-uptime = 204
restore-step-user-progress = 63
[11:09:33.0359-GMT]{4>8} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103064F:48;0x11030C05:53}
restore-step-names = {0x1103064F:check_apfs_format;0x11030C05:restore_invert}
restore-step-uptime = 204
restore-step-user-progress = 63
[11:09:33.0367-GMT]{4>8} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x1103064F:48;0x11030C05:53;0x11030648:55}
restore-step-names = {0x1103064F:check_apfs_format;0x11030C05:restore_invert;0x11030648:cleanup_send_final_status}
restore-step-uptime = 204
restore-step-user-progress = 63

ERROR: Unable to successfully restore device
ERROR: Unable to restore device

scoynet, 2017-06-03:
Just tried with 10.3.2 and saw the same error about inverting the APFS image. Any thoughts?

Thanks

nikias, 2017-06-15:
It might be a defective unit so that the file systems appear to have errors. Have you tried an erase restore with ```-e```? Mind that this will wipe all user data though.
Does a restore with iTunes work?