libimobiledevice/idevicerestore #169: can a device be restored to its defaults?

daniel-bluesea, 2017-06-05:
can i ran:
idevicerestore -e

without supplying a firmware file so the device will restore to its defaults?

related issue: https://github.com/libimobiledevice/libimobiledevice/issues/509

braden-wright93, 2017-06-14:
Issue tickets are supposed to be for *issues* with the program, not general help.

You would be better off doing that option along with a .ipsw file that matches the current firmware, just to make sure it all goes correctly. You can get those files from https://ipsw.me and everything will work fine assuming you have the proper SHSH/SHSH2 blobs or Apple is still signing that firmware.

Alternatively, you could supply the `-L` option to make it restore the device with the latest firmware.

Please close your other issue rather than spamming this area further.

nikias, 2017-06-15:
No you can't, and this is not the purpose of idevicerestore.

closed 2017-06-15 by nikias

daniel-bluesea, 2017-06-19:
@braden-wright93 i reject your comment about me spamming "this area". i'm a satisfied customer of the libimobiledevice project. my questions/comments/suggestion are solely for the purpose of making this project better.

regarding the subject itself, i'm well aware of the restore procedure with an ipsw file. the issue i encountered involved a device with an older (unsigned) firmware version (9.3.5) that needed to do erase all content and settings. some third party app, that are heavily based on iTunes offers this option, so i wondered whether it is possible using libimopbiledevice.

 @nikias thanks for your reply