libimobiledevice/idevicerestore #171: iPhone restore to 10.3.2

rdg10, 2017-06-15:
Hi.  I'm having a failure in idevicerestore when trying to update to 10.3.2 on my iPhone.  

Attached is the log.  Looks like everything is working right up until the baseband  tss request at the end.  I'm compiling and running on Windows 10.  I've incorporated the latest changes in idevicerestore and libplist. 

Any help would be greatly appreciated.

[C39QR4RVGRWP.txt](https://github.com/libimobiledevice/idevicerestore/files/1078314/C39QR4RVGRWP.txt)

nikias, 2017-06-15:
Hi, the issue seems to be this:
```
Updating baseband (19)
About to send BasebandData...
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... TSS server returned: STATUS=126&MESSAGE=An internal error occurred.
ERROR: TSS request failed (status=126, message=An internal error occurred.)
```
Seems like some of the submitted data for the TSS request is missing or invalid... can you run the command again with -d ?

rdg10, 2017-06-16:
Nikias,

Here are two files with different amounts of debug info from two different runs.  

[C39QR4RVGRWP_full_debug.txt](https://github.com/libimobiledevice/idevicerestore/files/1080821/C39QR4RVGRWP_full_debug.txt)

[C39QR4RVGRWP_normal_debug.txt](https://github.com/libimobiledevice/idevicerestore/files/1080824/C39QR4RVGRWP_normal_debug.txt)

nikias, 2017-06-16:
Hi, I can see what's wrong, I just don't know why yet. The Baseband TSS request doesn't have the ECID (and thus some other entries) added. See error message:
```
ERROR: Unable to find required ApECID in parameters
```
However the ApECID is added here:
https://github.com/libimobiledevice/idevicerestore/blob/master/src/restore.c#L1550

Maybe you can add a ```debug_plist(parameters);``` in line 1557 so see if it gets added at all.

rdg10, 2017-06-16:
Nikias,

I did that and it looks like the ApECID is in there 

dict>
	<key>@Locality</key>
	<string>en_US</string>
	<key>@HostPlatformInfo</key>
	<string>windows</string>
	<key>@VersionInfo</key>
	<string>libauthinstall-293.1.16</string>
	<key>@UUID</key>
	<string>72ABCE56-3C3B-B03A-3E7C-DF993299E2A9</string>
	<key>ApECID</key>
	<integer>977853000795066</integer>
	<key>UniqueBuildID</key>
	<data>
	/JEPyccAY7w8jvwy+n31qZb9KU0=
	</data>
	<key>ApChipID</key>
	<integer>32771</integer>
	<key>ApBoardID</key>
	<integer>6</integer>
	<key>ApSecurityDomain</key>
	<integer>1</integer>
	<key>AOP</key>

[C39QR4RVGRWP.txt](https://github.com/libimobiledevice/idevicerestore/files/1081253/C39QR4RVGRWP.txt)

Roger

nikias, 2017-06-16:
Actually the line of code I told you to put in line 1557 creates the following output:
```
<plist version="1.0">
<dict>
	<key>ApECID</key>
	<integer>977853000795066</integer>
	<key>BbChipID</key>
	<integer>9572577</integer>
	<key>BbGoldCertId</key>
	<integer>3840149528</integer>
	<key>BbSNUM</key>
	<data>
	CB99Sg==
	</data>
</dict>
</plist>
```
So yes, the ApECID is there as expected. However according to the log output later in the actual TSS request it is not present anymore. So we need to figure out where it gets lost.
In Line 1588, we have this function call:
```
tss_parameters_add_from_manifest(parameters, build_identity);
```
You should add a ```debug_plist(parameters);``` directly after it.
Then, in line 1568, we have these two function calls. The first one prints the error message about the missing ```ApECID``` key.
```
/* add baseband parameters */
tss_request_add_common_tags(request, parameters, NULL);
tss_request_add_baseband_tags(request, parameters, NULL);
```
So let's add a ```debug_plist(request);``` in between these two function calls.
Then paste the log file here.

rdg10, 2017-06-17:
Here it is...

[C39QR4RVGRWP.txt](https://github.com/libimobiledevice/idevicerestore/files/1082733/C39QR4RVGRWP.txt)

nikias, 2017-06-18:
The output directly after  ```tss_parameters_add_from_manifest()``` shows a malformed plist. Look at the start:
```
<plist version="1.0">
<dict>
	<integer>9572577</integer>
	<key>ApECID</key>
	<integer>977853000795066</integer>
	<key>BbChipID</key>
	<key>BbGoldCertId</key>
	<integer>3840149528</integer>
	<key>BbSNUM</key>
	<data>
	CB99Sg==
	</data>
[...]
```
The ```<key>BbChipID</key>``` is somehow not in the right place (in this case it should be the topmost entry in the dict, the value is there. That's how the ApECID gets lost.
Can you please make sure you are using the latest libplist code (the one that doesn't require libxml)?

rdg10, 2017-06-19:
Somehow that fixed itself after a full rebuild of libplist but now that I've moved to other phones there seem to be new issues....  

[DNPKQS14F8H2.txt](https://github.com/libimobiledevice/idevicerestore/files/1085333/DNPKQS14F8H2.txt)
[C39M374XFFDP.txt](https://github.com/libimobiledevice/idevicerestore/files/1085335/C39M374XFFDP.txt)

rdg10, 2017-06-19:
Here is a log with debugging turned on...

[DNPKQS14F8H2.txt](https://github.com/libimobiledevice/idevicerestore/files/1085420/DNPKQS14F8H2.txt)

rdg10, 2017-06-20:
My iPad is experiencing the same issue

[DMPJNZ2MF184.txt](https://github.com/libimobiledevice/idevicerestore/files/1088385/DMPJNZ2MF184.txt)

nikias, 2017-06-22:
Can you add a
```debug_plist(packet_info);```
into ```asr.c``` line 226 right before the ```if (asr_send(...``` and repost a (debug) log output again please.

rdg10, 2017-06-22:
Here it is...

[DMPJNZ2MF184.txt](https://github.com/libimobiledevice/idevicerestore/files/1095005/DMPJNZ2MF184.txt)

nikias, 2017-06-23:
@rdg10 your output looks a little weird. It seems a bit out of order and you have some additional output (like "LATEST FIRMWARE : END"). Did you modify the source code in any way other than for what we tested here?
In the last run the RestoreDeviceTree image cannot be sent to the device; it's a different error condition as before. Also, your code sends the RestoreKernelCache before sending the RestoreDeviceTree. The original code doesn't do this, as you can see here: https://github.com/libimobiledevice/idevicerestore/blob/master/src/recovery.c#L212-L221
Also, the log shows multiple devices connected. Can you please just connect one device so we can figure out the real problem?
Please check if your actual error now is ```ERROR: ASR was unable to validate the filesystem```.

rdg10, 2017-06-23:
I did add in my own comments at some point when I was having issues.  Same code with the same output as when I started this issue thread though.  Here is the new file.
[DMPJNZ2MF184.txt](https://github.com/libimobiledevice/idevicerestore/files/1099012/DMPJNZ2MF184.txt)

rdg10, 2017-06-23:
I just looked and my code is identical to your link with the highlighted functions.  I'm not sure how that could happen out of order.

/* send logo and show it */
	if (recovery_send_applelogo(client, build_identity) < 0) {
		error("ERROR: Unable to send AppleLogo\n");
		return -1;
	}

	/* send ramdisk and run it */
	if (recovery_send_ramdisk(client, build_identity) < 0) {
		error("ERROR: Unable to send Ramdisk\n");
		return -1;
	}

	/* send devicetree and load it */
	if (recovery_send_devicetree(client, build_identity) < 0) {
		error("ERROR: Unable to send DeviceTree\n");
		return -1;
	}

	if (recovery_send_kernelcache(client, build_identity) < 0) {
		error("ERROR: Unable to send KernelCache\n");
		return -1;
	}

nikias, 2017-06-24:
It seems like it can't even open the filesystem file. Can you add this:
```	error("ERROR: Could not open file %s for reading: %s\n", filesystem, strerror(errno));```
at line 204 of asr.c, right before the ```return -1;```.

rdg10, 2017-09-23:
Everything is working now as expected.

closed 2017-09-23 by rdg10