libimobiledevice/idevicerestore #175: Failed Inverting

nick-merlino, 2017-07-11:
Background: Half way through the OTA upgrade from IOS 11 beta 2 to beta 3, my phone went into recovery mode. Attempting to "update" back to 10.3.2 through iTunes led to error 6 and then error 7 on a later attempt. Apple support said that their current state of my phone is 10.3.2 on their end.

Upon trying idevicerestore with flags -l and -d, about half way through the loading bar I'm getting "invert_apfs_image : failed inverting : /System/Library/Filesystems/apfs.fs/apfs_invert /dev/disk0s1 0 apfs_invert_asr_img"

It seemed that this issue was encountered before but it was supposedly solved and added to the GitHub earlier this year. However I'm curious to how to fix this, as it seems the fix isn't working. 

Below is the idevicerestore log.

Extracting filesystem from IPSW
[==================================================] 100.0%
DEBUG: tss_response_get_path_by_entry: No entry 'iBEC' in TSS response
NOTE: No path for component iBEC in TSS, will fetch from build_identity
Extracting iBEC.d10.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (416053 bytes)...
opening device 05ac:1281 @ 0xfa130000...
Setting to configuration 1
Setting to interface 0:0
Recovery Mode Environment:
iBoot build-version=iBoot-3406.60.10
iBoot build-style=RELEASE
Sending AppleLogo...
DEBUG: tss_response_get_path_by_entry: No entry 'AppleLogo' in TSS response
NOTE: No path for component AppleLogo in TSS, will fetch from build_identity
Extracting applelogo@2x~iphone.t8010.im4p...
Personalizing IMG4 component AppleLogo...
Sending AppleLogo (12821 bytes)...
ramdisk-size=0x10000000
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreRamDisk' in TSS response
NOTE: No path for component RestoreRamDisk in TSS, will fetch from build_identity
Extracting 058-72873-060.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (42424499 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreDeviceTree' in TSS response
NOTE: No path for component RestoreDeviceTree in TSS, will fetch from build_identity
Extracting DeviceTree.d10ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (150644 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreKernelCache' in TSS response
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.iphone9...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (13861231 bytes)...
About to restore device... 
Waiting for device...
Attempt 1 to connect to restore mode device...
Attempt 2 to connect to restore mode device...
Attempt 3 to connect to restore mode device...
Attempt 4 to connect to restore mode device...
Attempt 5 to connect to restore mode device...
Attempt 6 to connect to restore mode device...
Attempt 7 to connect to restore mode device...
Attempt 8 to connect to restore mode device...
Attempt 9 to connect to restore mode device...
Attempt 10 to connect to restore mode device...
Attempt 11 to connect to restore mode device...
restore_is_current_device: Connected to com.apple.mobile.restored, version 14
Attempt 12 to connect to restore mode device...
Device ffffffffffffffffffffffffffffffff00000001 is now connected in restore mode...
Connecting now...
Connected to com.apple.mobile.restored, version 14
Device ffffffffffffffffffffffffffffffff00000001 has successfully entered restore mode
Hardware Information:
BoardID: 8
ChipID: 32784
UniqueChipID: 616732478013498
ProductionMode: true
Previous restore exit status: 0x100
Starting FDR listener thread
Connecting to FDR client at port 1082
[... 1953 lines cut ...]
restore-step-codes = {0x1107064F:{0:14};0x11070C05:{0:14}}
restore-step-domains = {0x1107064F:{0:"AMRestoreErrorDomain"};0x11070C05:{0:"AMRestoreErrorDomain"}}
restore-step-error = {0x1107064F:"[0]D(failed to invert APFS image)"}
restore-step-uptime = 187
restore-step-user-progress = 53
[23:22:02.0469-GMT]{4>6} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103064F:48;0x11030C05:53;0x11030647:54}
restore-step-names = {0x1103064F:check_apfs_format;0x11030C05:restore_invert;0x11030647:cleanup_check_result}
restore-step-uptime = 187
restore-step-user-progress = 53
[23:22:02.0469-GMT]{4>6} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103064F:48;0x11030C05:53}
restore-step-names = {0x1103064F:check_apfs_format;0x11030C05:restore_invert}
restore-step-uptime = 187
restore-step-user-progress = 53
[23:22:02.0470-GMT]{4>6} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x1103064F:48;0x11030C05:53;0x11030648:55}
restore-step-names = {0x1103064F:check_apfs_format;0x11030C05:restore_invert;0x11030648:cleanup_send_final_status}
restore-step-uptime = 187
restore-step-user-progress = 53

ERROR: Unable to successfully restore device
No data to read
FDR 0x7f89704cbe50 terminating...
ERROR: Unable to restore device

nikias, 2017-07-11:
Did you try to restore with ```-e``` (will erase all contents on the device!) ?

nick-merlino, 2017-07-12:
I know that's an option however I was an idiot and have no backups of my phone. It contains photos from the past 10 years+ that I would rather not lose. Is it hopeless to recover?

nikias, 2017-07-12:
I don't think there's much we can do here. However the complaining file system is the root filesystem and not the user filesystem which should be overwritten with the new (well, or old) image from the ipsw. But somehow after the transfer something goes wrong... When trying with iTunes is it failing at about the same stage?

nick-merlino, 2017-07-12:
Yes, it seems to fail at the same point, complaining about error 6 (USB error)

nick-merlino, 2017-07-12:
See update on top of post

closed 2017-07-12 by nick-merlino

nikias, 2017-07-12:
Thanks for the update. I think it must be some APFS filesystem specifics that changed in newer iOS versions that are not compatible with older iOS versions, and that's why it fails maybe. So updating to the lastest available version seems to help here.