libimobiledevice/idevicerestore #177: FDR timeout?

quanganhdo, 2017-07-25:
Hi,

I'm having some problems trying to restore my iPad:

`Sending NORData now...
Done sending NORData
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
Checking filesystems (15)
No data to read
FDR 0x7fdfa5f05a10 timeout waiting for command
FDR 0x7fdfa5f05a10 waiting for message...
No data to read
No data to read
FDR 0x7fdfa5f05a10 timeout waiting for command
FDR 0x7fdfa5f05a10 waiting for message...
No data to read
Checking filesystems (15)
FDR 0x7fdfa5f05a10 got sync message
Connecting to FDR client at port 49158
FDR Received 131 bytes
Got device identifier 6b8b4567327b23c6643c9869
FDR connected in reply to sync message, starting command thread
FDR 0x7fdfa5f05a10 waiting for message...
FDR 0x7fdfa7090e40 waiting for message...`

There were endless "No data to read" messages like this. On the actual iPad, the screen stuck at boot screen with the logo and a 0% progress bar. 

Any ideas what's happening? Thanks!

nikias, 2017-07-26:
Can you just try again a few times? The output doesn't really show what is going wrong here.

nullpixel, 2017-09-20:
I get the same issue :(

jalcock0, 2017-10-06:
[debug.log](https://github.com/libimobiledevice/idevicerestore/files/1363345/debug.log)
Hi Nikias,

I have attached more debug logs for endless "No data to read" messages on iPhone 8 in recovery mode

demhademha, 2021-06-04:
Are you on linux?