libimobiledevice/idevicerestore #181: 3G[S] not restoring

JohnJ1101, 2017-08-15:
Does not work in iTunes either, any fix?
 If it helps: I was trying to use @axi0mx 's  alloc8 to "upgrade" from 3.1 (also downgraded alloc8) to 5.0.1 (backup restore), and I used sn0wbreeze. It seems to have flashed NAND, and now my phone is stuck in DFU. I tried again selecting new bootrom, but it wouldn't restore. I tried using redsn0w's "Downgrade from iPad baseband" and it failed/did nothing. Any and all help is appreciated.

iDeviceRestore Log: https://pastebin.com/my5QsdrJ

(Going to post on alloc8 issues as well)

nikias, 2017-08-15:
It just stops at "Partition NAND device" ?

JohnJ1101, 2017-08-15:
Then it says error 28, apparently that was not copied with the rest

JohnJ1101, 2017-08-15:
NAND could not be partitioned

JohnJ1101, 2017-08-15:
Tried running ipwndfu --remove-24kpwn and got this:
https://pastebin.com/urUQT6Gi

JohnJ1101, 2017-08-15:
Fixed!!! I just sent the 3.0 iBSS (./ipwndfu -p -f 'filename.dfu'), screen turned white, plugged into another PC that had windows, restored to 6.1.6 in iTunes, and it started restoring. Will reopen if it fails, but closing now!

closed 2017-08-15 by JohnJ1101