libimobiledevice/idevicerestore #184: iPhone 8 iOS 11 Can't restore.

Kanong017, 2017-09-23:
please update

taanders, 2017-09-26:
Here are some log debug log from iPhone 8 device:
Sending RestoreKernelCache (15982955 bytes)...
STEP:1:PROGRESS:90
About to restore device...
Waiting for device...
Attempt 1 to connect to restore mode device...
Attempt 2 to connect to restore mode device...
Attempt 3 to connect to restore mode device...
Attempt 4 to connect to restore mode device...
Attempt 5 to connect to restore mode device...
restore_is_current_device: Connected to com.apple.mobile.restored, version 15
Attempt 6 to connect to restore mode device...
Device ffffffffffffffffffffffffffffffff00000022 is now connected in restore mode...
Connecting now...
Connected to com.apple.mobile.restored, version 15
Device ffffffffffffffffffffffffffffffff00000022 has successfully entered restore mode
Hardware Information:
BoardID: 10
ChipID: 32789
UniqueChipID: 4048234377047982
ProductionMode: true
Starting FDR listener thread
Connecting to FDR client at port 1082
About to do ctrl handshake
FDR sending 89 bytes:
common.c:printing 287 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Command</key>
        <string>BeginCtrl</string>
        <key>CtrlProtoVersion</key>
        <integer>2</integer>
</dict>
</plist>
FDR Sent 89 bytes
FDR Received 105 bytes
common.c:printing 334 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Command</key>
        <string>BeginCtrl</string>
        <key>CtrlProtoVersion</key>
        <integer>2</integer>
        <key>ConnPort</key>
        <integer>49158</integer>
</dict>
</plist>
Ctrl handshake done (ConnPort = 49158)
STEP:1:PROGRESS:100
FDR 036C8C38 waiting for message...
About to send NORData...
DEBUG: tss_response_get_path_by_entry: No entry 'LLB' in TSS response
NOTE: Could not get LLB path from TSS data, will fetch from build identity
Found firmware path Firmware/all_flash
NOTE: zip_name_locate: 'Firmware/all_flash/manifest' not found in archive.
Getting firmware manifest from build identity
Extracting LLB.d20.RELEASE.im4p...
Personalizing IMG4 component LLB...
Extracting applelogo@2x~iphone.im4p...
Personalizing IMG4 component AppleLogo...
Extracting batterycharging0@2x~iphone.im4p...
Personalizing IMG4 component BatteryCharging0...
Extracting batterycharging1@2x~iphone.im4p...
Personalizing IMG4 component BatteryCharging1...
Extracting batteryfull@2x~iphone.im4p...
Personalizing IMG4 component BatteryFull...
Extracting batterylow0@2x~iphone.im4p...
Personalizing IMG4 component BatteryLow0...
Extracting batterylow1@2x~iphone.im4p...
Personalizing IMG4 component BatteryLow1...
Extracting glyphplugin@1334~iphone-lightning.im4p...
Personalizing IMG4 component BatteryPlugin...
Extracting DeviceTree.d201ap.im4p...
Personalizing IMG4 component DeviceTree...
Extracting liquiddetect@1334~iphone-lightning.im4p...
Personalizing IMG4 component Liquid...
Extracting recoverymode@1334~iphone-lightning.im4p...
Personalizing IMG4 component RecoveryMode...
Extracting iBoot.d20.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Extracting sep-firmware.d201.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Extracting sep-firmware.d201.RELEASE.im4p...
Personalizing IMG4 component SEP...
common.c:supressed printing 24816700 bytes plist...
Sending NORData now...
Done sending NORData
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
Updating S3E Firmware (58)
ERROR: Unable to receive message from FDR 036C8C38 (-4). 0/2 bytes
FDR 036C8C38 terminating...
No data to read
No data to read
No data to read
No data to read
No data to read
No data to read

Kanong017, 2017-09-26:
@taanders me too,Is there any solution?

taanders, 2017-09-26:
And here is some debug log from iPhone 8 Plus started on normal mode:
```
Personalizing IMG4 component Multitouch...
Sending FUD data now...
Done sending FUD data
No data to read
FDR 036C5020 timeout waiting for command
FDR 036C5020 waiting for message...
Updating baseband (19)
About to send BasebandData...
DEBUG: Unknown file format passed to mbn_parse
WARNING: size mismatch when parsing MBN file. Continuing anyway.
DEBUG: Unknown file format passed to mbn_parse
WARNING: size mismatch when parsing MBN file. Continuing anyway.
Reading data from C:\Users\TAPIO-~1\AppData\Local\Temp\ffffffffffffffffffffffffffffffff000000252
Sending BasebandData now...
Done sending BasebandData
Updating Baseband in progress...
About to send BasebandData...
DEBUG: Unknown file format passed to mbn_parse
WARNING: size mismatch when parsing MBN file. Continuing anyway.
DEBUG: Unknown file format passed to mbn_parse
WARNING: size mismatch when parsing MBN file. Continuing anyway.
Reading data from C:\Users\TAPIO-~1\AppData\Local\Temp\ffffffffffffffffffffffffffffffff000000253
Sending BasebandData now...
Done sending BasebandData
Updating Baseband completed.
Updating SE Firmware (59)
DEBUG: Got FirmwareUpdaterData request:
common.c:printing 979 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Arguments</key>
        <dict>
                <key>MessageArgInfo</key>
                <dict>
                        <key>SE,ChipID</key>
                        <integer>115</integer>
                        <key>SE,ID</key>
                        <data>
                        BCY0E9dBgAFxcwIxdJQgAxXevuUuVFHx
                        </data>
                        <key>SE,Nonce</key>
                        <data>
                        w3NDjoU8Fl05FvppF/4jF+wV7+w=
                        </data>
                        <key>SE,OSUPubKeyID</key>
                        <data>
                        p800Mng1cTNLZzbdPzWEwI9KtaJdVE1+NzuicYuTvZU=
                        </data>
                        <key>SE,RootKeyIdentifier</key>
                        <data>
                        YzErNV1rVT5vssRti6cpWKctuCBprCmvZDv6ysfO+uI=
                        </data>
                </dict>
                <key>MessageArgType</key>
                <string>FirmwareResponseData</string>
                <key>MessageArgUpdaterLoopCount</key>
                <integer>0</integer>
                <key>MessageArgUpdaterName</key>
                <string>SE</string>
        </dict>
        <key>DataType</key>
        <string>FirmwareUpdaterData</string>
        <key>MsgType</key>
        <string>DataRequestMsg</string>
</dict>
</plist>
ERROR: Unable to find component node for SE,Firmware
ERROR: Unable get path for 'SE,Firmware' component
ERROR: restore_send_firmware_updater_data: Couldn't get SE firmware data
ERROR: Unable to send FirmwareUpdater data
ERROR: Unable to successfully restore device
```

Kanong017, 2017-09-26:
@taanders iPhone 7 is S3E firmware, iPhone 8 is S4E firmware, there is no this possible?

LegitSoulja, 2017-09-26:
I need help aswell. I am running futurerestore on Ubuntu but failed to connect to gp.apple.com socket. 

I seen that I need to install iTunes however I am no Ubuntu. Any way around this? Thanks

LegitSoulja, 2017-09-26:
```
$ ./futurerestore_linux -t blob.shsh2 -b Mav10-5.24.00.Release.bbfw -p BuildManifest.plist -s sep-firmware.n61.RELEASE.im4p -m BuildManifest.plist iPhone_4.7_10.0.2_14A456_Restore.ipsw
Version: b99eb8140d8e6c23f34e950102bb79e61c72384d - 152
Libipatcher Version: a6cc5cfa8d421987d06f6157130ef3c75391efa4 - 39
Odysseus Support: yes
[INFO] 64bit device detected
futurerestore init done
reading ticket blob.shsh2 done
[TSSC] opening BuildManifest.plist
WARNING: Unable to find BbSkeyId node
[TSSR] User specified not to request a Baseband ticket.
Request URL set to https://gs.apple.com/TSS/controller?action=2
TSS server returned: <html>
<head><title>403 Forbidden</title></head>
<body bgcolor="white">
<center><h1>403 Forbidden</h1></center>
<hr><center>Shield</center>
</body>
</html>

Sending TSS request attempt 1... Request URL set to https://17.171.36.30/TSS/controller?action=2
Failed to connect to 17.171.36.30 port 443: Connection timed out
Sending TSS request attempt 2... Request URL set to https://17.151.36.30/TSS/controller?action=2

```

taanders, 2017-09-27:
Some logs from iPhone 8 when restore was started on recovery mode and bypassed error lines above:
```
bbupdater: 1506500817.453: sHandleCapureProgress: dynamic.nvm:  79.8%
bbupdater: 1506500817.455: sHandleCapureProgress: dynamic.nvm:  81.1%
bbupdater: 1506500817.460: sHandleCapureProgress: dynamic.nvm:  82.4%
bbupdater: 1506500817.464: sHandleCapureProgress: dynamic.nvm:  83.7%
bbupdater: 1506500817.467: sHandleCapureProgress: dynamic.nvm:  85.0%
bbupdater: 1506500817.471: sHandleCapureProgress: dynamic.nvm:  86.3%
bbupdater: 1506500817.474: sHandleCapureProgress: dynamic.nvm:  87.6%
bbupdater: 1506500817.479: sHandleCapureProgress: dynamic.nvm:  88.9%
bbupdater: 1506500817.482: sHandleCapureProgress: dynamic.nvm:  90.1%
bbupdater: 1506500817.485: sHandleCapureProgress: dynamic.nvm:  91.4%
bbupdater: 1506500817.489: sHandleCapureProgress: dynamic.nvm:  92.7%
bbupdater: 1506500817.493: sHandleCapureProgress: dynamic.nvm:  94.0%
bbupdater: 1506500817.497: sHandleCapureProgress: dynamic.nvm:  95.3%
bbupdater: 1506500817.501: sHandleCapureProgress: dynamic.nvm:  96.6%
bbupdater: 1506500817.504: sHandleCapureProgress: dynamic.nvm:  97.9%
bbupdater: 1506500817.507: sHandleCapureProgress: dynamic.nvm:  99.2%
bbupdater: 1506500817.511: sHandleCapureProgress: dynamic.nvm:  100.0%
bbupdater: 1506500817.512: ICEStateMachinePerformSnapshot: ---------------------UpdateAck---------------------
bbupdater: 1506500817.524: sUpdateAck: deleting backup file /mnt3/bbfs/backup/static.nvm
bbupdater: 1506500817.525: sUpdateAck: deleting backup file /mnt3/bbfs/backup/dynamic.nvm
bbupdater: 1506500817.526: ICEStateMachinePerformSnapshot: device has no further updates
bbupdater: 1506500817.527: sDeRegister: De-registering for NVM updates
bbupdater:      END: Final stage
bbupdater: END: Command: performNextStage
send_bbupdate_status: sending baseband update status to host
update_baseband: success
update_baseband: calling CFNotificationCenterRemoveEveryObserver
[08:26:57.0544-GMT]{5>8} CHECKPOINT END: FIRMWARE:[0x1303] update_baseband
restore-step-ids = {0x11030635:60}
restore-step-names = {0x11030635:update_device_firmware}
restore-step-uptime = 195
restore-step-user-progress = 55
[08:26:57.0549-GMT]{5>8} CHECKPOINT BEGIN: FIRMWARE:[0x1309] update_se
restore-step-ids = {0x11030635:60;0x11031309:75}
restore-step-names = {0x11030635:update_device_firmware;0x11031309:update_se}
restore-step-uptime = 195
restore-step-user-progress = 55
entering update_se
ramrod_display_set_granular_progress_forced: 85.000000
opening /usr/lib/updaters/libSEUpdater.dylib
copy_updater_options: SE options: <CFBasicHash 0x108500b70 [0x10541ed70]>{type = mutable dict, count = 0,
entries =>
}

copy_updater_options: creating SEUpdater instance with options=<CFBasicHash 0x108500b70 [0x10541ed70]>{type = mutable dict, count = 3,
entries =>
        0 : <CFString 0x1044f7820 [0x10541ed70]>{contents = "PostFDRSealing"} = <CFBoolean 0x10541f2d8 [0x10541ed70]>{value = false}
        1 : <CFString 0x1044f7a00 [0x10541ed70]>{contents = "RestoreInternal"} = <CFBoolean 0x10541f2d8 [0x10541ed70]>{value = false}
        2 : <CFString 0x1044f79e0 [0x10541ed70]>{contents = "RestoreSystemPartition"} = /mnt1
}

creating SEUpdater obj
SE: [1506500817.732][WARNING] parseStartOptions: Additional actions: FTA will be deactivated, MFD will be performed
SE: [1506500817.930][WARNING] NFDriverSecureElementGetCertificates:296  Storing cert 0 of type 1 and length 237
SE: [1506500817.931][WARNING] NFDriverSecureElementGetCertificates:296  Storing cert 1 of type 2 and length 195
SE: [1506500817.938][WARNING] NFDriverSecureElementGetCertificates:296  Storing cert 2 of type 3 and length 195
copying SE updater response for loop 0
unable to convert message to property list
restored_receive_message: CFPropertyListCreateFromXMLData: The operation couldn\U2019t be completed. (NSCocoaErrorDomain error 3840.)

update_generic_firmware: Underlying error: (null)
Update[SE] error - The operation couldn\U2019t be completed. (RamrodErrorDomain error 1020 - update_generic_firmware: faile to copy response for SE)
[08:26:58.0099-GMT]{5>8} CHECKPOINT FAILURE:(FAILURE:-1) FIRMWARE:[0x1309] update_se [0]D(update_generic_firmware: faile to copy response for SE)
restore-step-results = {0x11071309:{0:-1}}
restore-step-codes = {0x11071309:{0:1020}}
restore-step-domains = {0x11071309:{0:"RamrodErrorDomain"}}
restore-step-error = {0x11071309:"[0]D(update_generic_firmware: faile to copy response for SE)"}
restore-step-uptime = 196
[... 2 lines cut ...]
0: RamrodErrorDomain/3fc: update_generic_firmware: faile to copy response for SE
unable to convert ramrod error 1020
[08:26:58.0106-GMT]{5>8} CHECKPOINT FAILURE:(FAILURE:-1) RESTORED:[0x0635] update_device_firmware [0]D(failed to update device firmware)[1]D(ramrod firmware update failed)[2]D(update_generic_firmware: faile to copy response for SE)
restore-step-results = {0x11070635:{0:-1};0x11071309:{0:-1}}
restore-step-codes = {0x11070635:{0:-1};0x11071309:{0:1020}}
restore-step-domains = {0x11070635:{0:"AMRestoreErrorDomain"};0x11071309:{0:"RamrodErrorDomain"}}
restore-step-error = {0x11070635:"[0]D(failed to update device firmware)[1]D(ramrod firmware update failed)[2]D(update_generic_firmware: faile to copy response for SE)"}
restore-step-uptime = 196
restore-step-user-progress = 85
[08:26:58.0110-GMT]{5>8} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x11030635:60;0x11031309:75;0x11030647:76}
[... cut ...]
restore-step-ids = {0x11030635:60;0x11031309:75}
restore-step-names = {0x11030635:update_device_firmware;0x11031309:update_se}
restore-step-uptime = 196
restore-step-user-progress = 85
[08:26:58.0114-GMT]{5>8} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x11030635:60;0x11031309:75;0x11030648:77}
restore-step-names = {0x11030635:update_device_firmware;0x11031309:update_se;0x11030648:cleanup_send_final_status}
restore-step-uptime = 196
restore-step-user-progress = 85
```

taanders, 2017-09-28:
Are those logs helpful for solving this issue or should I add full logs about those failed restore processes? It feels like iPhone 8 waits for some kind of message from PC-side and cannot continue before gets that message or some message before is given on wrong way.

On iTunes logs there are this kind of event at the end:
```
[06:44:01.0282] <Restore Device 0000025762236D20>: operation 59 progress 0
[06:44:01.0798] requested restore behavior: Erase
[06:44:01.0845] amai: AMAuthInstallRequestSendSync: SSO function returned NULL and no SSO token was provided, SSO disabled.
[06:44:01.0861] amai: AMAuthInstallDebugWriteObject: debug object written: file://localhost/C:/Users/Testuser/AppData/Local/Temp/Per8CFC.tmp/amai/debug/tss-request.plist
[06:44:01.0861] amai: tss_submit_job_with_retry: TSS Connection attempt 1 of 3.  (Will retry if TSS_ERR_SERVER_NOT_REACHABLE.)
[06:44:02.0099] amai: tss_submit_job: HttpQueryInfo returned 200
[06:44:02.0099] amai: AMAuthInstallRequestSendSync: received tss response (server version: 2.1.0)
[06:44:02.0114] amai: AMAuthInstallDebugWriteObject: debug object written: file://localhost/C:/Users/Testuser/AppData/Local/Temp/Per8CFC.tmp/amai/debug/tss-response.plist
[06:44:02.0316] Completed checkpoint id: 0x635
[06:44:02.0317] Completed checkpoint id: 0x663
[06:44:02.0317] <Restore Device 0000025762236D20>: operation 17 progress -1
[06:44:03.0484] <Restore Device 0000025762236D20>: operation 50 progress -1
[06:44:13.0624] Completed checkpoint id: 0xF03
[06:44:13.0624] Completed checkpoint id: 0x64D
[06:44:13.0624] Completed checkpoint id: 0x664
[06:44:13.0640] <Restore Device 0000025762236D20>: operation 25 progress -1
[06:44:13.0640] Completed checkpoint id: 0x639
[06:44:13.0655] Completed checkpoint id: 0x65E
[06:44:13.0693] Completed checkpoint id: 0x63C
[06:44:13.0718] Completed checkpoint id: 0x1501
[06:44:13.0721] Completed checkpoint id: 0x668
[06:44:13.0721] Completed checkpoint id: 0x63D
[06:44:13.0721] Completed checkpoint id: 0x63F
[06:44:13.0736] Completed checkpoint id: 0x640
[06:44:13.0736] Completed checkpoint id: 0x669
[06:44:13.0736] <Restore Device 0000025762236D20>: operation 29 progress -1
[06:44:13.0791] <Restore Device 0000025762236D20>: operation 29 progress -1
[06:44:13.0801] <Restore Device 0000025762236D20>: operation 29 progress -1
[06:44:13.0816] Completed checkpoint id: 0x642
[06:44:13.0831] Completed checkpoint id: 0x647
[06:44:13.0831] device returned success
[06:44:13.0831] Sending acknowledgement that final status was received
[06:44:13.0831] <Restore Device 0000025762236D20>: Restore succeeded
[06:44:13.0831] Restore completed, status:0
```

nikias, 2017-09-28:
Guys, can you please update the code to lastest from git, and then try out this patch:
https://gist.github.com/nikias/3ce0a34964e8f8a256cf793540b95328
And report back if this works for you? Thanks.

jalcock0, 2017-09-29:
I restored the iPhone 8 Plus successfully and its working . 
thank you very much

Kanong017, 2017-09-29:
@nikias Thank you for your help

nikias, 2017-09-29:
Thanks for testing! The code is now in git: 875c9696ce691db571f8910dce062ae560703a51

closed 2017-09-29 by nikias

LegitSoulja, 2017-09-30:
@nikias does it properly support downgrading?

LegitSoulja, 2017-09-30:
And I have my blob, I believe it is 10.0.2, or before that. I however tested it but it failed to be failed using TSS checker.

I created my blob on IOS using telegram, which uses TSS as an API I believe.

nikias, 2017-09-30:
@LegitSoulja idevicerestore can only downgrade if Apple still signs the firmware. Everything else is not supported.

LegitSoulja, 2017-09-30:
@nikias ok :/