libimobiledevice/idevicerestore #185: fails in Sending TSS request

ankitrwt, 2017-09-25:
idevicerestore -d -e -l -C . 

Sending TSS request attempt 1... Request URL set to https://17.171.36.30/TSS/controller?action=2
Failed to connect to 17.171.36.30 port 443: Connection timed out
Sending TSS request attempt 2... Request URL set to https://17.151.36.30/TSS/controller?action=2
Failed to connect to 17.151.36.30 port 443: Connection timed out
Sending TSS request attempt 3... Request URL set to http://gs.apple.com/TSS/controller?action=2
TSS server returned: <html>
<head><title>403 Forbidden</title></head>
<body bgcolor="white">
<center><h1>403 Forbidden</h1></center>
<hr><center>Shield</center>
</body>
</html>

Rayee666, 2017-09-25:
I got the same error.
Pls help.

4kzeng, 2017-09-25:
iTunes  also get failed,  https-sniff-tools found that gs.apple.com server refuses request.

ankitrwt, 2017-09-25:
i tried on iPhone4,4s,5,5s,6 & 6s and i got same error, 
debug

[root@System]# idevicerestore -d -e -l -C .
Reading data from ./version.xml
NOTE: using cached version data
opening device 05ac:1281...
Setting to configuration 1
Setting to interface 0:0
Found device in Recovery mode
opening device 05ac:1281...
Setting to configuration 1
Setting to interface 0:0
Identified device as n53ap, iPhone6,2
Latest firmware is iPhone_4.0_64bit_11.0_15A372_Restore.ipsw
Verifying './iPhone_4.0_64bit_11.0_15A372_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 11.0
Product Build: 15A372 Major: 15
opening device 05ac:1281...
Setting to configuration 1
Setting to interface 0:0
INFO: device serial number is DX3QWJDXFRC4
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Getting device's ECID for TSS request
Found ECID 5723206868272
Getting ApNonce in recovery mode... a1 91 2d 0f ca c0 43 ff 57 a0 13 13 54 37 fd f4 5c e6 d9 0b 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... bd cf 7d b4 d3 af 67 1f 1e 34 80 eb ee 37 ac 0a 7e 64 e3 ce 
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbFDRSecurityKeyHash node
WARNING: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry AppleLogo
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging0
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging1
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryFull
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow0
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow1
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryPlugin
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry DeviceTree
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry KernelCache
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry LLB
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry OS
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry RecoveryMode
[... 396 lines cut ...]
</body>
</html>

Failed to connect to 17.151.36.30 port 443: Connection timed out
Sending TSS request attempt 10... Request URL set to http://17.171.36.30/TSS/controller?action=2
Recv failure: Connection reset by peer
Sending TSS request attempt 11... Request URL set to http://17.151.36.30/TSS/controller?action=2
Recv failure: Connection reset by peer
Sending TSS request attempt 12... Request URL set to https://gs.apple.com/TSS/controller?action=2
TSS server returned: <html>
<head><title>403 Forbidden</title></head>
<body bgcolor="white">
<center><h1>403 Forbidden</h1></center>
<hr><center>Shield</center>
</body>
</html>

Recv failure: Connection reset by peer
Sending TSS request attempt 13... Request URL set to https://17.171.36.30/TSS/controller?action=2
Failed to connect to 17.171.36.30 port 443: Connection timed out
Sending TSS request attempt 14... Request URL set to https://17.151.36.30/TSS/controller?action=2
Failed to connect to 17.151.36.30 port 443: Connection timed out
ERROR: TSS request failed: Failed to connect to 17.151.36.30 port 443: Connection timed out (status=-1)
Segmentation fault (core dumped)
[root@System]#

nicolassan, 2017-09-25:
Same server problem with DFU in itunes, maybe a server problem

ankitrwt, 2017-09-25:
update your itunes then it will work fine, @nicolassan

daniel-bluesea, 2017-09-25:
@ankitrwt its not an itunes forum, we need to solve this problem for idevicerestore

daniel-bluesea, 2017-09-26:
anyway.. it seems that the issue was fixed by apple
@ankitrwt please verify and close this issue, 
thanks

ankitrwt, 2017-09-26:
yes issue was resolved but still idevicerestore -d -e -l -C . not working for ios 11
@Rayee666  @daniel-bluesea @4kzeng @posixninja
please verify

ankitrwt, 2017-09-26:
debug,

[root@System]# idevicerestore -d -e -l -C .
Reading data from ./version.xml
NOTE: using cached version data
opening device 05ac:1281...
Setting to configuration 1
Setting to interface 0:0
Found device in Recovery mode
opening device 05ac:1281...
Setting to configuration 1
Setting to interface 0:0
Identified device as j81ap, iPad5,3
Latest firmware is iPad_64bit_TouchID_11.0_15A372_Restore.ipsw
Verifying './iPad_64bit_TouchID_11.0_15A372_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 11.0
Product Build: 15A372 Major: 15
opening device 05ac:1281...
Setting to configuration 1
Setting to interface 0:0
INFO: device serial number is DMRNME2YG5VV
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Getting device's ECID for TSS request
Found ECID 1751011606743078
Getting ApNonce in recovery mode... 9b d9 ea 96 4b 6f ff fa 89 c3 d9 ca 51 ec 1b a3 b2 15 ce 5d 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 7c 00 f5 01 2f 29 67 b3 e7 84 2c ee 0c 99 d7 b9 75 a3 9e 25 
WARNING: Unable to find BbChipID node
NOTE: Unable to find BbProvisioningManifestKeyHash node
NOTE: Unable to find BbActivationManifestKeyHash node
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbFDRSecurityKeyHash node
WARNING: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry AppleLogo
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging0
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging1
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryFull
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow0
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow1
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryPlugin
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry DeviceTree
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry KernelCache
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry LLB
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry OS
DEBUG: Adding action EPRO to TSS entry
[... 737 lines cut ...]
Attempt 160 to connect to restore mode device...
Attempt 161 to connect to restore mode device...
Attempt 162 to connect to restore mode device...
Attempt 163 to connect to restore mode device...
Attempt 164 to connect to restore mode device...
Attempt 165 to connect to restore mode device...
Attempt 166 to connect to restore mode device...
Attempt 167 to connect to restore mode device...
Attempt 168 to connect to restore mode device...
Attempt 169 to connect to restore mode device...
Attempt 170 to connect to restore mode device...
Attempt 171 to connect to restore mode device...
Attempt 172 to connect to restore mode device...
Attempt 173 to connect to restore mode device...
Attempt 174 to connect to restore mode device...
Attempt 175 to connect to restore mode device...
Attempt 176 to connect to restore mode device...
Attempt 177 to connect to restore mode device...
Attempt 178 to connect to restore mode device...
Attempt 179 to connect to restore mode device...
Attempt 180 to connect to restore mode device...
ERROR: Unable to connect to device in restore mode
ERROR: Unable to open device in restore mode
ERROR: Unable to restore device
[root@System]#

nikias, 2017-09-26:
TSS works fine, and I just restored an iPad5,1 (j96ap) successfully to iOS 11.
```
$ idevicerestore -e ~/Downloads/iPad_64bit_TouchID_11.0_15A372_Restore.ipsw
NOTE: Updated version data.
Found device in Normal mode
Identified device as j96ap, iPad5,1
Extracting BuildManifest from IPSW
Product Version: 11.0
Product Build: 15A372 Major: 15
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Found ECID x0778829xxxxxxxx
Getting ApNonce in normal mode... 8c 0a 66 e2 7e 18 11 40 38 15 0c 60 bf 1b 29 50 8f ad 68 10
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... 28 3e f9 77 96 13 6d 71 cf ad 22 4a 26 fc 18 99 a3 ef 3f 72
WARNING: Unable to find BbChipID node
WARNING: Unable to find BbSkeyId node
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting filesystem from IPSW
[==================================================] 100.0%
Entering recovery mode...
INFO: device serial number is Fxxxxxxxxxxx
Extracting iBEC.ipad5.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (706740 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-4076.1.44
iBoot build-style=RELEASE
Sending AppleLogo...
Extracting applelogo@2x~ipad.im4p...
Personalizing IMG4 component AppleLogo...
Sending AppleLogo (23186 bytes)...
ramdisk-size=0x20000000
Extracting 058-59982-349.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (59114995 bytes)...
Extracting DeviceTree.j96ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (135472 bytes)...
Extracting kernelcache.release.ipad5...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (13826677 bytes)...
About to restore device...
Waiting for device...
Device ffffffffc33f1f691a52102ffffffffcffffffff is now connected in restore mode...
Connecting now...
Connected to com.apple.mobile.restored, version 15
Device ffffffffc33f1f691a52102ffffffffcffffffff has successfully entered restore mode
Hardware Information:
BoardID: 8
ChipID: 28672
UniqueChipID: x0778829xxxxxxxx
ProductionMode: true
Starting FDR listener thread
About to send NORData...
Found firmware path Firmware/all_flash
Getting firmware manifest from build identity
Extracting LLB.ipad5.RELEASE.im4p...
Personalizing IMG4 component LLB...
Extracting applelogo@2x~ipad.im4p...
Personalizing IMG4 component AppleLogo...
Extracting batterycharging0@2x~ipad.im4p...
Personalizing IMG4 component BatteryCharging0...
Extracting batterycharging1@2x~ipad.im4p...
Personalizing IMG4 component BatteryCharging1...
Extracting batteryfull@2x~ipad.im4p...
Personalizing IMG4 component BatteryFull...
Extracting batterylow0@2x~ipad.im4p...
Personalizing IMG4 component BatteryLow0...
Extracting batterylow1@2x~ipad.im4p...
Personalizing IMG4 component BatteryLow1...
Extracting glyphplugin@2x~ipad-lightning.im4p...
Personalizing IMG4 component BatteryPlugin...
Extracting DeviceTree.j96ap.im4p...
Personalizing IMG4 component DeviceTree...
Extracting recoverymode@2x~ipad-lightning.im4p...
Personalizing IMG4 component RecoveryMode...
Extracting iBoot.ipad5.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Extracting sep-firmware.j96.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Extracting sep-firmware.j96.RELEASE.im4p...
Personalizing IMG4 component SEP...
Sending NORData now...
Done sending NORData
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
Checking filesystems (15)
Checking filesystems (15)
About to send FDR Trust data...
Sending FDR Trust data now...
Done sending FDR Trust Data
Unmounting filesystems (29)
Unmounting filesystems (29)
Creating partition map (11)
Creating filesystem (12)
About to send filesystem...
Connected to ASR
Validating the filesystem
Filesystem validated
Sending filesystem now...
[==================================================] 100.0%
Done sending filesystem
Verifying restore (14)
[==================================================] 100.0%
Checking filesystems (15)
Checking filesystems (15)
Mounting filesystems (16)
Mounting filesystems (16)
About to send KernelCache...
Extracting kernelcache.release.ipad5...
Personalizing IMG4 component KernelCache...
Sending KernelCache now...
Done sending KernelCache
Installing kernelcache (27)
About to send DeviceTree...
Extracting DeviceTree.j96ap.im4p...
Personalizing IMG4 component DeviceTree...
Sending DeviceTree now...
Done sending DeviceTree
Unknown operation (61)
Flashing firmware (18)
[==================================================] 100.0%
Updating gas gauge software (47)
Updating gas gauge software (47)
Updating Stockholm (55)
About to send FUD data...
Sending FUD data now...
Done sending FUD data
About to send FUD data...
Sending FUD data now...
Done sending FUD data
Updating SE Firmware (59)
Fixing up /var (17)
Creating system key bag (50)
Modifying persistent boot-args (25)
Unmounting filesystems (29)
Unmounting filesystems (29)
Got status message
Status: Restore Finished
Cleaning up...
DONE
```
Also I used ideviceactivation to activate it, without problems.

ankitrwt, 2017-09-26:
i tried on 5s, 6, Air 2 it fails on all but in 4,4s,5 it works fine
any suggestion pls

Orbif, 2017-09-26:
@ankitrwt With the -C parameter, you need to specify a cache path. This is an optional parameter.

Try this : idevicerestore -d -e -l

ankitrwt, 2017-09-26:
it keeps stuck on 
FDR 0x3475440 timeout waiting for command
tried on 5s and 6

@nikias 

[root@System]# idevicerestore -d -e -l -C .
Reading data from ./version.xml
NOTE: using cached version data
opening device 05ac:1281...
Setting to configuration 1
Setting to interface 0:0
Found device in Recovery mode
opening device 05ac:1281...
Setting to configuration 1
Setting to interface 0:0
Identified device as n53ap, iPhone6,2
Latest firmware is iPhone_4.0_64bit_11.0_15A372_Restore.ipsw
Verifying './iPhone_4.0_64bit_11.0_15A372_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 11.0
Product Build: 15A372 Major: 15
opening device 05ac:1281...
Setting to configuration 1
Setting to interface 0:0
INFO: device serial number is DX3QWJDXFRC4
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Getting device's ECID for TSS request
Found ECID 5723206868272
Getting ApNonce in recovery mode... 99 0e 88 2a 8e fa 8d 5c 97 a7 d1 04 ee 9b a2 1f e9 fe 1f b6 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... bd cf 13 70 e1 3b fe 7b 8a d3 1d d5 aa 1c ed 41 82 7d f8 e5 
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbFDRSecurityKeyHash node
WARNING: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry AppleLogo
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging0
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging1
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryFull
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow0
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow1
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryPlugin
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry DeviceTree
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry KernelCache
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry LLB
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry OS
[... 1443 lines cut ...]
No data to read
No data to read
FDR 0x3475440 timeout waiting for command
FDR 0x3475440 waiting for message...
No data to read
No data to read
FDR 0x3475440 timeout waiting for command
FDR 0x3475440 waiting for message...
No data to read
No data to read
FDR 0x3475440 timeout waiting for command
FDR 0x3475440 waiting for message...
No data to read
No data to read
FDR 0x3475440 timeout waiting for command
FDR 0x3475440 waiting for message...
No data to read
No data to read
FDR 0x3475440 timeout waiting for command
FDR 0x3475440 waiting for message...
No data to read
No data to read
FDR 0x3475440 timeout waiting for command
FDR 0x3475440 waiting for message...
No data to read

daniel-bluesea, 2017-09-26:
@ankitrwt according to your debug info you attached you used again! the -C flag that @Orbif suggested you to omit.

also, (i think) you can attached and/or summarize the important debug info, no need for full output for every post you're writing
thanks :)

nikias, 2017-09-26:
@daniel-bluesea @Orbif he did specify a cache path "." which is the current directory and that has nothing to do with the actual problem. I just saw you are trying from recovery mode. I will try that and also from DFU with my test device.

nikias, 2017-09-26:
@ankitrwt are you sure you are using the latest code? At least the iPad Air 2 should restore, it is very similar to the iPad mini 4 which worked for me.

ankitrwt, 2017-09-26:
yes my all iphones are in recory mode but i started restoring thm in normal mode
@nikias

ankitrwt, 2017-09-26:
and yes i am using all latest codes from github.com/libimobiledevice

LegitSoulja, 2017-09-26:
I need help aswell. I am running futurerestore on Ubuntu but failed to connect to gp.apple.com socket.

I seen that I need to install iTunes however I am on Ubuntu. Any way around this? Thanks

```
$ ./futurerestore_linux -t blob.shsh2 -b Mav10-5.24.00.Release.bbfw -p BuildManifest.plist -s sep-firmware.n61.RELEASE.im4p -m BuildManifest.plist iPhone_4.7_10.0.2_14A456_Restore.ipsw
Version: b99eb8140d8e6c23f34e950102bb79e61c72384d - 152
Libipatcher Version: a6cc5cfa8d421987d06f6157130ef3c75391efa4 - 39
Odysseus Support: yes
[INFO] 64bit device detected
futurerestore init done
reading ticket blob.shsh2 done
[TSSC] opening BuildManifest.plist
WARNING: Unable to find BbSkeyId node
[TSSR] User specified not to request a Baseband ticket.
Request URL set to https://gs.apple.com/TSS/controller?action=2
TSS server returned: <html>
<head><title>403 Forbidden</title></head>
<body bgcolor="white">
<center><h1>403 Forbidden</h1></center>
<hr><center>Shield</center>
</body>
</html>

Sending TSS request attempt 1... Request URL set to https://17.171.36.30/TSS/controller?action=2
Failed to connect to 17.171.36.30 port 443: Connection timed out
Sending TSS request attempt 2... Request URL set to https://17.151.36.30/TSS/controller?action=2

```

ankitrwt, 2017-09-27:
@LegitSoulja 
TSS was not working on Monday, its working fine now...

now its stuck on 
FDR 0x3475xxx timeout waiting for command
FDR 0x3475xxx waiting for message... 
No data to read 

and shows error message
ERROR: unable to receive message from FDR 0x3475xxx (-4). 0/2 bytes

any suggestions?
@nikias @LegitSoulja

LegitSoulja, 2017-09-27:
@ankitrwt i'll switch back over and run another test.

LegitSoulja, 2017-09-27:
@ankitrwt I got a different result this time

```
Version: b99eb8140d8e6c23f34e950102bb79e61c72384d - 152
Libipatcher Version: a6cc5cfa8d421987d06f6157130ef3c75391efa4 - 39
Odysseus Support: yes
[INFO] 64bit device detected
futurerestore init done
reading ticket blob.shsh2 done
[TSSC] opening BuildManifest.plist
WARNING: Unable to find BbSkeyId node
[TSSR] User specified not to request a Baseband ticket.
Request URL set to https://gs.apple.com/TSS/controller?action=2
TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.)
Sending TSS request attempt 1... [Error] sep firmware isn't signed
Failed with errorcode=-3
```

LegitSoulja, 2017-09-27:
Wait how can I check my blob and see what firmware it was on at the time?

nikias, 2017-09-27:
@LegitSoulja can you please keep futurerestore off this ticket and bug tracker? This is solely for idevicerestore. Thanks.

nikias, 2017-09-29:
The original issue with TSS is no longer a problem. Is the FDR thing still broken? Because for me, it is not.

nikias, 2017-09-29:
@ankitrwt while checking your log output I came across this error:
```
Unknown data request 'DeviceTree' received
```
Which is  proof that you are not using the latest code. This was fixed on June 8th with commit b99782b2. Please update your code.

nikias, 2017-11-10:
Closing. This issue was not caused by idevicerestore.

closed 2017-11-10 by nikias

mailinglists35, 2019-12-14:
if anyone hits this in 2019:

find ip of gs.apple.com (it varies according to where you are in the world, for me in europe was:)
````
$ host gs.apple.com
gs.apple.com is an alias for gs.apple.com.akadns.net.
gs.apple.com.akadns.net has address 17.137.163.1
````

put it in /etc/hosts

````
$ grep apple /etc/hosts
17.137.163.1 gs.apple.com
````

run idevicerestore again.

somehow idevicerestore insisted in connecting to a different IP than the one returned by dns

````
Request URL set to https://gs.apple.com/TSS/controller?action=2
Empty reply from server
Sending TSS request attempt 1... Request URL set to https://17.171.36.30/TSS/controller?action=2
Failed to connect to 17.171.36.30 port 443: Connection timed out
Sending TSS request attempt 2... Request URL set to https://17.151.36.30/TSS/controller?action=2
Failed to connect to 17.151.36.30 port 443: Connection timed out
Sending TSS request attempt 3... Request URL set to http://gs.apple.com/TSS/controller?action=2
Empty reply from server
Sending TSS request attempt 4... Request URL set to http://17.171.36.30/TSS/controller?action=2
Failed to connect to 17.171.36.30 port 80: Connection timed out
Sending TSS request attempt 5... Request URL set to http://17.151.36.30/TSS/controller?action=2
Failed to connect to 17.151.36.30 port 80: Connection timed out
Sending TSS request attempt 6... Request URL set to https://gs.apple.com/TSS/controller?action=2
Empty reply from server
Sending TSS request attempt 7... Request URL set to https://17.171.36.30/TSS/controller?action=2
Failed to connect to 17.171.36.30 port 443: Connection timed out
Sending TSS request attempt 8... Request URL set to https://17.151.36.30/TSS/controller?action=2
^C
````

after that,
````
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received

````

$ sudo -E dnf info idevicerestore
Last metadata expiration check: 1:35:57 ago on Sb 14 dec 2019 09:48:06 +0200.
Installed Packages
Name         : idevicerestore
Version      : 0.0
Release      : 1.gitc09c07b.fc31
Architecture : x86_64
Size         : 213 k
Source       : idevicerestore-0.0-1.gitc09c07b.fc31.src.rpm
Repository   : @System
From repo    : copr:copr.fedorainfracloud.org:r3pek:idevicerestore
Summary      : Tool to restore firmware files to iOS devices
URL          : https://github.com/libimobiledevice/idevicerestore.git
License      : LGPLv3
Description  : The idevicerestore tool allows to restore firmware files to iOS devices.
             : 
             : It is a full reimplementation of all granular steps which are performed during
             : restore of a firmware to a device.
             : 
             : In general, upgrades and downgrades are possible, however subject to
             : availability of SHSH blobs from Apple for signing the firmware files.

mailinglists35, 2019-12-14:
>somehow idevicerestore insisted in connecting to a different IP than the one returned by dns

apparently the build I have from fedora copr has two apple activation servers IP hardcoded:

$ strings /usr/bin/idevicerestore|grep -- '://17\.'
https://17.171.36.30/TSS/controller?action=2
https://17.151.36.30/TSS/controller?action=2
http://17.171.36.30/TSS/controller?action=2
http://17.151.36.30/TSS/controller?action=2

Per apple support document, they only mention `gs.apple.com` https://support.apple.com/en-us/HT201999

So instead of mangling the `gs.apple.com` hosts entry, just firewall the other two, and the dns resolution will take care of the rest:

````
Request URL set to https://gs.apple.com/TSS/controller?action=2
Empty reply from server
Sending TSS request attempt 1... Request URL set to https://17.171.36.30/TSS/controller?action=2
Failed to connect to 17.171.36.30 port 443: Connection refused
Sending TSS request attempt 2... Request URL set to https://17.151.36.30/TSS/controller?action=2
Failed to connect to 17.151.36.30 port 443: Connection refused
Sending TSS request attempt 3... Request URL set to http://gs.apple.com/TSS/controller?action=2
Empty reply from server
Sending TSS request attempt 4... Request URL set to http://17.171.36.30/TSS/controller?action=2
Failed to connect to 17.171.36.30 port 80: Connection refused
Sending TSS request attempt 5... Request URL set to http://17.151.36.30/TSS/controller?action=2
Failed to connect to 17.151.36.30 port 80: Connection refused
Sending TSS request attempt 6... Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 7... response successfully received
````

lalaRLH, 2024-04-23:
Still an issue in 2024. Thanks to the ignorant response to immediately mark this as not an issue with the code. Whilst this indeed worked fine for me in the United States, it fails in the UK as the domain record here points to a different IP. It’s not aligned to Apple guidelines to hardcode the IP, rather it’s very clear to resolve the host name. So I’m made up for whoever is sat in the United States but for the rest of us using this has been infinitely problematic for years. I have built and compiled entirely from source.

lalaRLH, 2024-04-23:
> if anyone hits this in 2019:
>
> find ip of gs.apple.com (it varies according to where you are in the world, for me in europe was:)
>
> ```
> $ host gs.apple.com
> gs.apple.com is an alias for [...]

Thank you so much.

nikias, 2024-04-23:
@lalaRLH I wouldn't say "ignorant", and also not "infinitely problematic for years" since I have been using this for years myself, mostly from Germany but also other non-US countries, and I know many people using this all over the globe have never had an issue with that specifically. Otherwise the issue tracker would explode with complaints... I am really sure the problem is by far only affecting a small amount of people.

The actual code to handle the actual connections for the request is performed by libcurl which is a separate library, and might be affected by some system settings like a proxy or similar things. So in general the statement that idevicerestore is not to blame here is true. Second, the hardcoded IP addresses that @mailinglists35 pointed out are not used as a first attempt, instead https://gs.apple.com is being used as the very first one, you can also see from the output:
```
Request URL set to https://gs.apple.com/TSS/controller?action=2
Empty reply from server
```
Why the server, which is the official one, is giving an empty reply, needs to be investigated to understand what is happening. The other addresses are supposed to act as a fallback, but looks like they are no longer valid. We might want to investigate and add a few more recent ones, just in case the DNS resolution doesn't work. But this problem will hit us again in the future probably, so maybe not the best choice.

I would suggest to add
```diff
diff --git a/src/tss.c b/src/tss.c
index 80591e7..904204d 100644
--- a/src/tss.c
+++ b/src/tss.c
@@ -1548,6 +1548,7 @@ plist_t tss_request_send(plist_t tss_request, const char* server_url_string)

                /* disable SSL verification to allow download from untrusted https locations */
                curl_easy_setopt(handle, CURLOPT_SSL_VERIFYPEER, 0);
+               curl_easy_setopt(handle, CURLOPT_VERBOSE, 1);

                curl_easy_setopt(handle, CURLOPT_ERRORBUFFER, curl_error_message);
                curl_easy_setopt(handle, CURLOPT_WRITEFUNCTION, (curl_write_callback)&tss_write_callback);
```
So we can see what is happening. Also, before fiddling with firewall or hosts files, there is a command line switch that can be used:
```
idevicerestore --server https://17.137.163.1/TSS/controller?action=2 IPSWFILE
```

lalaRLH, 2024-04-23:
All points noted and they’re valid, it’s something I only ever see when I’m in the UK. I’ve got a stack of devices to DFU and I find the debug output extremely helpful to identify whether a users Mac is doing anything odd, so thank you. It’s extra comfort to see step by step what’s occurring if that users had a threat notification. 

It absolutely works fine in the US. 

All of this being said the UK’s mass market ISPs all supply free equipment and some make it extremely difficult / often against the terms of service to use something else. They seemingly issue £300 routers wifi6 routers to all on internet plans around £25 a month. Seems pretty great until you observe DNS interception from hell. It sends most traffic to its usual local DNS but at times randomly picks traffic to forward instead to Trend Micro and/or Amazon cloud. 

DNS often hits a dead revolver in Manchester which then reroutes to Amsterdam after timeouts, via a partner network and back down from there into London, where it’s DNS is processed via the London internet exchange.

I apologise for jumping the gun on this. I’d be not in the least bit surprised if this wonderfully engineered setup was triggering concerns over the authenticity of data requests. 

https://community.virginmedia.com/t5/Networking-and-WiFi/Hub-5-Intercepting-all-DNS-queries/td-p/5378038 

Thanks for your suggestion I will give it a go.

nikias, 2024-04-23:
Also something I am working on currently is splitting out the TSS code to a separate library, so I am open to figuring out the best solution for everyone here :)

nikias, 2024-04-23:
I see, that makes a lot of sense now. I am not sure how to best handle this, but maintaining a list of official Apple IP addresses might be something. The ones hardcoded actually are not reachable for me at all and then there is
```
❯ host gs.apple.com
gs.apple.com is an alias for gs.apple.com.v.aaplimg.com.
gs.apple.com.v.aaplimg.com has address 17.111.103.15
```
Maybe I just need to stick with the official hostname and have people manually specify an IP if there are any problems...

lalaRLH, 2024-04-23:
I’m a fan of this site to identify the global footprint, probably as it looks kind of neat with the map. I can’t find it but there’s another one which also shows uptime overtime.

https://uptime.com/freetools/global-uptime-test/665769/gs.apple.com

If you really want detail, web check is pretty revealing https://web-check.xyz/

lalaRLH, 2024-04-23:
I find the Apple developer stuff logical in general and it applies often much more broadly. Some info here: https://developer.apple.com/documentation/technotes/tn3151-choosing-the-right-networking-api#DNS-alternatives 

note: I’m almost certain my issues are using ISP kit and working as an infosec manager everything I own is loaded with encryption. So from my side we’re good on this point, but always happy to talk networking! 😝