libimobiledevice/idevicerestore #188: iphone 5s recovery fail (extracting BuildManifest)

jun0s, 2017-10-03:
Linux dbox 4.13.4 #2 SMP Sun Oct 1 09:39:15 MDT 2017 i686 GNU/Linux
Debian 32bits
libzip 1.3.0

Latest firmware is iPhone_4.0_64bit_11.0.1_15A402_Restore.ipsw
Verifying 'iPhone_4.0_64bit_11.0.1_15A402_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
ERROR: zip_open: iPhone_4.0_64bit_11.0.1_15A402_Restore.ipsw: 5
ERROR: zip_open: iPhone_4.0_64bit_11.0.1_15A402_Restore.ipsw: 5
ERROR: Invalid archive
ERROR: Unable to extract BuildManifest from iPhone_4.0_64bit_11.0.1_15A402_Restore.ipsw. Firmware file might be corrupt.

can anybody can help me? thanks!!

nikias, 2017-10-03:
Seems like an issue with libzip even though you seem to have a really new version. Error code 5 is a 'read error'. Can you make sure that Large File Support has been selected? You can check config.log if it says ```LFS_CFLAGS=' -D_LARGE_FILES=1'```
This is no guarantee that it works but it might be the issue if this is missing.

jun0s, 2017-10-04:
Thanks for your time @nikias..... No, and i can´t understand why being the last version of libzip the support for large files needs to be activated, i assumed that it is, since 2013... i find just one option that disable the large files support... with "configure" --disable-largefiles, so "large files" support its the default option... maybe its gcc problem? (gcc (Debian 6.3.0-18) 6.3.0 20170516) the support in kernel for large files its ok, i open the iphone file with anothers tools, i use ziptool with 500M files without problem and then with 2.6 gigas iphone file (ziptool iphonefile stat 1) with this result.. 

can't stat iPhone_4.0_64bit_11.0.1_15A402_Restore.ipsw: Value too large for defined data type.

can i force the "large files" parameter to the Makefile file? what do you think? 

thank you very much.. and i am sorry for my very bad english. :(

jun0s, 2017-10-05:
Well this is a libzip problem with 32bits systems, i solved adding -D_FILE_OFFSET_BITS=64 to the compiler flags, i close this thank you.

closed 2017-10-05 by jun0s

nikias, 2017-10-05:
Compiler flags are for libzip, right? Just to clarify for other who run into this problem.

jun0s, 2017-10-05:
sorry... YES, the compilers flags are for any version of libzip.

jun0s, 2017-10-06:
Hi @nikias ... what about -x option, still works? because i am trying a iphone 5s recovery with the baseband faulty, using the "-x iphonefile" it´s possible works?

nikias, 2017-10-06:
No this doesn't work for newer devices and firmware.