libimobiledevice/idevicerestore #189: DFU mode restore of iPhone 8 fails

pekn, 2017-10-04:
DFU mode restore of iPhone 8 fails

`$ sudo ./idevicerestore -e -l
NOTE: using cached version data
Found device in DFU mode
Identified device as d201ap, iPhone10,4
Latest firmware is iPhone10,4_11.0.2_15A421_Restore.ipsw
Verifying 'iPhone10,4_11.0.2_15A421_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 11.0.2
Product Build: 15A421 Major: 15
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Found ECID 2653710371139502
Getting ApNonce in dfu mode... 03 1f aa d7 92 7c 69 c5 48 24 59 5a 27 2c 7b e9 c0 da 93 04 7a 86 7b 1d 4a 0a fd 13 f6 2c 69 73 
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode... ff c5 e6 16 af 2b 29 8b 72 2b b7 5a 5d 14 f6 89 c6 23 ad 75 
WARNING: Unable to find BbSkeyId node
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Using cached filesystem from 'iPhone10,4_11.0.2_15A421_Restore/058-55270-396.dmg'
Extracting iBSS.d20.RELEASE.im4p...
Personalizing IMG4 component iBSS...
Sending iBSS (831476 bytes)...
[==================================================]  99.8%ERROR: Unable to send iBSS component: Unable to upload data to device
ERROR: Unable to send iBSS to device
ERROR: Unable to place device into recovery mode from DFU mode`

It always seems to fail during sending of iBSS.. percentage varies (e.g. 99.8 or 25.1).

nikias, 2017-10-04:
Maybe Apple changed something in the DFU transfer protocol for those devices, weird though if they use the same USB ID and not a new one. I will try to figure this out, I don't have a device though to test it.

nikias, 2017-10-04:
Hmm also maybe the component personalization doesn't work properly. Can you run it with -k parameter? This will keep the personalized files. Then upload the file iBSS somewhere or email it to me (my email address is in the source header files).

LeThuyVFS, 2017-10-05:
@nikias 
Give yourself your email address.

pekn, 2017-10-05:
Here's iBSS: https://www.dropbox.com/s/vw6vcz21b8sla42/iBSS?dl=0

rekyagis, 2017-10-05:
@pekn How can I put iPhone 8 into DFU mode?
I will try to test with my device but I can not put it into DFU mode. (Pressing power and volume-down buttons will make it Recovery mode.)

pekn, 2017-10-05:
@rekyagis Same way as iPhone 7/7Plus, i.e. first turn off power from your iPhone, then
press volume down and power key for 8-10 seconds, then release power key but continue pressing volume down for 5 seconds. (You may need to try a couple of times in order to succeed.. you can use lsusb command to check if phone entered to DFU mode)

rekyagis, 2017-10-05:
@pekn Thanks, I have succeeded to put my device to DFU mode.
I have tested with my iPhone 8 Plus (A1898/iPhone10,2) and it is succeeded. The log is attached.
Is the issue only with iPhone 8 (not Plus) or only with iPhone10,4/10,5?
[iPhone8Plus.log](https://github.com/libimobiledevice/idevicerestore/files/1358996/iPhone8Plus.log)

nikias, 2017-10-05:
Interestingly, it would be the same for both devices; since DFU restore worked with an 8 Plus it should also just work with the 8.
@pekn what platform/system are you running it on?
@rekyagis also what's yours?
Maybe it is something else, like libzip failing to uncompress iBSS (I didn't have a chance yet to look into the file uploaded by @pekn)

rekyagis, 2017-10-06:
I am on "Ubuntu 16.04.3 LTS (GNU/Linux 4.4.0-96-generic x86_64)".
idevicerestore and libirecovery are from git repository. Other libraries are the original one on the system.

pekn, 2017-10-06:
I am on Ubuntu 16.04 (with latest updates).

nikias, 2017-10-06:
Just for completeness: I restored my iPad mini 4 from DFU without any problems, on macOS Sierra (libirecovery with IOKit support). 
@pekn The only reason why I it wouldn't work is if you'd had a really old libusb or something but you both should have the same default ubuntu one. Not really sure what's happening. Maybe your USB chipset is the reason? You can see with lsusb what kind of USB controllers you have @rekyagis @pekn. Other than that I am out of ideas here...

rekyagis, 2017-10-06:
My PC is DELL INSPIRON 1501 (released about 10 years ago, with AMD Turion 64 CPU). The information of its USB Host Controller (from dmesg) is attached.
[dmesg-usb.txt](https://github.com/libimobiledevice/idevicerestore/files/1362838/dmesg-usb.txt)
I also succeeded on Ubuntu16.04 running over MacBook with VMWare Fusion.

4kzeng, 2017-10-10:
iTunes also cannot restore my iPhone8 from DFU yet, iTunes's bug??
Now,  the only one thing has meaning is to exit the "White apple"  boot Loop, it can stay in 
Recovery  after a DFU restore.

daniel-bluesea, 2017-10-22:
same problem as @pekn describes, iphone8 is failing to be restored from dfu mode while iphone8 plus is successfully done.

@4kzeng @nikias iTunes error: The iPhone "#####" could not be restored. An unknown error occurred (2005).

@pekn @nikias maybe its a windows issue?

daniel-bluesea, 2017-10-22:
@4kzeng i managed to restore from dfu using iTunes. please check if its not a network issue (e.g. firewall and similar)

daniel-bluesea, 2017-10-22:
@nikias it seems that idevicerestore (via libirecovery) fail to upload the iboot (iBSS) to the device..

nikias, 2017-10-22:
@daniel-bluesea the error you mentioned is not an actual error. It's a debug statement.
What looks more like an error to me is this:
```
Setting to interface 0:0
WARNING: set interface failed, error -8
```
Not sure why this would happen though.

daniel-bluesea, 2017-10-22:
@nikias and its happening only with iphone 8 (10,4 ipsw).. around "irecv_send_buffer" function in the libirecovery..

does the interface error/warning imply something to the usb interface between the device and my pc?

do you need more debug information from me?
do you have some tips how can i further debug the device or the restore from dfu operation?
do you manage to reproduce it yourself with an iphone8? :)

daniel-bluesea, 2017-10-22:
@nikias i thought maybe it could be a windows issue (plenty of #ifndef WIN32 at that part of the code) , but @pekn  reported that it happened on its ubuntu machine..

daniel-bluesea, 2017-10-23:
@pekn any new insights or is it still an issue for you?

daniel-bluesea, 2017-10-23:
@nikias btw, the "WARNING: set interface failed, error -8" also occurs when restoring from dfu iphone 8 plus, but it doesnt prevent the uploading of the iBSS and to go into recovery mode (and then successful restore)..

daniel-bluesea, 2017-10-23:
@nikias in dfu mode the packet size is 2048 bytes, the iphone 8 iBSS file size is 831476 bytes which is 405 packets + last packet with size 2036, but to the last packet the code is adding 16 bytes with dfu_xbuf and some more data and total size that is sent to the device is 2052 (2036 + 16) which is more then 2048.
the "irecv_usb_control_transfer" function return -1 for this last packet..

could be the problem?

daniel-bluesea, 2017-10-23:
@nikias fixed! the last packet + 16 is the reason iphone 8 is failing to restore from dfu

Dantee296, 2017-10-23:
@daniel-bluesea share the patch files , where u made the changes in code ?

Dantee296, 2017-10-23:
Done sending NORData
ERROR: Unable to recieve message from FDR..

nikias, 2017-10-23:
@daniel-bluesea ohhhh I see. I didn't check the size of the iBSS so I didn't realize this. Good catch! So you already have a working patch?

4kzeng, 2017-10-24:
use daniel-bluesea 's   method, it works! 
if (i+1 == packets) {
    //libirecovery.c line 1783
    if( size + 16 > packet_size ){
	bytes = bytes = irecv_usb_control_transfer(client, 0x21, 1, i, 0, &buffer[i * packet_size], size, USB_TIMEOUT);
	size = 0;
    }
     for (j = 0; j < 2; j++) {
	dfu_hash_step(h1, dfu_xbuf[j*6 + 0]);
        ....

Dantee296, 2017-10-24:
@4kzeng can u please create a patch file for it

daniel-bluesea, 2017-10-24:
@nikias i did a local fix in my local code that is specific to the iphone 8 iBSS size.
for publishing a patch i think a more generic solution is needed.

nikias, 2017-10-24:
Guys, you can try this patch:
https://gist.github.com/nikias/94fbec21886ea3b0512c1e4fa013e403
I actually tried to _always_ split the last packet and the checksum buffer and it worked. Not sure if it's generally safe to assume this would work for all devices, so it would only split if the last packet is too large.

4kzeng, 2017-10-26:
iPhone8 and mini4 work for this patch.

nikias, 2017-10-26:
@4kzeng any reason why you did remove your comment? I put the fix in in the main repository now:
https://github.com/libimobiledevice/libirecovery/commit/b398a9ab735ea63ffd17d7c063503521460ed83f
Would be good to hear if it actually works for you @daniel-bluesea @Dantee296

daniel-bluesea, 2017-10-29:
@nikias sorry man, i was working with my local fix.. i'll check your fix and let you know asap

ChrisCheney, 2017-10-31:
With current git of libirecovery and idevicerestore my 8+ appears to get stuck when trying to RestoreKernelCache as shown below. I have tried several times and it appears to always get stuck at the same point. The iPhone shows the Apple logo and blank percentage bar ready to show loading...

...
Personalizing IMG4 component AppleLogo...
Sending AppleLogo (19656 bytes)...
ramdisk-size=0x20000000
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreRamDisk' in TSS response
NOTE: No path for component RestoreRamDisk in TSS, will fetch from build_identity
Extracting 058-84556-096.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (59136297 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreDeviceTree' in TSS response
NOTE: No path for component RestoreDeviceTree in TSS, will fetch from build_identity
Extracting DeviceTree.d21ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (173775 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreKernelCache' in TSS response
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.iphone10...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (16054293 bytes)...
About to restore device... 
Waiting for device...
Attempt 1 to connect to restore mode device...
Attempt 2 to connect to restore mode device...
Attempt 3 to connect to restore mode device...
Attempt 4 to connect to restore mode device...
...

nikias, 2017-10-31:
Question is: Does the device show the Apple logo when you try, and does it show a spinner below it later, and ideally a progress bar?

ChrisCheney, 2017-10-31:
It shows the Apple logo with the empty progress bar below it and eventually times out and reboots into restore mode with the iTunes logo, after trying ~ 100 or so times. I ended up restoring the phone with a spare Windows system. I can do further testing if wanted when I update my other iPhone 8+.

nikias, 2017-10-31:
In that case the device accepted the RestoreKernelCache. What platform did you try to restore? Linux? If so, maybe usbmuxd wasn't running so communication in restore mode wasn't possible.

pekn, 2017-11-03:
Tested with latest idevicerestore etc with iPhone 8. Still fails, but goes a bit farther.

```
NOTE: using cached version data
Found device in DFU mode
Identified device as d201ap, iPhone10,4
Latest firmware is iPhone10,4_11.1_15B93_Restore.ipsw
Verifying 'iPhone10,4_11.1_15B93_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 11.1
Product Build: 15B93 Major: 15
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Extracting filesystem from IPSW
[==================================================] 100.0%
Found ECID 2653710371139502
Getting ApNonce in dfu mode... 4d a5 93 84 2f cd ef 3c 5e 5c 0a 9c 83 8a b1 34 8f b1 f5 15 db ed 69 ce 94 27 8e c7 27 6f e1 4d 
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode... c6 6e ad 70 dd 4a 5a b0 79 29 61 22 93 34 f5 21 da 64 a3 d4 
WARNING: Unable to find BbSkeyId node
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBSS.d20.RELEASE.im4p...
Personalizing IMG4 component iBSS...
Sending iBSS (867941 bytes)...
[==================================================] 100.0%
ERROR: Unable to connect to device in DFU mode
ERROR: Unable to get ApNonce from device!
ERROR: Unable to place device into recovery mode from DFU mode
```

Device does not boot to recovery mode - instead it boots to normal mode.

nikias, 2017-11-05:
@daniel-bluesea I have no idea. Right now I am trying to make idevicerestore work with the iPhone X but after too many hours I figured out that it was the cable that made is cause to fail (even with iTunes!).
The cable from the box works (the actual restore process needs to be updated for the iPhone X though).

daniel-bluesea, 2017-11-06:
@nikias it seems that iphone8 & iphone8+ are suffering from inconsistent restoring success...
i'm using most updated projects and dlls.
lets wait and see if its "global" issue or only mine :)
i dont want to delay your iPhone X coding

daniel-bluesea, 2017-12-03:
@nikias to continue our discussion regarding "iphone8 & iphone8+ are suffering from inconsistent restoring success..."

it is reproduced quite rarely but when it does, only iTunes can overcome this issue and successfully restore the device. the idevicerestre is repeatedly failing on "Got status message
Status: Verification Error"

whats this "Verification Error" issue?
is there a way to solve it/overcome it/workaround it?
do you want me to open new issue instead of continuing this one? :)

iphone8plus_restore.txt

nikias, 2017-12-03:
Open a new issue please. The log says something about an error regarding SE Firmware update...

daniel-bluesea, 2017-12-03:
i opened.