libimobiledevice/idevicerestore #192: DEVICE WITH NAND-CHIP ISSUE IS STUCK AT "FIRMWARE FLASHING"

Dantee296, 2017-10-19:
its stuck at  "FIRMWARE FLASHING" 
want to atleast tell if fail or pass it stuck there

@nikias  what to do where to change to know that it has a nand chip issue and firmware flashing wont work ?

need to add timeout ?
Please Help @nikias

Dantee296, 2017-10-24:
@nikias ??

nikias, 2017-10-24:
This is not enough information to comment on it. If it is "stuck" on that step ("Flashing Firmware" I suppose) then idevicerestore is waiting for feedback from the device. Nothing much we can do here since the actual restore process is handled by the device itself. As you mentioned NAND-chip issue this is probably the reason it will not succeed.

Dantee296, 2017-10-25:
ok Thanks

closed 2017-11-05 by Dantee296