libimobiledevice/idevicerestore #193: iPhone 6 IOS 11 Can't restore

eklass, 2017-10-21:
Hey guys,

You do a great job in developing this software. Thanks for that!
Like https://github.com/libimobiledevice/idevicerestore/issues/184 I cant restore my iPhone 6.

Is there something required to restore with an iPhone 6?

Log:
Reading data from version.xml
NOTE: using cached version data
opening device 05ac:1281 @ 0x14122000...
Setting to configuration 1
Setting to interface 0:0
Found device in Recovery mode
opening device 05ac:1281 @ 0x14122000...
Setting to configuration 1
Setting to interface 0:0
Identified device as n61ap, iPhone7,2
Extracting BuildManifest from IPSW
Product Version: 11.0.2
Product Build: 15A421 Major: 15
opening device 05ac:1281 @ 0x14122000...
Setting to configuration 1
Setting to interface 0:0
INFO: device serial number is xxx
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Using cached filesystem from '/Users/xxx/Downloads/iPhone_4.7_11.0.2_15A421_Restore/058-45615-483.dmg'
Getting device's ECID for TSS request
Found ECID xxx
Getting ApNonce in recovery mode... 36 94 08 de 22 00 23 c0 b5 fa 66 17 1d 83 7e 6e f3 a8 ee c7 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... b8 a8 63 c7 78 71 33 47 ea 31 c3 20 ab 73 2c aa bf 36 50 c4 
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
WARNING: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry AppleLogo
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging0
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging1
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryFull
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow0
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow1
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryPlugin
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry DeviceTree
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry KernelCache
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry LLB
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry OS
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
[... 662 lines cut ...]
Checking filesystems (15)
No data to read
Checking filesystems (15)
About to send FDR Trust data...
Sending FDR Trust data now...
Done sending FDR Trust Data
FDR 0x7f93c6fb5b20 timeout waiting for command
FDR 0x7f93c6fb5b20 waiting for message...
No data to read
No data to read
FDR 0x7f93c6fb5b20 timeout waiting for command
FDR 0x7f93c6fb5b20 waiting for message...
No data to read
No data to read
FDR 0x7f93c6fb5b20 timeout waiting for command
FDR 0x7f93c6fb5b20 waiting for message...
No data to read
No data to read
FDR 0x7f93c6fb5b20 timeout waiting for command
FDR 0x7f93c6fb5b20 waiting for message...
No data to read
No data to read

Is there a way/solution, that i can restore my iphone?
Restoring with 3utools were also not successfull.

jcschoen, 2017-10-31:
Not sure if you figured your issue out, but I was able to upgrade my iPhone 6, which was running iOS 11.0.2, and I'm not sure if this will help if you want to wipe you phone, but maybe someone else will find it useful. First, I used the command `idevicerestore -u [my uuid] -l` which put my phone into recovery mode, but did not restore. With my phone still in recovery mode, I ran `lsusb` to check if it was recognized by my OS, which it was. Then, I ran `idevicerestore -l` which ended up working. Best of luck!

Dantee296, 2017-11-05:
@eklass i believe its a NAND Chip issue