libimobiledevice/idevicerestore #196: FirmwareData of Savage

rekyagis, 2017-11-08:
Thank you for your great job, supporting Savage firmware.

I found that FirmwareData of Savage is different from the one with iTunes.
Please see the attachment:
[idevicerestore_FirmwareResponse_Savage.plist.txt](https://github.com/libimobiledevice/idevicerestore/files/1453202/idevicerestore_FirmwareResponse_Savage.plist.txt)
 -> FirmwareResponse for Savage by idevicerestore.
[itunes_FirmwareResponse_Savage.plist.txt](https://github.com/libimobiledevice/idevicerestore/files/1453207/itunes_FirmwareResponse_Savage.plist.txt)
 -> FirmwareResponse for Savage by iTunes. (usb monitoring)

idevicerestore uses Savage/Savage.B2-Prod.fw for it.
iTunes uses different data for it. I can not find any file with the same data in IPSW.

I am using the same firmware (iPhone10,6_11.1_15B93_Restore.ipsw) and device for idevicerestore and iTunes.
I have tried several times with iTunes (both windows and mac), the FirmwareData of Savage is not changed. iTunes always uses the same data for it.

The FirmwareData of Savage is different, but idevicerestore succeed to restore.

Do you have any idea?

nikias, 2017-11-10:
I will take a look. This is indeed weird.

nikias, 2017-11-10:
So the data is identical, except that iTunes is adding 16 bytes at the beginning:
```
00 00 00 00 32 05 00 00  00 00 00 00 00 00 00 00
```

nikias, 2017-11-11:
Seems like FaceID doesn't work on the phone I restored. The additional data header might be the reason. Will try later if it works

rekyagis, 2017-11-14:
@nikias Thank you for your commit for this. It works well. The FirmwareData for Savage is now the same as iTune's.

Please let me ask a question.
Why should we use "Savage.B2-Prod.fw" from 4 candidate files in IPSW: Savage.B0-Prod.fw, Savage.B0-Prod.vt.fw, Savage.B2-Prod.fw, and Savage.B2-Prod.vt.fw? (*.vt.fw is empty though)
Do you have any idea what the other files are?

nikias, 2017-11-14:
I don't know exactly, but iTunes used the B2. I couldn't figure out from MobileDevice framework why it choses B2 over B0, nor any hint about it in the build identity (BuildManifest.plist) maybe I will look again and try to understand.

nikias, 2017-11-14:
Fixed with commit https://github.com/libimobiledevice/idevicerestore/commit/08d610d5811ed0aa3fecf48ff9e9cee2190b1981

closed 2017-11-14 by nikias