libimobiledevice/idevicerestore #197: iPhone 8 Plus failed to restore

jalcock0, 2017-11-09:
I am using the latest version of idevicerestore and its failed to restore on the iPhone 8 Plus device.I can see the progress bar moving more than 50 % and then disconnects. I have attached the logs.
[iPhone8plus_failedtorestore.txt](https://github.com/libimobiledevice/idevicerestore/files/1458115/iPhone8plus_failedtorestore.txt)

rekyagis, 2017-11-10:
Please try the attached patch.
[idevicerestore_tss_development_update_payload_hash.patch.txt](https://github.com/libimobiledevice/idevicerestore/files/1460808/idevicerestore_tss_development_update_payload_hash.patch.txt)

"DevelopmentUpdatePayloadHash" key of SE,UpdatePayload"  in the tss request  should be removed.

The patch treats "DevelopmentUpdatePayloadHash" (for "SE,UpdatePayload") the same way as "DevelopmentCMAC" (for "SE,Firmware").
I am not sure "SE,IsDev" is effective even for "'SE,UpdatePayload", but this will work well for production mode.

4kzeng, 2017-11-10:
Good job, @rekyagis, My iPhone 8 have this problem after doing OTA UPDATE.  I removed "DevelopmentUpdatePayloadHash",  now,  it worked again,  i am very happy to see the "100%" and "DONE".

nikias, 2017-11-10:
@rekyagis this seems good to me. What's your real name and github email address so I can commit in your name? You can send it to me privately. See commit log for my email address.

jalcock0, 2017-11-11:
Thank you very much @rekyagis . I applied the patch and iPhone 8 plus restored successfully with the iOS version 11.1.1.

zbalaton, 2017-11-13:
How about this instead:
https://github.com/libimobiledevice/idevicerestore/pull/200/commits/9e3a1348a93637b1f70fa53dc0a1ddb9e145246c
It's a bit more complex but should be safer than guessing by what's in the buildID.

Kanong017, 2017-11-27:
The original USB line must be used

nikias, 2017-11-27:
@zbalaton I prefer your solution, but instead of adding a new parameter to tss_request_add_se_tags I would just get the SE,ChipID again from parameters.

nikias, 2018-04-27:
This should be working now with latest commit b8b3035500eb2311aab366aaa3e5069ee422e8e3.

closed 2018-04-27 by nikias