libimobiledevice/idevicerestore #199: TSS Request Failing

rdg10, 2017-11-11:
Hi.  Everything had been working fine but with the new IOS versions and the latest code I'm getting TSS Request Failures.  I've attached the debug info.  Any help would be greatly appreciated.

[c69b0516082ad3ca7b68c84411923d677c1a1b77.txt](https://github.com/libimobiledevice/idevicerestore/files/1463947/c69b0516082ad3ca7b68c84411923d677c1a1b77.txt)

nikias, 2017-11-11:
Can you check if it works with the patch attached here? https://github.com/libimobiledevice/idevicerestore/issues/197

rdg10, 2017-11-12:
I'm running on windows so I don't have diff.  Do you have source code to the changed functions instead of the diff patch you could post?

nikias, 2017-11-12:
Just check the patch. It is just adding 5 lines to the source code (the lines starting with +)

rdg10, 2017-11-12:
I added the lines and same issue...

Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... TSS server returned: STATUS=100&MESSAGE=An internal error occurred.
ERROR: TSS request failed (status=100, message=An internal error occurred.)
ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device

rdg10, 2017-11-13:
Any ideas?

rekyagis, 2017-11-14:
The issue seems in first TSS request, not in TSS for "SE,UpdatePayload". So, unfortunately, the patch attached #197 won't work.

The plist of TSS request in the log file seems bad format.
You can see the request from line 88 of "c69b0516082ad3ca7b68c84411923d677c1a1b77.txt", and the issue is at line 405-407.
> 
	&lt;key&gt;BbChipID&lt;/key&gt;
	&lt;key&gt;BbProvisioningManifestKeyHash&lt;/key&gt;
	&lt;key&gt;BbProvisioningManifestKeyHash&lt;/key&gt;

There is no value for "BbChipID", and the key "BbProvisioningManifestKeyHash" is duplicated.
TSS server may return "An internal error occurred." if this bad formatted plist is requested.

I have no idea why the plist is wrong. Is there some issue in libplist?

nikias, 2017-11-14:
@rekyagis oh good point. It looks like libplist is broken...

rdg10, 2017-11-15:
After updating libplist and libzip/zlib it all seems to be working now.  Thank you.

closed 2017-11-15 by rdg10

izaman1, 2019-10-14:
> Just check the patch. It is just adding 5 lines to the source code (the lines starting with +)

can you help me how to add that patch in linux

nikias, 2019-10-15:
The patch is not required anymore, it is already applied in the latest code from git.