libimobiledevice/idevicerestore #201: ERROR: tss_send_request: Unhandled status code 132 iphone 7 (global)

henosvk, 2017-11-21:
any idea how to fix this ? 

Odysseus Support: yes
[INFO] 64bit device detected
futurerestore init done
reading ticket /Users/henry/Desktop/downgrade/8287423015092282_iPhone9,1_10.2-14C92.shsh2 done
[TSSC] opening /Users/henry/Desktop/downgrade/BuildManifest.plist
WARNING: Unable to find BbSkeyId node
[TSSR] User specified not to request a Baseband ticket.
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Did set sep+baseband path and firmware
[TSSC] opening /Users/henry/Desktop/downgrade/BuildManifest.plist
WARNING: Unable to find BbSkeyId node
[TSSR] User specified to request only a Baseband ticket.
ERROR: Unable to get BasebandFirmware node
ERROR: Unable to find required BbGoldCertId in parameters
Request URL set to https://gs.apple.com/TSS/controller?action=2
TSS server returned: STATUS=132&MESSAGE=An internal error occurred.
ERROR: tss_send_request: Unhandled status code 132
Sending TSS request attempt 1... Request URL set to https://17.171.36.30/TSS/controller?action=2
ERROR: tss_send_request: Unhandled status code 132
Sending TSS request attempt 2... Request URL set to https://17.151.36.30/TSS/controller?action=2
ERROR: tss_send_request: Unhandled status code 132
Sending TSS request attempt 3... Request URL set to http://gs.apple.com/TSS/controller?action=2
TSS server returned: STATUS=132&MESSAGE=An internal error occurred.
ERROR: tss_send_request: Unhandled status code 132
Sending TSS request attempt 4... Request URL set to http://17.171.36.30/TSS/controller?action=2
ERROR: tss_send_request: Unhandled status code 132
Sending TSS request attempt 5... Request URL set to http://17.151.36.30/TSS/controller?action=2

gair, 2017-11-23:
Maybe this?
https://github.com/libimobiledevice/idevicerestore/issues/199

henosvk, 2017-11-24:
nope im trying to downgrade from ios 11 to 10 maybe thats the prob ?

gair, 2017-11-24:
Yeah, you can't do that. Apple isn't signing 10 anymore.

henosvk, 2017-11-24:
well yea I know but I have saved blobs

henosvk, 2017-11-24:
yup I made a mistake when saving blobs .. I saved for iphone 9,1 and found out that I have ip 9,3 .. ill slap myself ..

nikias, 2017-11-25:
Even if you saved the correct blobs you would have no way to use them. There is no downgrade attack for the iPhone 7.

closed 2017-11-25 by nikias