libimobiledevice/idevicerestore #202: repeatedly failed restore - iphone8

daniel-bluesea, 2017-12-03:
@nikias this is a rare phenomena but when it happens only iTunes can successfully restore the device while idevicerestore is repeatedly failing to do so.

is usually happens (when it happens) with devices in recovery mode (you can see the previous restore.log in the file attached starting at: "Previous Restore Log Received:"), when the restore process receives from the device a status message with a verification error:

Done sending FirmwareUpdater data
Got status message
Status: Verification Error
Log is available:
...

the log is also available in the attached file.

according to the code this error is crucial and can not be overcome..
whats the nature of this error? what can be the cause of that? (iTunes version??)

iphone8plus_restore.txt

daniel-bluesea, 2017-12-03:
@nikias do you think it might be related to: https://github.com/libimobiledevice/idevicerestore/issues/197 ?

daniel-bluesea, 2017-12-04:
after applying patch from #197 everything was fine for a while and then suddenly i got the folowing error:

Got status message
Status: Failed to recover FDR data.

from extra debug info:
_AMSupportHttpURLSessionSendSyncNoRetry: HTTP request failed with error Error Domain=NSURLErrorDomain Code=-999 "cancelled" UserInfo={NSErrorFailingURLStringKey=https://gg.apple.com/ca/authorize, NSLocalizedDescription=cancelled, NSErrorFailingURLKey=https://gg.apple.com/ca/authorize}
AMSupportHttpURLSessionSendSync: SSL trust evaluation failed
AMFDRCreateError: _AMFDRHttpMessageSendSync: SSL trust evaluation failed: code=22
AMFDRCreateError: _AMFDRHttpRequestSendSyncNoRetry: _AMFDRHttpMessageSendSync failed: code=8
AMFDRCreateError: _AMFDRPermissionsRequestWithString: AMFDRPermissionsRequest failed: code=8
AMFDRCreateError: _AMFDRPermissionsRequest: Unexpected permission request error: code=12
AMFDRCreateError: AMFDRDataHTTPLoadPersistent: Permissions request failed.: code=12
AMFDRCreateError: AMFDRDataHTTPCopy: AMFDRDataHTTPLoadPersistent failed: code=8
AMFDRCreateError: AMFDRDataCopy: seal:00008015-00152D1A2040E92E get data failed - foundData is NULL: code=10
AMFDRCreateError: AMFDRSealedDataPopulate: Unexpected error for seal:00008015-00152D1A2040E92E: 22: code=8
AMFDRCreateError: AMFDRDataErrorIsBenign: AMFDRSealedDataPopulate failed to report what data is missing: code=10
AMFDRCreateError: AMFDRDataRecover: AMFDRSealedDataPopulate with amfdrRemote failed with a fatal error: code=10
AMFDRDataRecoverCurrentDevice(sealData=false) failed, errCode=19
0: AMFDRError/13
1: AMFDRError/a: AMFDRSealedDataPopulate with amfdrRemote failed with a fatal error
2: AMFDRError/a: AMFDRSealedDataPopulate failed to report what data is missing
3: AMFDRError/8: Unexpected error for seal:00008015-00152D1A2040E92E: 22
4: AMFDRError/a: seal:00008015-00152D1A2040E92E get data failed - foundData is NULL
5: AMFDRError/8: AMFDRDataHTTPLoadPersistent failed
6: AMFDRError/c: Permissions request failed.
7: AMFDRError/c: Unexpected permission request error
8: AMFDRError/8: AMFDRPermissionsRequest failed
9: AMFDRError/8: _AMFDRHttpMessageSendSync failed
10: AMFDRError/16: SSL trust evaluation failed
[10:08:27.0774-GMT]{5>8} CHECKPOINT FAILURE:(FAILURE:53) RESTORED:[0x0634] fdr_recover foundData is NULL)[7]D(AMFDRDataHTTPLoadPersistent failed)[8]D(Permissions request failed.)[9]D(Unexpected permission request error)[10]D(AMFDRPermissionsRequest failed)[11]D(_AMFDRHttpMessageSendSync failed)[12]D(SSL trust evaluation failed)
restore-step-results = {0x11070634:{0:53}}
restore-step-codes = {0x11070634:{0:53}}
restore-step-domains = {0x11070634:{0:"AMRestoreErrorDomain"}}
restore-step-error = {0x11070634:"foundData is NULL)[7]D(AMFDRDataHTTPLoadPersistent failed)[8]D(Permissions request failed.)[9]D(Unexpected permission request error)[10]D(AMFDRPermissionsRequest failed)[11]D(_AMFDRHttpMessageSendSync failed)[12]D(SSL trust evaluation failed)"}
restore-step-uptime = 103
restore-step-user-progress = 52
[10:08:27.0779-GMT]{5>8} CHECKPOINT NOTICE: (NVRAM set) restore-step-user-progress=52 [sync=true] (first failure)
[10:08:27.0779-GMT]{5>8} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x11030634:59;0x11030647:60}
restore-step-names = {0x11030634:fdr_recover;0x11030647:cleanup_check_result}
restore-step-uptime = 103
restore-step-user-progress = 52
[10:08:27.0782-GMT]{5>8} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x11030634:59}
restore-step-names = {0x11030634:fdr_recover}
restore-step-uptime = 103
restore-step-user-progress = 52
[10:08:27.0785-GMT]{5>8} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x11030634:59;0x11030648:61}
restore-step-names = {0x11030634:fdr_recover;0x11030648:cleanup_send_final_status}
restore-step-uptime = 103
restore-step-user-progress = 52

ERROR: Unable to successfully restore device
FDR 036AFA40 got payload of 45 bytes, now try to proxy it
Sending 45 bytes of data
FDR 036AFA40 Received 51 bytes reply data, sending to device
No data to read
FDR 036ADC58 timeout waiting for command
FDR 036ADC58 terminating...
ERROR: Unable to restore device

nikias, 2017-12-12:
Something is wrong with the SE firmware update it seems... But I can't really see what...
```
SE: [1509627132.740][ERR] SEUpdaterExecCommand: [1509627132.562][ERR] assertSW: Fail to send Imaeg4 Apdu : 0x6f85 8
update_generic_firmware: Operation failed: Error Domain=RamrodErrorDomain Code=1017 "update_generic_firmware: failed to execute performNextStage on SE updater" UserInfo={NSDescription=update_generic_firmware: failed to execute performNextStage on SE updater, NSUnderlyingError=0x106813c20 {Error Domain=SEUpdaterErrorDomain Code=8 "[1509627132.740][ERR] SEUpdaterExecCommand: [1509627132.562][ERR] assertSW: Fail to send Imaeg4 Apdu : 0x6f85 8" UserInfo={NSLocalizedDescription=[1509627132.740][ERR] SEUpdaterExecCommand: [1509627132.562][ERR] assertSW: Fail to send Imaeg4 Apdu : 0x6f85 8}}}
update_generic_firmware: Underlying error: Error Domain=SEUpdaterErrorDomain Code=8 "[1509627132.740][ERR] SEUpdaterExecCommand: [1509627132.562][ERR] assertSW: Fail to send Imaeg4 Apdu : 0x6f85 8" UserInfo={NSLocalizedDescription=[1509627132.740][ERR] SEUpdaterExecCommand: [1509627132.562][ERR] assertSW: Fail to send Imaeg4 Apdu : 0x6f85 8}
Update[SE] error - The operation couldn\U2019t be completed. (RamrodErrorDomain error 1017 - update_generic_firmware: failed to execute performNextStage on SE updater)
[12:52:12.0751-GMT]{5>8} CHECKPOINT FAILURE:(FAILURE:-1) FIRMWARE:[0x1309] update_se [0]D(update_generic_firmware: failed to execute performNextStage on SE updater)[1]LD([1509627132.740][ERR] SEUpdaterExecCommand: [1509627132.562][ERR] assertSW: Fail to send Imaeg4 Apdu : 0x6f85 8)
restore-step-results = {0x11071309:{0:-1}}
restore-step-codes = {0x11071309:{0:1017}}
restore-step-domains = {0x11071309:{0:"RamrodErrorDomain"}}
restore-step-error = {0x11071309:"[0]D(update_generic_firmware: failed to execute performNextStage on SE updater)[1]LD([1509627132.740][ERR] SEUpdaterExecCommand: [1509627132.562][ERR] assertSW: Fail to send Imaeg4 Apdu : 0x6f85 8)"}
restore-step-uptime = 170
restore-step-user-progress = 85
[12:52:12.0755-GMT]{5>8} CHECKPOINT NOTICE: (NVRAM set) restore-step-user-progress=85 [sync=true] (first failure)
0: RamrodErrorDomain/3f9: update_generic_firmware: failed to execute performNextStage on SE updater
1: SEUpdaterErrorDomain/8
unable to convert ramrod error 1017
[12:52:12.0759-GMT]{5>8} CHECKPOINT FAILURE:(FAILURE:-1) RESTORED:[0x0635] update_device_firmware  firmware)[1]D(ramrod firmware update failed)[2]D(update_generic_firmware: failed to execute performNextStage on SE updater)[3]LD([1509627132.740][ERR] SEUpdaterExecCommand: [1509627132.562][ERR] assertSW: Fail to send Imaeg4 Apdu : 0x6f85 8)
restore-step-results = {0x11070635:{0:-1};0x11071309:{0:-1}}
restore-step-codes = {0x11070635:{0:-1};0x11071309:{0:1017}}
restore-step-domains = {0x11070635:{0:"AMRestoreErrorDomain"};0x11071309:{0:"RamrodErrorDomain"}}
restore-step-error = {0x11070635:" firmware)[1]D(ramrod firmware update failed)[2]D(update_generic_firmware: failed to execute performNextStage on SE updater)[3]LD([1509627132.740][ERR] SEUpdaterExecCommand: [1509627132.562][ERR] assertSW: Fail to send Imaeg4 Apdu : 0x6f85 8)"}
```

gsmdenis, 2017-12-31:
Get same problem:

Sending FirmwareResponse data now...
Done sending FirmwareUpdater data
Verification Error

nikias, 2018-04-27:
Maybe this was related to #197 and the latest commit b8b3035500eb2311aab366aaa3e5069ee422e8e3 fixed it?

nikias, 2019-01-26:
I think this issue has been resolved with all the commits that have been added in the meantime. Feel free to re-open if this is not the case.

closed 2019-01-26 by nikias