libimobiledevice/idevicerestore #203: Similar vulnerability to CVE-2016-5104

Young-X, 2017-12-08:
As CVE-2016-5104 mentioned, the socket_create function in common/socket.c in libimobiledevice and libusbmuxd allows remote attackers to bypass intended access restrictions and communicate with services on iOS devices by connecting to an IPv4 TCP socket. 
There is a similar function named socket_create in idevicerestore/src/socket.c that passes INADDR_ANY to htonl. I think this is also a vulnerability.
Please confirm this issue.

Patch for CVE-2016-5104
[](https://github.com/libimobiledevice/libimobiledevice/commit/df1f5c4d70d0c19ad40072f5246ca457e7f9849e)
[image]

[](https://github.com/libimobiledevice/libusbmuxd/commit/4397b3376dc4e4cb1c991d0aed61ce6482614196)
[image]

socket_create function in idevicerestore/src/socket.c:
[image]

nikias, 2017-12-12:
Neither ```socket_create``` nor ```socket_accept``` are used in the code, so this is not a vulnerability. However since it uses the same code it makes sense to change it accordingly.

nikias, 2017-12-12:
But certainly it doesn't make sense to assign a CVE for this. It's dead code.