libimobiledevice/idevicerestore #204: idevicerestore not finding the right ipsw file

jalcock0, 2017-12-12:
Hi,

the idevicerestore not finding the right ipsw file for iPhone 6s and trying to flash with the iPhone 8 Plus  firmware "iPhone_5.5_P3_11.0_11.2_15C114_Restore.ipsw.

nikias, 2017-12-12:
I don't know what you are doing exactly, but when I query for the iPhone 6s (iPhone8,1) I am getting this download link:
http://appldnld.apple.com/ios11.2/091-51435-20171206-BA105B14-D5F7-11E7-82AC-2BB9DC24E03F/iPhone_4.7_11.2_15C114_Restore.ipsw

nikias, 2017-12-12:
However I am planning to use the ipsw.me API to get the firmware information. I also have the idea to show all available (i.e. still being signed by Apple) firmwares with the user being able to select one.

jalcock0, 2017-12-12:
Please apologies nikias and its my mistake. Nothing wrong with the idevicerestore

closed 2017-12-12 by nikias