libimobiledevice/idevicerestore #207: Failed to restore iPad 2

Orbif, 2017-12-21:
Hello,

I have a crash of idevicerestore on Windows when I try to restore an iPad 2.

Attached the log before the crash.

[ipad2-crash.txt](https://github.com/libimobiledevice/idevicerestore/files/1579813/ipad2-crash.txt)

I don't have the problem when using an older version of libimobiledevice and idevicerestore (march 2017 version) 

Any idea why I have this problem with latest version ?

nikias, 2017-12-21:
That log doesn’t help because it has no information about the crash itself. Might be something in the code handling the baseband update. It should not be an issue in libimobiledevice. Any chance you can run it in a debugger?

Orbif, 2017-12-22:
I ran the command in SimpleProgramDebugger.

The report is attached here : 
[debugger-infos.txt](https://github.com/libimobiledevice/idevicerestore/files/1581916/debugger-infos.txt)