libimobiledevice/idevicerestore #208: Detecting wrong model of iPhone

securmk, 2017-12-31:
As title says iDeviceRestore is recognizing my iPhone as iphone5,1 but it's global model (iphone5,2). I tried to use a 5,1 firmware but it says that my device is not eligible for the update. It shows the following in console:

tim$ ./idevicerestore -t "custom.ipsw"
NOTE: using cached version data
Found device in DFU mode
Identified device as iPhone5,1
Extracting BuildManifest from IPSW
ERROR: Could not make sure this firmware is suitable for the current device. Refusing to continue.

I tried in pwned DFU, normal mode, recovery and normal dfu but won't work. What I can do?

julienkrijnen8, 2019-03-03:
Maybe reinstall idevicerestore?

circularsprojects, 2024-03-18:
im getting the same issue (like 5 years later.. i know)
im using an iphone 3G (1,2), but idevicerestore detects it as an iphone 2G (1,1)
```
# bin/macos/idevicerestore -ec "./iPhone1,2_4.2.1_8C148_CustomHJ.ipsw" 
Found device in WTF mode
Sending data (62602 bytes)...
[==================================================] 100.0%
Found ECID 0
Identified device as m68ap, iPhone1,1
Extracting Restore.plist from IPSW
ERROR: Could not make sure this firmware is suitable for the current device. Refusing to continue.
```
i know for a fact it is an iphone 3G, as it was on 4.2.1 before i restored it to 3.1.3 (and lukezgd's legacy ios kit detects it as an iphone 3G)