libimobiledevice/idevicerestore #211: /usr/bin/ld: idevicerestore-thread.o: undefined reference to symbol 'pthread_once@@GLIBC_2.2.5'

ChrisCheney, 2018-02-20:
Trying to build on Fedora 27 I found a few issues:

Need to add curl devel and bzip2 devel to requirements on the doc.

After adding those the following error occurred:

```
# PKG_CONFIG_PATH=/usr/local/lib/pkgconfig/ make
make  all-recursive
make[1]: Entering directory '/root/a/idevicerestore-master'
Making all in src
make[2]: Entering directory '/root/a/idevicerestore-master/src'
  CC       idevicerestore-idevicerestore.o
  CC       idevicerestore-common.o
  CC       idevicerestore-tss.o
  CC       idevicerestore-fls.o
  CC       idevicerestore-mbn.o
  CC       idevicerestore-img3.o
  CC       idevicerestore-img4.o
  CC       idevicerestore-ipsw.o
  CC       idevicerestore-normal.o
  CC       idevicerestore-dfu.o
  CC       idevicerestore-recovery.o
  CC       idevicerestore-restore.o
restore.c: In function ‘restore_device’:
restore.c:2151:24: warning: initialization makes integer from pointer without a cast [-Wint-conversion]
  thread_t fdr_thread = NULL;
                        ^~~~
  CC       idevicerestore-asr.o
  CC       idevicerestore-fdr.o
  CC       idevicerestore-limera1n.o
  CC       idevicerestore-download.o
  CC       idevicerestore-locking.o
  CC       idevicerestore-socket.o
  CC       idevicerestore-thread.o
thread.c: In function ‘thread_alive’:
thread.c:63:9: warning: implicit declaration of function ‘pthread_kill’; did you mean ‘pthread_self’? [-Wimplicit-function-declaration]
  return pthread_kill(thread, 0) == 0;
         ^~~~~~~~~~~~
         pthread_self
  CCLD     idevicerestore
idevicerestore-idevicerestore.o: In function `idevicerestore_start':
/root/a/idevicerestore-master/src/idevicerestore.c:632: warning: the use of `tempnam' is dangerous, better use `mkstemp'
/usr/bin/ld: idevicerestore-thread.o: undefined reference to symbol 'pthread_once@@GLIBC_2.2.5'
//usr/lib64/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:447: idevicerestore] Error 1
make[2]: Leaving directory '/root/a/idevicerestore-master/src'
make[1]: *** [Makefile:424: all-recursive] Error 1
make[1]: Leaving directory '/root/a/idevicerestore-master'
make: *** [Makefile:356: all] Error 2
```

M3taSapien, 2018-07-10:
Came across your post while I was looking into this issue myself. I was able to piece together similar issues at different stages of compilation from other forums and found the underlying cause for this error specifically was due to missing  bzip2-devel library. Fixed the issue by running:
`dnf install -y bzip2-devel-1.0.6-24.fc27.x86_64 bzip2-static-1.0.6-24.fc27.x86_64`

followed by rebuilding from src : 
**Solution:** Add'-lz' to makefile / GCC flags if not missing required compression libs (bzip, libcurl, etc) and rebuild.
**Background:** DSO is a dynamic shared object or a shared library. Adding -lz to the LDFLAGS to include location of your missing library at build time.

maenpaa24, 2018-08-12:
Hi! I am in arch. I think I have all the required dependencies installed. I have been looking for every package containing bzip and not being incompatible and installed them. When I do:
`
export LDFLAGS=$LDFLAGS:-lz`

then the autogen.sh promps an new error

```
checking for gcc... gcc
checking whether the C compiler works... no
configure: error: in `/src/apple/iphone/idevicerestore':
configure: error: C compiler cannot create executables
See `config.log' for more details
```
Do you have any idea?

mounaiban, 2019-01-25:
If anyone is still getting this problem, try the patch in https://github.com/libimobiledevice/idevicerestore/issues/220#issuecomment-412364421.

I ran into this problem on Fedora Workstation 29 and the patch worked for me. I'll test on F27 when I've got some more time to `kill`.

mounaiban, 2024-05-18:
I think this issue can be closed now. I managed to successfully compile 653349a, without any changes to the code, on Fedora 40.

The README instructions seem to be still up-to-date, but is not 100% correct for Fedora. [Package names follow a different format, and the `libcurl` dependency is packaged differently](https://gist.github.com/mounaiban/ed90ddaa4660dde0d39b8274649033cc#summary).