libimobiledevice/idevicerestore #215: 'ERROR: Unable to discover device model' on iPad 5 (2017)

iakdev, 2018-03-16:
iPad6,11 (J71tAP) won't get recorgnized correctly.

I get the follwoing ouput on macOS and Windows:

[INFO] 64bit device detected
futurerestore init done
reading ticket myshsh.shsh2 done
[Error] ERROR: Unable to discover device model
[Error] Fail code=-2
Failed with errorcode=-2

iakdev, 2018-03-17:
I think the problem might be that only "j71sap" but not "j71tap" for iPad6,11 is in libirecovery/src/libirecovery.c

iakdev, 2018-03-17:
Adding

{"iPad6,11",  "j71tap", 0x10, 0x8003 },

to irecv_devices[] in said file fixed the problem for me.

nikias, 2018-03-19:
Fixed with commit https://github.com/libimobiledevice/libirecovery/commit/260cb0684bf1cb7d2cdef8ce4a1ec6d832eb1d1c

closed 2018-03-19 by nikias

jailbreak2018, 2018-05-15:
I'm having the same issue but on mac, I was wondering if you could help me since I'm confused about the j71tap and other things thank you

iakdev, 2018-05-15:
This isn’t a problem with libmobiledevice anymore.
encounter needs to update his futurerestore branch or you build it yourself from scratch using the latest libmobiledevice stack.

See https://github.com/encounter/futurerestore/issues/32