libimobiledevice/idevicerestore #216: idevicerestore hangs when phone is unplugged during restore

Blefish, 2018-03-22:
Hi

It sometimes happens that the phone loses communication midway through restore, be it due to any reason.
idevicerestore hangs in such cases due to this line:
https://github.com/libimobiledevice/idevicerestore/blob/master/src/restore.c#L2342

Does this error (no data to read) happen from normal restore process as well? Would there be a way to drop out of the loop if communication is truly lost with the phone?

nikias, 2018-03-22:
We can detect the unplug of the device and should add it. I think we had this at some point in the past but there were some issues. However it wasn't used to break out of the loop IIRC.
I will come up with something better ;)

Blefish, 2018-04-25:
Actually it seems like this also happens if phone is not physically unplugged. I am not sure what the cause might be, however today I attempted to update iPhone 7 to 11.3.1 and it failed.
```
2018-04-25 10:25:50: /usr/local/bin/idevicerestore -e -l -C XXX -i 1049D20885E526
2018-04-25 10:25:51: NOTE: using cached version data
2018-04-25 10:25:51: Found device in Normal mode
2018-04-25 10:25:51: Identified device as d101ap, iPhone9,3
2018-04-25 10:25:51: Latest firmware is iPhone_4.7_P3_11.0_11.3.1_15E302_Restore.ipsw
2018-04-25 10:25:51: Verifying 'XXX/iPhone_4.7_P3_11.0_11.3.1_15E302_Restore.ipsw'...
2018-04-25 10:25:55: Checksum matches.
2018-04-25 10:25:55: Extracting BuildManifest from IPSW
2018-04-25 10:25:55: Product Version: 11.3.1
2018-04-25 10:25:55: Product Build: 15E302 Major: 15
2018-04-25 10:25:55: Device supports Image4: true
2018-04-25 10:25:55: Variant: Customer Erase Install (IPSW)
2018-04-25 10:25:55: This restore will erase your device data.
2018-04-25 10:25:55: Using cached filesystem from 'XXX/iPhone_4.7_P3_11.0_11.3.1_15E302_Restore/058-97716-130.dmg'
2018-04-25 10:25:55: Found ECID 4584766062322982
2018-04-25 10:25:55: Getting ApNonce in normal mode... 9a c0 f6 6b d1 15 44 b6 1a 44 77 86 29 28 ab 17 51 ed 0d 9e 62 06 c7 0d 73 ab 37 dd 8d bc b2 df 
2018-04-25 10:25:55: Trying to fetch new SHSH blob
2018-04-25 10:25:55: Getting SepNonce in normal mode... aa 05 61 60 cd 91 77 95 49 1a b3 a5 cd 7f 55 02 9e 16 ec 68 
2018-04-25 10:25:55: WARNING: Unable to find BbSkeyId node
2018-04-25 10:25:55: Request URL set to https://gs.apple.com/TSS/controller?action=2
2018-04-25 10:25:56: Sending TSS request attempt 1... response successfully received
2018-04-25 10:25:56: Received SHSH blobs
2018-04-25 10:25:56: Entering recovery mode...
2018-04-25 10:26:01: [USB] 05ac:12a8 left
2018-04-25 10:26:06: [USB] 05ac:1281 arrived
2018-04-25 10:26:06: [USB] Serial number CPID:8010 CPRV:11 CPFM:03 SCEP:01 BDID:0C ECID:001049D20885E526 IBFL:3D SRNM:[DNPTMFZSHG7F]
2018-04-25 10:26:08: INFO: device serial number is DNPTMFZSHG7F
2018-04-25 10:26:08: Extracting iBEC.d10.RELEASE.im4p...
2018-04-25 10:26:08: Personalizing IMG4 component iBEC...
2018-04-25 10:26:08: Sending iBEC (439094 bytes)...
2018-04-25 10:26:08: [USB] 05ac:1281 left
2018-04-25 10:26:10: [USB] 05ac:1281 arrived
2018-04-25 10:26:10: [USB] Serial number CPID:8010 CPRV:11 CPFM:03 SCEP:01 BDID:0C ECID:001049D20885E526 IBFL:3C SRNM:[DNPTMFZSHG7F]
2018-04-25 10:26:15: Recovery Mode Environment:
2018-04-25 10:26:15: iBoot build-version=iBoot-4076.50.126
2018-04-25 10:26:15: iBoot build-style=RELEASE
2018-04-25 10:26:15: Sending AppleLogo...
2018-04-25 10:26:15: Extracting applelogo@2x~iphone.im4p...
2018-04-25 10:26:15: Personalizing IMG4 component AppleLogo...
2018-04-25 10:26:15: Sending AppleLogo (12820 bytes)...
2018-04-25 10:26:15: ramdisk-size=0x20000000
2018-04-25 10:26:15: Extracting 058-97541-138.dmg...
2018-04-25 10:26:15: Personalizing IMG4 component RestoreRamDisk...
2018-04-25 10:26:15: Sending RestoreRamDisk (79287987 bytes)...
2018-04-25 10:26:25: Extracting DeviceTree.d101ap.im4p...
2018-04-25 10:26:25: Personalizing IMG4 component RestoreDeviceTree...
2018-04-25 10:26:25: Sending RestoreDeviceTree (153126 bytes)...
2018-04-25 10:26:25: Extracting kernelcache.release.iphone9...
2018-04-25 10:26:25: Personalizing IMG4 component RestoreKernelCache...
2018-04-25 10:26:25: Sending RestoreKernelCache (15630385 bytes)...
2018-04-25 10:26:27: About to restore device... 
2018-04-25 10:26:27: Waiting for device...
2018-04-25 10:26:28: [USB] 05ac:1281 left
2018-04-25 10:26:35: [USB] 05ac:12a8 arrived
2018-04-25 10:26:35: [USB] Serial number ffffffffffffffffffffffffffffffffffffffff
2018-04-25 10:26:35: Device ffffffffffffffffffffffffffffffff00000007 is now connected in restore mode...
2018-04-25 10:26:35: Connecting now...
2018-04-25 10:26:35: Connected to com.apple.mobile.restored, version 15
2018-04-25 10:26:35: Device ffffffffffffffffffffffffffffffff00000007 has successfully entered restore mode
2018-04-25 10:26:35: Hardware Information:
2018-04-25 10:26:35: BoardID: 12
2018-04-25 10:26:35: ChipID: 32784
2018-04-25 10:26:35: UniqueChipID: 4584766062322982
2018-04-25 10:26:35: ProductionMode: true
2018-04-25 10:26:35: Starting FDR listener thread
2018-04-25 10:26:35: About to send NORData...
2018-04-25 10:26:35: Found firmware path Firmware/all_flash
2018-04-25 10:26:35: Getting firmware manifest from build identity
[... 20 lines cut ...]
2018-04-25 10:26:35: Extracting recoverymode@1334~iphone-lightning.im4p...
2018-04-25 10:26:35: Personalizing IMG4 component RecoveryMode...
2018-04-25 10:26:35: Extracting iBoot.d10.RELEASE.im4p...
2018-04-25 10:26:35: Personalizing IMG4 component iBoot...
2018-04-25 10:26:35: Extracting sep-firmware.d101.RELEASE.im4p...
2018-04-25 10:26:35: Personalizing IMG4 component RestoreSEP...
2018-04-25 10:26:35: Extracting sep-firmware.d101.RELEASE.im4p...
2018-04-25 10:26:36: Personalizing IMG4 component SEP...
2018-04-25 10:26:36: Sending NORData now...
2018-04-25 10:26:36: Done sending NORData
2018-04-25 10:26:37: About to send RootTicket...
2018-04-25 10:26:37: Sending RootTicket now...
2018-04-25 10:26:37: Done sending RootTicket
2018-04-25 10:26:37: Waiting for NAND (28)
2018-04-25 10:26:37: Updating S3E Firmware (58)
2018-04-25 10:26:44: Checking filesystems (15)
2018-04-25 10:26:44: ERROR: Unable to receive message from FDR 0x3127f90 (-4). 0/2 bytes
2018-04-25 10:26:44: [USB] 05ac:12a8 left
[... cut ...]
The process did not end and I killed it manually. The phone showed progress bar with 0% the whole time and did not reboot on its own (which I've normally seen it do). When I tried to call ireovery -n to restart the phone, it failed. The phone did not respond in any way (ideviceinfo, ideviceenterrecovery did not respond too) and the only way to get it out was to hold down volume down+power.

The USB messages are added by myself. Is there any way to debug this further? I was not able to replicate it after doing it one more time, when it succeeded to update to 11.3.1. I've been running it in a loop for more than 15 times already.

nikias, 2018-04-25:
Hmm weird. Looks like the phone reconnects on its own here...

Blefish, 2018-04-25:
I wonder if the phone expects the firmware flashing to continue but since the connection is lost, it gets stuck in the loop?

nikias, 2018-04-25:
Maybe it is something new and we need to react to it? Is it happening at exactly the same stage (Checking filesystems)?

Blefish, 2018-07-27:
I noticed that on this stage, the phone does not reboot on its own and is completely frozen. If it happens later in the process, phone reboots and I can retrigger the update one more time and hope this time it succeeds.

closed 2023-04-29 by Blefish