libimobiledevice/idevicerestore #217: iPhone 8 Plus and iPhone X can't restore 11.3 15E5216a

taanders, 2018-03-27:
iPhone 8 Plus and iPhone X devices are failing with some kind of Verification Error with images iPhone_5.5_P3_11.0_11.3_15E5216a_Restore.ipsw and iPhone10,3,iPhone10,6_11.3_15E5216a_Restore.ipsw:
`Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
common.c:printing 1728 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>@ServerVersion</key>
	<string>2.1.0</string>
	<key>SE,Ticket</key>
	<data>
	MIIEHBYESU00TQIBADGCAVT/hOqFnEKCAUswggFHFgRNQU5CMYIBPf+E6oWcUIHCMIG/
	FgRNQU5QMYG2/4eblcZpCzAJFgRzZWNpAgFz/4ebldJkIjAgFgRzZWlkBBgEDyEz6kSA
	AXJ3GUh4lDhjoDJL78HBi77/h5uV1mkqMCgWBHNla2kEIGMxKzVda1U+b7LEbYunKVin
	Lbggaawpr2Q7+srHzvri/4ebldxjHjAcFgRzZW5jBBSMVCHMI0rwsSgkjZ2kf4OiPBSZ
	lP+Hm8nsbh4wHBYEc3J2bgQUDKlOdiW+twiuR3Ph0igC2BMBOwr/hPqJlFBsMGoWBE9C
	SlAxYv+Hm5XIaCowKBYEc2VkaAQgetfBlnWYIcGEdnNIlf6vqUculEveDpLVyf8wul8t
	BDv/h5uV4GgqMCgWBHNlcGgEIH0UICt4lBsuAQzsKCb+QMu8NCo1p2uAxPs7ICQD3Vr2
	BEcwRQIhAKBIs7RA39gYIsYFbbOtbVESqeTTPHKj3+3j/T86yZU9AiBX6OO4tXtJ4Uyt
	Vqa/kUMd9wqtyeyMkOFC9KAgURgqWzCCAm4wggJqMIICEKADAgECAghfkiMnEnJj4TAK
	BggqhkjOPQQDAjBDMR8wHQYDVQQDDBZTRSBTZWN1cmUgQm9vdCBSb290IENBMRMwEQYD
	VQQKDApBcHBsZSBJbmMuMQswCQYDVQQGEwJVUzAeFw0xNjEwMjExODQ4MTlaFw0zNTEw
	MzEyMjU2MDZaMEwxKDAmBgNVBAMMH1NFNzMtUHJvZHVjdGlvbi1SZXZBLURhdGFDZW50
	ZXIxEzARBgNVBAoMCkFwcGxlIEluYy4xCzAJBgNVBAYTAlVTMFkwEwYHKoZIzj0CAQYI
	KoZIzj0DAQcDQgAEp1JFfAm3xC2tlrGkpwt20yk0lpYNoWrHE7VGYtWMWgEzsN1OOXD9
	GLG/aarM4DLeTIA5B3XxBAOmFd4hCun+E6OB5DCB4TAdBgNVHQ4EFgQU7Oueneaagj2L
	vEDrQ8RYvcXA6KwwDAYDVR0TAQH/BAIwADAfBgNVHSMEGDAWgBSWUzQbf9pr41zkx4qa
	IZhL4zC5YDAOBgNVHQ8BAf8EBAMCB4AwgYAGCiqGSIb3Y2QGAQ8BAf8EbzFt/4TqhZxQ
	QjBAFgRNQU5QMTj/h5uVxmkLMAkWBHNlY2kCAXP/h5uV0mQMMAoWBHNlaWSgAgUA/4eb
	ldxjDDAKFgRzZW5joAIFAP+E+omUUB0wGxYET0JKUDET/4ebleBoDDAKFgRzZXBooAIF
	ADAKBggqhkjOPQQDAgNIADBFAiEAkUy+2pXc47kUeHPommsfFUExhcQFmTQvnaYaHAAS
	EM4CIDm3mMtGbKktiXicqQs0mL9AZThK0d8C2MRw45NIqB7B
	</data>
</dict>
</plist>
Received SE ticket
Sending FirmwareResponse data now...
Done sending FirmwareUpdater data
Got status message
Status: Verification Error
Log is available:
SYSLOG: Mar 26 10:33:15 localhost bootlog[0] <Notice>: BOOT_TIME 1522060395 937377
[10:33:16.0383-GMT]{5>7} CHECKPOINT NOTICE: Image4 device: AP nonce clearable
entering ramrod_clear_ap_nonce
[10:33:16.0397-GMT]{5>7} CHECKPOINT NOTICE: AP nonce consumed
[10:33:16.0398-GMT]{5>7} CHECKPOINT NOTICE: Pre-existing NVRAM variable: auto-boot=false
[10:33:16.0399-GMT]{5>7} CHECKPOINT NOTICE: Pre-existing NVRAM variable: restore-outcome=initial_monitor_no_return
[10:33:16.0400-GMT]{5>7} CHECKPOINT PROGRESS: START (unknown) -> (initial_engine_no_return)
[10:33:16.0400-GMT]{5>7} CHECKPOINT NOTICE: NVRAM access available on initial check
restore-outcome = initial_engine_no_return
executing /usr/sbin/nvram restore-outcome=initial_engine_no_return
[10:33:16.0409-GMT]{5>7} CHECKPOINT BEGIN: MAIN:[0x0400] umask
restore-step-ids = {0x11030400:1}
restore-step-names = {0x11030400:umask}
restore-step-uptime = 1
restore-step-user-progress = -1
[10:33:16.0412-GMT]{5>7} CHECKPOINT END: MAIN:[0x0400] umask
restore-step-ids = {}
restore-step-names = {}
restore-step-uptime = 1
restore-step-user-progress = -1
[10:33:16.0415-GMT]{5>7} CHECKPOINT BEGIN: MAIN:[0x0402] setvbuf
restore-step-ids = {0x11030402:2}
restore-step-names = {0x11030402:setvbuf}
restore-step-uptime = 1
restore-step-user-progress = -1
[10:33:16.0417-GMT]{5>7} CHECKPOINT END: MAIN:[0x0402] setvbuf
restore-step-ids = {}
restore-step-names = {}
[... 519 lines cut ...]
restore-step-ids = {0x11030635:60;0x11031309:75}
restore-step-names = {0x11030635:update_device_firmware;0x11031309:update_se}
restore-step-uptime = 200
restore-step-user-progress = 85
[10:36:35.0634-GMT]{5>7} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x11030635:60;0x11031309:75;0x11030648:77}
restore-step-names = {0x11030635:update_device_firmware;0x11031309:update_se;0x11030648:cleanup_send_final_status}
restore-step-uptime = 200
restore-step-user-progress = 85

FDR 06B90800 timeout waiting for command
FDR 06B90800 waiting for message...
ERROR: Unable to successfully restore device
FDR 06B90800 timeout waiting for command
FDR 06B90800 waiting for message...
FDR 06B90800 timeout waiting for command
FDR 06B90800 waiting for message...
FDR 06B90800 timeout waiting for command
FDR 06B90800 waiting for message...
No data to read
FDR 06B90800 timeout waiting for command
FDR 06B90800 terminating...
ERROR: Unable to restore device`

iPhone 6S restore with firmware iPhone_4.7_11.3_15E5216a_Restore.ipsw is working ok.

nikias, 2018-03-27:
Have you checked #214?

taanders, 2018-03-27:
Yes, I tried that but it seems not to solve this verification error. I did not got that error message "ERROR: Unable to send FirmwareUpdater data" as on issue #214.

4kzeng, 2018-03-29:
try restore.c line 1836 add two line:
  plist_dict_remove_item(plist_dict_get_item(request,"SE,UpdatePayload"), "DevelopmentUpdatePayloadHash");
  plist_dict_remove_item(plist_dict_get_item(request,"SE,Firmware"), "DevelopmentCMAC");
  info("Sending SE TSS request...\n");
  response = tss_request_send(request, client->tss_url);

goodboydheeraj, 2018-03-30:
Sorry for hijacking the post. But can someone please look into this futurerestore issue that fails to restore 11.3 sep/baseband?

https://github.com/encounter/futurerestore/issues/17

Futurerestore uses idevicerestore, so though if i could get any help from you guys!

4kzeng, 2018-04-04:
@taanders Did you try?

taanders, 2018-04-27:
Both ways mine or 4kzeng's code seems to fix this kind of issue on iPhone 8 and iPhone 8 Plus restore.

nikias, 2018-04-27:
This should be working with latest commit b8b3035500eb2311aab366aaa3e5069ee422e8e3. Re-open if this should still be an issue.

closed 2018-04-27 by nikias