libimobiledevice/idevicerestore #218: undefined reference to symbol 'gzread'

vostai, 2018-04-20:
Hi guys, I have a problem when compiling idevicerestore. I've installed all of it dependencies. It's normal when I run `./autogen.sh`  but when I run `make` I stuck at this
`make  all-recursive
make[1]: Entering directory '/home/admin1/Downloads/idevicerestore-master'
Making all in src
make[2]: Entering directory '/home/admin1/Downloads/idevicerestore-master/src'
  CC       idevicerestore-idevicerestore.o
  CCLD     idevicerestore
idevicerestore-idevicerestore.o: In function `idevicerestore_start':
/home/admin1/Downloads/idevicerestore-master/src/idevicerestore.c:632: warning: the use of `tempnam' is dangerous, better use `mkstemp'
/usr/bin/ld: idevicerestore-idevicerestore.o: undefined reference to symbol 'gzread'
//usr/local/lib/libz.so.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
Makefile:447: recipe for target 'idevicerestore' failed
make[2]: *** [idevicerestore] Error 1
make[2]: Leaving directory '/home/admin1/Downloads/idevicerestore-master/src'
Makefile:425: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/admin1/Downloads/idevicerestore-master'
Makefile:357: recipe for target 'all' failed
make: *** [all] Error 2
`
What should I do now? Should I re-check the dependences? Thanks for reading and sorry for bad English :D

nikias, 2018-04-25:
Try this patch: https://gist.github.com/nikias/f5b16b933921e7557b9d2c27781dcd04

OothecaPickle, 2018-05-19:
@nikias i was having the exact same issue as @bkruggel on sierra, your patch fixed it! thx a bunch :)

nikias, 2018-06-01:
Fixed with commit 707856d9005a48ee006dfcc7e5424473cf9e7652

closed 2018-06-01 by nikias