libimobiledevice/idevicerestore #219: could not retrieve device serial number. Can't continue.

esamorokov, 2018-04-25:
I was told that it was previously resolved, however I did not find the resolution, please help.
Here is my debug log:

> downloading: 99%

> * Connection #0 to host updates-http.cdn-apple.com left intact
> 
> Verifying 'iPhone_4.0_64bit_11.3.1_15E302_Restore.ipsw'...
> Checksum matches.
> Extracting BuildManifest from IPSW
> Product Version: 11.3.1

> 
> Product Build: 15E302 Major: 15
> opening device 05ac:1227...
> found device with ECID 000e0410248113ba
> Setting to configuration 1
> Setting to interface 0:0
> Device supports Image4: true
> Variant: Customer Erase Install (IPSW)
> This restore will erase your device data.
> Extracting filesystem from IPSW
> [==================================================] 100.0%
> Getting device's ECID for TSS request
> Found ECID 3945117052375994
> Getting ApNonce in dfu mode... 79 3c 41 52 15 41 ac bd fa 84 51 41 4e ed a0 5c b4 b6 56 b7 
> Trying to fetch new SHSH blob
> Getting SepNonce in dfu mode... 57 23 e7 5b be 93 df a1 34 c1 3a db 3f 2f bd 3b d8 c0 99 ad 
> NOTE: Unable to find BbCalibrationManifestKeyHash node
> NOTE: Unable to find BbFactoryActivationManifestKeyHash node
> WARNING: Unable to find BbSkeyId node
> DEBUG: Applying restore request rules for entry AOP
> DEBUG: Adding action EPRO to TSS entry
> DEBUG: Adding action ESEC to TSS entry
> DEBUG: Applying restore request rules for entry AppleLogo
> DEBUG: Adding action EPRO to TSS entry
> DEBUG: Adding action ESEC to TSS entry
> DEBUG: Applying restore request rules for entry BatteryCharging0
> DEBUG: Adding action EPRO to TSS entry
> DEBUG: Adding action ESEC to TSS entry
> DEBUG: Applying restore request rules for entry BatteryCharging1
> DEBUG: Adding action EPRO to TSS entry
> DEBUG: Adding action ESEC to TSS entry
> DEBUG: Applying restore request rules for entry BatteryFull
> DEBUG: Adding action EPRO to TSS entry
> DEBUG: Adding action ESEC to TSS entry
> DEBUG: Applying restore request rules for entry BatteryLow0
> DEBUG: Adding action EPRO to TSS entry
> DEBUG: Adding action ESEC to TSS entry
> DEBUG: Applying restore request rules for entry BatteryLow1
> DEBUG: Adding action EPRO to TSS entry
> DEBUG: Adding action ESEC to TSS entry
> DEBUG: Applying restore request rules for entry BatteryPlugin
> DEBUG: Adding action EPRO to TSS entry
> DEBUG: Adding action ESEC to TSS entry
> DEBUG: Applying restore request rules for entry DeviceTree
> DEBUG: Adding action EPRO to TSS entry
> DEBUG: Adding action ESEC to TSS entry
> DEBUG: Applying restore request rules for entry KernelCache
> DEBUG: Adding action EPRO to TSS entry
> DEBUG: Adding action ESEC to TSS entry
> DEBUG: Applying restore request rules for entry LLB
> DEBUG: Adding action EPRO to TSS entry
> DEBUG: Adding action ESEC to TSS entry
> DEBUG: Applying restore request rules for entry OS
> DEBUG: Adding action EPRO to TSS entry
> DEBUG: Adding action ESEC to TSS entry
> DEBUG: Applying restore request rules for entry RecoveryMode
> DEBUG: Adding action EPRO to TSS entry
> DEBUG: Adding action ESEC to TSS entry
[... 551 lines cut ...]
> Setting to configuration 1
> Setting to interface 0:0
> DEBUG: tss_response_get_path_by_entry: No entry 'iBSS' in TSS response
> NOTE: No path for component iBSS in TSS, will fetch from build_identity
> Extracting iBSS.n69.RELEASE.im4p...
> Personalizing IMG4 component iBSS...
> Sending iBSS (223704 bytes)...
> [==================================================] 100.0%
> opening device 05ac:1227...
> found device with ECID 000e0410248113ba
> Setting to configuration 1
> Setting to interface 0:0
> Nonce: 79 3c 41 52 15 41 ac bd fa 84 51 41 4e ed a0 5c b4 b6 56 b7 
> Setting to configuration 1
> DEBUG: tss_response_get_path_by_entry: No entry 'iBEC' in TSS response
> NOTE: No path for component iBEC in TSS, will fetch from build_identity
> Extracting iBEC.n69.RELEASE.im4p...
> Personalizing IMG4 component iBEC...
> Sending iBEC (354520 bytes)...
> [==================================================] 100.0%
> opening device 05ac:1281...
> found device with ECID 000e0410248113ba
> Setting to configuration 1
> Setting to interface 0:0
> ERROR: could not retrieve device serial number. Can't continue.

Blefish, 2018-04-26:
Are you able to use iTunes to restore? I have one sample phone with the same issue but even iTunes can't detect serial number and restore it.

nikias, 2018-04-26:
Oh the serial number is only used to match a device coming from recovery mode into restore mode. I think this check can be removed or turned into a warning.

esamorokov, 2018-04-27:
That would be great!

esamorokov, 2018-04-27:
No, non of the applications were able to restore, including iTunes.

cmdpwnd, 2019-07-25:
bump..

I also am getting this error. Is there a patch to disable the serial confirmation?

```
root@kali:~# idevicerestore /media/root/Elements/Files/vmware/Apple/iPhone_4.0_64bit_12.4_16G77_Restore.ipsw
Found device in DFU mode
Identified device as n51ap, iPhone6,1
Extracting BuildManifest from IPSW
Product Version: 12.4
Product Build: 16G77 Major: 16
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 048-77707-092.dmg
[==================================================] 100.0%
Found ECID --SNIP--
Getting ApNonce in dfu mode... --SNIP--
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode...--SNIP-- 
WARNING: Unable to find BbSkeyId node
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBSS.iphone6.RELEASE.im4p...
Personalizing IMG4 component iBSS...
Sending iBSS (174065 bytes)...
[==================================================] 100.0%
Nonce: --SNIP--
Extracting iBEC.iphone6.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (731057 bytes)...
[==================================================] 100.0%
ERROR: could not retrieve device serial number. Can't continue.
root@kali:~# 
```

cmdpwnd, 2019-07-25:
@nikias @esamorokov 

I made some modifications to bypass checking the serial number, and increased the `sleep(7)` to `sleep(25)` when waiting after *IBEC*. **I was able to get the phone to reboot into recovery mode with the apple logo & loading bar**, but the program either **segfaults** after *exactly 12 attempts* to `connect to restore mode device...` or will loop infinitely. I do not know the cause for the loop, and am trying to track the segfault, but gdb points at `0x0000000000000000 in ?? ()`. Source [`idevicerestore.c#1191`](https://github.com/libimobiledevice/idevicerestore/blob/0be1f2f64c5fc05cb9b6843b13c61193eec8e5b3/src/idevicerestore.c#L1191)

```
--SNIP, THIS WAS REALLY LONG--
About to restore device... 
Waiting for device...
[New Thread 0x7ffff596f700 (LWP 1985)]
Attempt 1 to connect to restore mode device...
Attempt 2 to connect to restore mode device...
Attempt 3 to connect to restore mode device...
Attempt 4 to connect to restore mode device...
Attempt 5 to connect to restore mode device...
Attempt 6 to connect to restore mode device...
Attempt 7 to connect to restore mode device...
Attempt 8 to connect to restore mode device...
Attempt 9 to connect to restore mode device...
Attempt 10 to connect to restore mode device...
Attempt 11 to connect to restore mode device...
Attempt 12 to connect to restore mode device...

Thread 4 "idevicerestore" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff596f700 (LWP 1985)]
0x0000000000000000 in ?? ()
(gdb) 
(gdb) where
#0  0x0000000000000000 in ?? ()
#1  0x00007fffe8001ff0 in ?? ()
#2  0x0000000000000000 in ?? ()
(gdb) list
1188		}
1189	}
1190	
1191	void idevicerestore_set_progress_callback(struct idevicerestore_client_t* client, idevicerestore_progress_cb_t cbfunc, void* userdata)
1192	{
1193		if (!client)
1194			return;
1195		client->progress_cb = cbfunc;
1196		client->progress_cb_data = userdata;
1197	}
(gdb) 
```
Killing it in loop has not been very helpful either
```
About to restore device... 
Waiting for device...
[New Thread 0x7ffff596f700 (LWP 3219)]
Attempt 1 to connect to restore mode device...
Attempt 2 to connect to restore mode device...
Attempt 3 to connect to restore mode device...
Attempt 4 to connect to restore mode device...
Attempt 5 to connect to restore mode device...
Attempt 6 to connect to restore mode device...
Attempt 7 to connect to restore mode device...
Attempt 8 to connect to restore mode device...
Attempt 9 to connect to restore mode device...
Attempt 10 to connect to restore mode device...
^C
Thread 1 "idevicerestore" received signal SIGINT, Interrupt.
0x00007ffff781b720 in __GI___nanosleep (requested_time=requested_time@entry=0x7fffffffd250, remaining=remaining@entry=0x7fffffffd250)
    at ../sysdeps/unix/sysv/linux/nanosleep.c:28
28	../sysdeps/unix/sysv/linux/nanosleep.c: No such file or directory.
(gdb) list
23	in ../sysdeps/unix/sysv/linux/nanosleep.c
(gdb) where
#0  0x00007ffff781b720 in __GI___nanosleep (requested_time=requested_time@entry=0x7fffffffd250, remaining=remaining@entry=0x7fffffffd250)
    at ../sysdeps/unix/sysv/linux/nanosleep.c:28
#1  0x00007ffff781b62a in __sleep (seconds=0, seconds@entry=1) at ../sysdeps/posix/sleep.c:55
#2  0x000055555556758d in restore_open_with_timeout (client=client@entry=0x55555559bed0) at restore.c:427
#3  0x000055555556bd4c in restore_device (client=0x55555559bed0, build_identity=0x5555557c2500, 
    filesystem=0x5555557c2240 "/media/root/Elements/Files/vmware/Apple/iPhone_4.0_64bit_12.4_16G77_Restore/048-77707-092.dmg") at restore.c:2275
#4  0x000055555555c976 in idevicerestore_start (client=0x55555559bed0) at idevicerestore.c:1047
#5  0x0000555555558f58 in main (argc=<optimized out>, argv=<optimized out>) at idevicerestore.c:1322
(gdb) print client
No symbol "client" in current context.
(gdb) 
```

Blefish, 2019-07-25:
I believe the "Attempt x to connect to restore mode device..." also waits for devices that match the serial number, as seen here: https://github.com/libimobiledevice/idevicerestore/blob/master/src/restore.c#L379

So you could remove the restore_is_current_device function call and recheck.

cmdpwnd, 2019-07-25:
@Blefish 

I'd commented out the section of `restore_is_current_device` function which checks the serial, it also checks the `UDID`. I might try removing the call to that function where you mentioned, like this:

```diff
diff --git a/src/dfu.c b/src/dfu.c
index 8c65b93..5da087e 100644
--- a/src/dfu.c
+++ b/src/dfu.c
@@ -404,7 +404,7 @@ int dfu_enter_recovery(struct idevicerestore_client_t* client, plist_t build_ide
 
 	dfu_client_free(client);
 
-	sleep(7);
+	sleep(25);
 
 	// Reconnect to device, but this time make sure we're not still in DFU mode
 	if (recovery_client_new(client) < 0) {
@@ -419,7 +419,7 @@ int dfu_enter_recovery(struct idevicerestore_client_t* client, plist_t build_ide
 	irecv_get_mode(client->recovery->client, &mode);
 
 	if (mode == IRECV_K_DFU_MODE) {
-		error("ERROR: Unable to connect to recovery device\n");
+		error("ERROR: Unable to connect to recovery device, still in DFU mode\n");
 		if (client->recovery->client) {
 			irecv_close(client->recovery->client);
 			client->recovery->client = NULL;
diff --git a/src/idevicerestore.c b/src/idevicerestore.c
index 1918987..f184a4c 100644
--- a/src/idevicerestore.c
+++ b/src/idevicerestore.c
@@ -1022,12 +1022,12 @@ int idevicerestore_start(struct idevicerestore_client_t* client)
 
 	// now finally do the magic to put the device into restore mode
 	if (client->mode->index == MODE_RECOVERY) {
-		if (client->srnm == NULL) {
+		/*if (client->srnm == NULL) {
 			error("ERROR: could not retrieve device serial number. Can't continue.\n");
 			if (delete_fs && filesystem)
 				unlink(filesystem);
 			return -1;
-		}
+		}*/
 		if (recovery_enter_restore(client, build_identity) < 0) {
 			error("ERROR: Unable to place device into restore mode\n");
 			plist_free(buildmanifest);
diff --git a/src/restore.c b/src/restore.c
index 4e6b91d..fd3b3af 100644
--- a/src/restore.c
+++ b/src/restore.c
@@ -247,7 +247,7 @@ irecv_device_t restore_get_irecv_device(struct idevicerestore_client_t* client)
 		return NULL;
 	}
 
-	if (client->srnm == NULL) {
+	/*if (client->srnm == NULL) {
 		restore_error = restored_get_value(restore, "SerialNumber", &node);
 		if (restore_error != RESTORE_E_SUCCESS || !node || plist_get_node_type(node) != PLIST_STRING) {
 			error("ERROR: Unable to get SerialNumber from restored\n");
@@ -260,7 +260,7 @@ irecv_device_t restore_get_irecv_device(struct idevicerestore_client_t* client)
 		info("INFO: device serial number is %s\n", client->srnm);
 		plist_free(node);
 		node = NULL;
-	}
+	}*/
 
 	restore_error = restored_get_value(restore, "HardwareModel", &node);
 	restored_client_free(restore);
@@ -313,10 +313,10 @@ static int restore_is_current_device(struct idevicerestore_client_t* client, con
 	if (!client) {
 		return 0;
 	}
-	if (!client->srnm) {
+	/*if (!client->srnm) {
 		error("ERROR: %s: no SerialNumber given in client data\n", __func__);
 		return 0;
-	}
+	}*/
 
 	idevice_t device = NULL;
 	idevice_error_t device_error;
@@ -349,7 +349,7 @@ static int restore_is_current_device(struct idevicerestore_client_t* client, con
 
 	plist_t node = NULL;
 	restore_error = restored_get_value(restored, "SerialNumber", &node);
-	if ((restore_error != RESTORE_E_SUCCESS) || !node || (plist_get_node_type(node) != PLIST_STRING)) {
+	/*if ((restore_error != RESTORE_E_SUCCESS) || !node || (plist_get_node_type(node) != PLIST_STRING)) {
 		error("ERROR: %s: Unable to get SerialNumber from restored\n", __func__);
 		restored_client_free(restored);
 		idevice_free(device);
@@ -357,7 +357,7 @@ static int restore_is_current_device(struct idevicerestore_client_t* client, con
 			plist_free(node);
 		}
 		return 0;
-	}
+	}*/
 	restored_client_free(restored);
 	idevice_free(device);
 
@@ -365,9 +365,9 @@ static int restore_is_current_device(struct idevicerestore_client_t* client, con
 	plist_get_string_val(node, &this_srnm);
 	plist_free(node);
 
-	if (!this_srnm) {
+	/*if (!this_srnm) {
 		return 0;
-	}
+	}*/
 
 	return (strcasecmp(this_srnm, client->srnm) == 0);
}
@@ -376,7 +376,7 @@ static void restore_device_event_cb(const idevice_event_t *event, void *user_dat
 {
        if (event->event == IDEVICE_DEVICE_ADD) {
                struct idevicerestore_client_t* client = (struct idevicerestore_client_t*)user_data;
-               if (!restore_device_connected && restore_is_current_device(client, event->udid)) {
+               if (!restore_device_connected /*&& restore_is_current_device(client, event->udid)*/) {
                        restore_device_connected = 1;
                        client->udid = strdup(event->udid);
                }
@@ -398,10 +398,10 @@ int restore_open_with_timeout(struct idevicerestore_client_t* client) {
                return -1;
        }
 
-       if(client->srnm == NULL) {
+       /*if(client->srnm == NULL) {
                error("ERROR: no SerialNumber in client data!\n");
                return -1;
-       }
+       }*/
 
        // create our restore client if it doesn't yet exist
        if(client->restore == NULL) {

```

Blefish, 2019-07-25:
@cmdpwnd You can keep the call, but make sure it returns 1 instead in that case (current patch returns 0) :). Any luck with this though?

cmdpwnd, 2019-07-25:
@Blefish 

I removed the call & have the same results...

cmdpwnd, 2019-07-25:
@Blefish 

Since `strcasecmp` depends on `this_srnm`, I may chase that.

Blefish, 2019-07-25:
You may also run autogen.sh with --enable-debug-code and recompile. This will allow gdb to properly show you the crashing line of code.

cmdpwnd, 2019-07-25:
I bypassed the comparison in `restore.c`, and now get the infinite loop every time. Also, the phone goes black on 3rd attempt to connect. I think that's why it's looping, but not sure what is triggering the shutdown on the phone...

```diff
@@ -365,11 +365,11 @@ static int restore_is_current_device(struct idevicerestore_client_t* client, con
        plist_get_string_val(node, &this_srnm);
        plist_free(node);
 
-       if (!this_srnm) {
+       /*CDUNN-SERIALif (!this_srnm) {
                return 0;
-       }
+       }*/
 
-       return (strcasecmp(this_srnm, client->srnm) == 0);
+       return 1;//(strcasecmp(this_srnm, client->srnm) == 0);
 }
```

cmdpwnd, 2019-07-26:
@Blefish @nikias 

I believe I've solved the serial number issue. Now I'm facing NVRAM not available.. close here & open new issue or not? *Note: one thing I find tricky is the sleep in `dfu.c`; when the backlight comes on & device initializes, the sooner you can reconnect the better... I've varied this from `sleep(2)` to `sleep(25)` to hook at the right time.*

```diff
diff --git a/src/dfu.c b/src/dfu.c
index 8c65b93..459cd41 100644
--- a/src/dfu.c
+++ b/src/dfu.c
@@ -404,7 +404,7 @@ int dfu_enter_recovery(struct idevicerestore_client_t* client, plist_t build_ide
 
 	dfu_client_free(client);
 
-	sleep(7);
+	sleep(4);
 
 	// Reconnect to device, but this time make sure we're not still in DFU mode
 	if (recovery_client_new(client) < 0) {
@@ -419,7 +419,7 @@ int dfu_enter_recovery(struct idevicerestore_client_t* client, plist_t build_ide
 	irecv_get_mode(client->recovery->client, &mode);
 
 	if (mode == IRECV_K_DFU_MODE) {
-		error("ERROR: Unable to connect to recovery device\n");
+		error("ERROR: Unable to connect to recovery device, still in DFU mode\n");
 		if (client->recovery->client) {
 			irecv_close(client->recovery->client);
 			client->recovery->client = NULL;
diff --git a/src/idevicerestore.c b/src/idevicerestore.c
index 1918987..3a6f587 100644
--- a/src/idevicerestore.c
+++ b/src/idevicerestore.c
@@ -1022,12 +1022,12 @@ int idevicerestore_start(struct idevicerestore_client_t* client)
 
 	// now finally do the magic to put the device into restore mode
 	if (client->mode->index == MODE_RECOVERY) {
-		if (client->srnm == NULL) {
+		/*if (client->srnm == NULL) {
 			error("ERROR: could not retrieve device serial number. Can't continue.\n");
 			if (delete_fs && filesystem)
 				unlink(filesystem);
 			return -1;
-		}
+		}*/
 		if (recovery_enter_restore(client, build_identity) < 0) {
 			error("ERROR: Unable to place device into restore mode\n");
 			plist_free(buildmanifest);
diff --git a/src/restore.c b/src/restore.c
index 4e6b91d..4d25290 100644
--- a/src/restore.c
+++ b/src/restore.c
@@ -247,7 +247,7 @@ irecv_device_t restore_get_irecv_device(struct idevicerestore_client_t* client)
 		return NULL;
 	}
 
-	if (client->srnm == NULL) {
+	/*if (client->srnm == NULL) {
 		restore_error = restored_get_value(restore, "SerialNumber", &node);
 		if (restore_error != RESTORE_E_SUCCESS || !node || plist_get_node_type(node) != PLIST_STRING) {
 			error("ERROR: Unable to get SerialNumber from restored\n");
@@ -260,7 +260,7 @@ irecv_device_t restore_get_irecv_device(struct idevicerestore_client_t* client)
 		info("INFO: device serial number is %s\n", client->srnm);
 		plist_free(node);
 		node = NULL;
-	}
+	}*/
 
 	restore_error = restored_get_value(restore, "HardwareModel", &node);
 	restored_client_free(restore);
@@ -313,10 +313,10 @@ static int restore_is_current_device(struct idevicerestore_client_t* client, con
 	if (!client) {
 		return 0;
 	}
-	if (!client->srnm) {
+	/*if (!client->srnm) {
 		error("ERROR: %s: no SerialNumber given in client data\n", __func__);
 		return 0;
-	}
+	}*/
 
 	idevice_t device = NULL;
 	idevice_error_t device_error;
@@ -349,7 +349,7 @@ static int restore_is_current_device(struct idevicerestore_client_t* client, con
 
 	plist_t node = NULL;
 	restore_error = restored_get_value(restored, "SerialNumber", &node);
-	if ((restore_error != RESTORE_E_SUCCESS) || !node || (plist_get_node_type(node) != PLIST_STRING)) {
+	/*if ((restore_error != RESTORE_E_SUCCESS) || !node || (plist_get_node_type(node) != PLIST_STRING)) {
 		error("ERROR: %s: Unable to get SerialNumber from restored\n", __func__);
 		restored_client_free(restored);
 		idevice_free(device);
@@ -357,7 +357,7 @@ static int restore_is_current_device(struct idevicerestore_client_t* client, con
 			plist_free(node);
 		}
 		return 0;
-	}
+	}*/
 	restored_client_free(restored);
 	idevice_free(device);
 
@@ -365,18 +365,18 @@ static int restore_is_current_device(struct idevicerestore_client_t* client, con
 	plist_get_string_val(node, &this_srnm);
 	plist_free(node);
 
-	if (!this_srnm) {
+	/*if (!this_srnm) {
 		return 0;
-	}
+	}*/
 
-	return (strcasecmp(this_srnm, client->srnm) == 0);
+	return 1;//(strcasecmp(this_srnm, client->srnm) == 0);
 }
 
 static void restore_device_event_cb(const idevice_event_t *event, void *user_data)
 {
 	if (event->event == IDEVICE_DEVICE_ADD) {
 		struct idevicerestore_client_t* client = (struct idevicerestore_client_t*)user_data;
-		if (!restore_device_connected && restore_is_current_device(client, event->udid)) {
+		if (!restore_device_connected /*&& restore_is_current_device(client, event->udid)*/) {
 			restore_device_connected = 1;
 			client->udid = strdup(event->udid);
 		}
@@ -398,10 +398,10 @@ int restore_open_with_timeout(struct idevicerestore_client_t* client) {
 		return -1;
 	}
 
-	if(client->srnm == NULL) {
+	/*if(client->srnm == NULL) {
 		error("ERROR: no SerialNumber in client data!\n");
 		return -1;
-	}
+	}*/
 
 	// create our restore client if it doesn't yet exist
 	if(client->restore == NULL) {
@@ -424,7 +424,7 @@ int restore_open_with_timeout(struct idevicerestore_client_t* client) {
 			info("Device %s is now connected in restore mode...\n", client->udid);
 			break;
 		}
-		sleep(1);
+		usleep(25*10000);//sleep(1);
 	}
 	idevice_event_unsubscribe();
 
@@ -435,9 +435,11 @@ int restore_open_with_timeout(struct idevicerestore_client_t* client) {
 
 	info("Connecting now...\n");
 	device_error = idevice_new(&device, client->udid);
+        info("**Evaluated device_error**\n");
 	if (device_error != IDEVICE_E_SUCCESS) {
 		return -1;
 	}
+        info("Success!\n");
 
 	restore_error = restored_client_new(device, &restored, "idevicerestore");
 	if (restore_error != RESTORE_E_SUCCESS) {
```

iha3ker1920, 2020-12-24:
I’m having this same issue on iPhone 4 3,3 trying to restore swapped rootfs of switchboard from iPhone 4 3,1... please help

mehdij00, 2021-05-21:
Hi
i am trying to restore an iphone X  to last ios version 14.5.1 
my device is on DFU mode it does not exit this mode by every way that i examined  it 
after fail to restore  i checked got text below  please help me what should i do and tell me if it damaged any hardware on device  and must replaced.
thank you  
Mehdi Jafari 

۰۵:۴۰:۲۸.۸۷۶:  Start flash
۰۵:۴۰:۲۸.۸۷۶:  Checking PC DiskSpace
۰۵:۴۰:۲۸.۸۷۶:  Detecting Connection
۰۵:۴۰:۲۸.۸۷۹:  Found a device with mode: DFU 
۰۵:۴۰:۲۸.۸۷۹:  Detecting ProductType
۰۵:۴۰:۲۸.۸۷۹:  matching device type through original information
۰۵:۴۰:۲۸.۸۸۱:  Checking iDevice firmware E:\iPhone10,3,iPhone10,6_14.6_18F71_Restore.ipsw
۰۵:۴۰:۲۸.۸۸۱:  Extract BuildManifest.plist
۰۵:۴۰:۲۸.۸۸۱:  Extracting Restore.plist from IPSW
۰۵:۴۰:۲۸.۹۰۲:  Firmware version: 14.6
۰۵:۴۰:۲۸.۹۰۲:  Product Build: 18F71 Major: 18
۰۵:۴۰:۲۸.۹۰۴:  Is Image4 supported: TRUE
۰۵:۴۰:۲۸.۹۰۵:  Extracting Firmware
۰۵:۴۲:۴۵.۰۶۷:  Reading ECID
۰۵:۴۲:۴۵.۰۶۷:  Found ECID 3989307495956526
۰۵:۴۲:۴۵.۰۶۷:  Reading NONCE
۰۵:۴۲:۴۵.۰۶۷:  Requesting SHSH
۰۵:۴۲:۴۵.۰۶۷:  WARNING: Unable to find BbSkeyId node
۰۵:۴۲:۴۶.۶۵۵:  Enter Recovery Mode
۰۵:۴۳:۰۵.۰۰۴:  Sending AppleLogo
۰۵:۴۳:۰۵.۰۵۲:  Sending ISP
۰۵:۴۳:۰۶.۰۳۵:  Sending RestoreTrustCache
۰۵:۴۳:۰۶.۰۵۱:  Sending Ramdisk
۰۵:۴۳:۱۴.۹۸۳:  Sending DeviceTree
۰۵:۴۳:۱۵.۰۱۰:  Sending RestoreSep
۰۵:۴۳:۱۵.۰۸۶:  Sending KernelCache
۰۵:۴۳:۱۷.۴۰۱:  Restoring
۰۵:۴۳:۱۷.۴۰۱:  About to restore device... 
۰۵:۴۳:۲۵.۴۲۵:  WARNNING: restore_is_current_device: no SerialNumber given in client data
۰۵:۴۳:۲۵.۷۴۴:  Sending RootTicket
۰۵:۴۳:۲۵.۷۶۳:  Waiting for NAND
۰۵:۴۳:۲۵.۷۷۲:  Updating NAND Firmware
۰۵:۴۳:۲۷.۰۵۷:  ERROR: Unable to receive message from FDR 0D09CFD0 (-4). 0/2 bytes

[image]
[image]
[image]
[flashlog_۲۱۰۵۲۱_۰۵۴۴۵۷.txt](https://github.com/libimobiledevice/idevicerestore/files/6525318/flashlog_._.txt)