libimobiledevice/idevicerestore #220: Build fail

mrgreenlinux, 2018-05-05:
idevicerestore-idevicerestore.o: In function `idevicerestore_start':
/home/mrgreen/idevicerestore/src/idevicerestore.c:632: warning: the use of `tempnam' is dangerous, better use `mkstemp'
/usr/bin/ld: idevicerestore-thread.o: undefined reference to symbol 'pthread_once@@GLIBC_2.2.5'
/usr/lib/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:447: idevicerestore] Error 1
make[2]: Leaving directory '/home/mrgreen/idevicerestore/src'
make[1]: *** [Makefile:424: all-recursive] Error 1
make[1]: Leaving directory '/home/mrgreen/idevicerestore'
make: *** [Makefile:356: all] Error 2

Running under Arch Linux, do any of the big distro carry devicerestore as a package?

maenpaa24, 2018-08-12:
I have exactly the same problem. Did you manage to solve it?

maenpaa24, 2018-08-12:
#211

chenxiaolong, 2018-08-12:
The Arch Linux package has a patch to fix this: https://aur.archlinux.org/cgit/aur.git/tree/0001-configure.ac-check-for-pthreads.patch?h=idevicerestore-git

maenpaa24, 2018-08-13:
Oh great thanks!

mounaiban, 2019-01-25:
I had this problem too, on Fedora 29 Workstation. Applying the patch linked by @chenxiaolong above (https://github.com/libimobiledevice/idevicerestore/issues/220#issuecomment-412364421) or doing the one-line edit manually solved this problem for me.

The patch basically adds one line to `configure.ac` in the root of the source tree:

`AC_SEARCH_LIBS([pthread_create], [pthread])`

Under the section `# Checks for Libraries`.

mounaiban, 2023-02-12:
I have successfully built on Fedora 37 as of 7b89019, the latest commit at time of writing. No changes were necessary. I have written down the [process I followed](https://gist.github.com/mounaiban/ed90ddaa4660dde0d39b8274649033cc) in my gists. Maybe the edits to `configure.ac` in 0b269e6, 7377e50 or 6401650 helped fix this issue.

Is anyone still getting this problem, or can we finally close this issue?