libimobiledevice/idevicerestore #226: idevicerestore fails to flash iOS 12 Beta 5 or iOS Beta 6 to iPhone 8, iPhone 8 Plus and iPhone X

taanders, 2018-08-15:
I tried to flash iOS 12 Beta versions 5 and 6 to latest iPhone devices. For some reason idevicerestore cannot found device when it has changed it's mode ready for flashing state and idevicerestore process exits. Flashing iOS 12 Beta version 5 and 6 to iPhone 7 and earlier works ok.
```
PROGRESS:0
Reading data from version.xml
NOTE: using cached version data
PROGRESS:10
Found device in Normal mode
PROGRESS:20
Identified device as d201ap, iPhone10,4
PROGRESS:59
Extracting BuildManifest from IPSW
PROGRESS:80
Product Version: 12.0
Product Build: 16A5345f Major: 16
Device supports Image4: true
Variant: Developer Erase Install (IPSW)
This restore will erase your device data.
PROGRESS:0
Using cached filesystem from 'iPhone_4.7_P3_12.0_16A5345f_Restore/048-22206-073.dmg'
Extracting filesystem from IPSW completed
PROGRESS:20
Getting device's ECID for TSS request
Found ECID 2578999150133294
Getting ApNonce in normal mode... d0 0b 5d c1 32 bf d5 01 07 80 a7 96 8c b2 0d 94 04 0c 40 0b 8c 01 4c 69 64 4b 98 f4 a5 f1 72 c8
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... f6 c2 2c d9 e6 38 25 a6 a8 9f fc 64 c4 48 08 97 46 97 6c 79
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
WARNING: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry AOP
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry AppleLogo
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry AudioCodecFirmware
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging0
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging1
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryFull
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow0
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow1
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryPlugin
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry DeviceTree
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry KernelCache
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry LLB
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry Liquid
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry Multitouch
DEBUG: Adding action EPRO to TSS entry
[... 1280 lines cut ...]
Attempt 160 to connect to restore mode device...
Attempt 161 to connect to restore mode device...
Attempt 162 to connect to restore mode device...
Attempt 163 to connect to restore mode device...
Attempt 164 to connect to restore mode device...
Attempt 165 to connect to restore mode device...
Attempt 166 to connect to restore mode device...
Attempt 167 to connect to restore mode device...
Attempt 168 to connect to restore mode device...
Attempt 169 to connect to restore mode device...
Attempt 170 to connect to restore mode device...
Attempt 171 to connect to restore mode device...
Attempt 172 to connect to restore mode device...
Attempt 173 to connect to restore mode device...
Attempt 174 to connect to restore mode device...
Attempt 175 to connect to restore mode device...
Attempt 176 to connect to restore mode device...
Attempt 177 to connect to restore mode device...
Attempt 178 to connect to restore mode device...
Attempt 179 to connect to restore mode device...
Attempt 180 to connect to restore mode device...
ERROR: Unable to connect to device in restore mode
ERROR: Unable to open device in restore mode
ERROR: Unable to restore device
```

Blefish, 2018-08-21:
Noticed same on iPhone 8 Plus.
Worked fine on iPhone 6S Plus.

Blefish, 2018-08-22:
It shows Apple logo but never goes to showing the empty progress bar.

I experimented and 3uTools in windows is able to flash it. Is it possible to flash custom ipsw (the beta ipsw) with iTunes?

Johnny314, 2018-08-24:
idevice_event_subscribe(restore_device_event_cb, client); 
i have checked event successfully subscribe but event does not hit as unable to connect device in restore mode

Blefish, 2018-08-24:
I am not so sure, but it seems like 'bootx' should kick it into restore mode. But it's not going into that mode.

Some differences I found more, but they don't affect device behavior (iTunes 12.7.4.80):
* setenvnp instead of setenv for boot args
* getenv ramdisk-size is called before every item is sent to device
* no getenv ramdisk-delay
* no control for 0x21 USB control.

3uTools seems to set 0x21 and I found out they use libimobiledevice idevicerestore as a library, however it flashes successfully for some reason.

From the logs, when bootx is sent, unbind event is received in kernel, but no USB disconnect, which should happen when device boots to restore mode.

Blefish, 2018-08-28:
I managed to successfully flash iPhone 8 with iOS 12 Beta 11 iPhone_4.7_P3_12.0_16A5365b_Restore.ipsw
I will submit a pull request with more details.

taanders, 2018-09-14:
iPhone 8 and 8 Plus won't boot after flashing iOS 12 Beta 11, Beta 12 or Grand Master images. Devices stay at some kind of rebooting loop. @Blefish With your fixes other iOS devices are flashing ok but those iPhone 8 and 8 Plus are failing.

I think this is maybe still some kind of error on sending img4 data. When I commented out your fix for Apple logo then Apple logo was showing on device booting screen but after that devices were rebooting continuously.

I think also that I was able to flash those devices with iOS 12 Beta 5 but it has gone some way broken again.

I cannot see any errors on logs related this issue.

```
PROGRESS:0
NOTE: using cached version data
Reading data from version.xml
PROGRESS:10
Found device in Normal mode
PROGRESS:20
Identified device as d211ap, iPhone10,5
PROGRESS:59
Extracting BuildManifest from IPSW
PROGRESS:80
Product Version: 12.0
Product Build: 16A366 Major: 16
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
PROGRESS:0
Using cached filesystem from 'iPhone_5.5_P3_12.0_16A366_Restore/048-08759-231.dmg'
Extracting filesystem from IPSW completed
PROGRESS:20
Getting device's ECID for TSS request
Found ECID 4853077237710894
Getting ApNonce in normal mode... 5a f4 1c 6b 5b de ea cc c3 bd dd bc 2f c0 53 07 3b ef cb 32 f9 a6 1d c8 bc 43 ad ec 00 8c 0a 53 
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... c9 67 20 36 94 b3 26 a2 1e dc 4b b8 f4 90 04 47 89 8c 08 6c 
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
WARNING: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry AOP
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry AppleLogo
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry AudioCodecFirmware
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging0
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging1
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryFull
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow0
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow1
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryPlugin
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry DeviceTree
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry KernelCache
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry LLB
[... 2245 lines cut ...]
	hZxQQjBAFgRNQU5QMTj/h5uVxmkLMAkWBHNlY2kCAXP/h5uV0mQMMAoWBHNlaWSgAgUA
	/4ebldxjDDAKFgRzZW5joAIFAP+E+omUUB0wGxYET0JKUDET/4ebleBoDDAKFgRzZXBo
	oAIFADAKBggqhkjOPQQDAgNIADBFAiEAkUy+2pXc47kUeHPommsfFUExhcQFmTQvnaYa
	HAASEM4CIDm3mMtGbKktiXicqQs0mL9AZThK0d8C2MRw45NIqB7B
	</data>
</dict>
</plist>
Received SE ticket
Sending FirmwareResponse data now...
Done sending FirmwareUpdater data
Fixing up /var (17)
Creating system key bag (50)
Modifying persistent boot-args (25)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Got status message
Status: Restore Finished
Cleaning up...
DONE
PROGRESS:100
FDR 03ED8128 timeout waiting for command
FDR 03ED8128 terminating...
```

Blefish, 2018-09-14:
I was able to flash it with my fix on iPhone 8 Plus - iPhone_5.5_P3_12.0_16A366_Restore.ipsw

When I flashed it one a completely full (storage filled) device it failed with ASR error, but when I performed erase upgrade, it worked fine.
I am not sure what might cause your problem though.

jduranc, 2018-09-14:
Same here,

Flashing iPhone 8 or 8P full/erase mode finish with success result, but device stay in bootloop, white-black screen its blinking and no initial post-restore loading appears in device.

Blefish, 2018-09-14:
Is this on Windows or Linux? For the record, I am using it on Linux

jduranc, 2018-09-14:
> Is this on Windows or Linux? For the record, I am using it on Linux

I'm testing on windows.

As note, I also try using 3utools also and its working, after use 3utools on that devices, they can be restored successfully using idevicerestore.

taanders, 2018-09-17:
It might be something related it. There comes same kind of lines from iPhone X flashing.

taanders, 2018-09-17:
I was able to erase flash iOS 12 GM to iPhone 8 Plus when I had there pre-installed iOS 12 from beta.apple.com.

Upgrading with erase from iOS 11.4.1 to iOS 12 is still failing.

@Blefish Are you able to upgrade iPhone 8 or 8 Plus iOS version from 11.4.1 to 12 on Linux?

pekn, 2018-09-17:
First, I used idevicerestore to restore iOS 12 to iPhone 8 and the device went to boot loop. Then I used iTunes to restore iOS 12 to fix boot loop -> that of course succeeded. Now when I try to use idevicerestore, it succeeds as well.. so this seems to be rather nasty issue.

Blefish, 2018-09-17:
Yes, it worked fine for me, which is strange. I am using the same libimobiledevice libraries with some small changes (and the 2 commits I submitted) on Ubuntu 18.04.

After your comment I retried it like this (iPhone 8 Plus, d211ap, iPhone10,5):

Currently on iOS 12 GM iPhone_5.5_P3_12.0_16A366_Restore.ipsw
Restored with `idevicerestore -e to iPhone_5.5_P3_11.0_11.4.1_15G77_Restore.ipsw`
Restored with `idevicerestore -e to iPhone_5.5_P3_12.0_16A366_Restore.ipsw`
Device did post-restore progress-bar and booted like normal

Might be able to dig out iPhone 8 and test with it, it has never been upgraded to 12.0 yet.
I believe the issue is that some partition is not updated, which has been the case before. But I cannot understand why this is not happening in my case.

Blefish, 2018-09-18:
I dug out 2 iPhone 8's and they flashed fine to iPhone_4.7_P3_12.0_16A366_Restore.ipsw with idevicerestore. I used erase restore.

Both of them were never upgraded to 12.0 betas and one of them has never been on Beta software.

taanders, 2018-09-18:
I think also that there might be something wrong with iOS 12 Beta images. I updated iOS 12 Final version from OTA update to failing iPhone 8 and after that it is possible to flash device back to iOS 11.4.1 and then flashing to iOS 12 version works ok.

I got same results from iPhone 8 Plus that failed earlier and after installing iOS 12 OTA update it started work with idevicerestore.

I didn't find any reason for those failures when first flashing iOS 12 Beta version and then trying to flash iOS 12 Final version.

jduranc, 2018-09-19:
I'm still having issues restoring iP8 and iP8P devices, using the iOS 12 final version using windows, here my config.
* Windows 7
* iTunes: 12.9.0.167
* Apple mobile device support : 12.0.0.1039

@taanders I can see in your logs that you are using miniGW, could you tell me if you need another changes in order to get working the process?, I also see another post where you apply one fix for libirecovery, I try it also with no luck.

[output.txt](https://github.com/libimobiledevice/idevicerestore/files/2399004/output.txt)

I appreciate any help.

Orbif, 2018-09-20:
Hello,

I have the same problem than @jduranc with iPhone 8. The restore end without errors, but then the iPhone 8 doesn't restart.

No problem restoring iPhone 7.

I tried with the latest code including @Blefish patch.

Any ideas ?

nicolassan, 2018-09-20:
Hi,
same problem with iPhone 8 and IOS 12 release.
DFU don't start, i see the apple logo screen but the progress bar doesn't appear.

Orbif, 2018-09-21:
I tried with another iPhone 8.

I only have the problem (bootloop) when I restore iOS 11 device.
No problem restoring a device that was already upgraded to iOS 12 with itunes.

taanders, 2018-09-24:
@nicolassan and @Orbif have you tried to restore iOS 11 devices without using any iOS 12 Beta version on those devices earlier?

I had same kind bootloop issues with iOS 11.4.1 devices which had some iOS 12 Beta version installed earlier. @Blefish wrote that he was able to do iOS 12 restore process for devices which were not used any iOS 12 Betas.

Orbif, 2018-09-24:
@taanders I didn't tried any beta version of iOS 12. I have the problem when I restore iOS 11.4.1 to iOS 12 final.

Blefish, 2018-09-25:
I was finally able to replicate this problem on a 256GB iPhone 8. Apple xx 2 restored it properly so there is something imobiledevice is failing to do at the moment. Not sure why the other iPhone 8/8+ models I had did not fail this way.

jduranc, 2018-09-25:
I fixed it by adding next flags in options send to device. In function int restore_device

char* guid = generate_guid();
if (guid) {
	plist_dict_set_item(opts, "UUID", plist_new_string(guid));
	free(guid);
}

plist_dict_set_item(opts, "UpdateBaseband", plist_new_bool(1));

//Added for iOS 12
plist_t smt = plist_new_dict();
plist_dict_set_item(smt, "BBUpdateStatusMsg", plist_new_bool(0));

plist_dict_set_item(smt, "CheckpointMsg", plist_new_bool(1));
plist_dict_set_item(smt, "DataRequestMsg", plist_new_bool(0));
plist_dict_set_item(smt, "MsgType", plist_new_bool(0));
plist_dict_set_item(smt, "PreviousRestoreLogMsg", plist_new_bool(1));
plist_dict_set_item(smt, "ProgressMsg", plist_new_bool(0));

plist_dict_set_item(smt, "ProvisioningAck", plist_new_bool(0));
plist_dict_set_item(smt, "ProvisioningInfo", plist_new_bool(0));
plist_dict_set_item(smt, "ProvisioningStatusMsg", plist_new_bool(0));
plist_dict_set_item(smt, "ReceivedFinalStatusMsg", plist_new_bool(0));
plist_dict_set_item(smt, "StatusMsg", plist_new_bool(0));
plist_dict_set_item(opts, "SupportedMessageTypes", smt);

plist_t sep = plist_access_path(build_identity, 3, "Manifest", "SEP", "Info");
if (sep)
{
	node = plist_dict_get_item(sep, "RequiredCapacity");

	if (node && plist_get_node_type(node) == PLIST_STRING) {
		char* sval = NULL;
		plist_get_string_val(node, &sval);
		plist_dict_set_item(opts, "TZ0RequiredCapacity", plist_copy(node));
		free(sval);
		sval = NULL;
	}
}

plist_dict_set_item(opts, "WaitForDeviceConnectionToFinishStateMachine", plist_new_bool(0));
plist_dict_set_item(opts, "iTunesVersion", plist_new_string("iTunes 12.9.0.167"));

plist_dict_set_item(opts, "FormatForAPFS", plist_new_bool(1));
plist_dict_set_item(opts, "FormatForLwVM", plist_new_bool(0));
plist_dict_set_item(opts, "InstallDiags", plist_new_bool(0));
plist_dict_set_item(opts, "SkipPreflightPersonalization", plist_new_bool(0));

Tested on all models and its working again. In my tests works just by adding flag "TZ0RequiredCapacity", but I leave the others flags for future compatibility. @Blefish thanks for your help & support.

nicolassan, 2018-09-25:
Yes !!

Many thanks !!! I 'll try theses modifications quickly

Orbif, 2018-09-25:
Tested with iPhone 8 and iPhone 8+, It worked ! 
Thank you @jduranc and @Blefish  !

nicolassan, 2018-09-26:
Good for me too ! Many thanks !

Blefish, 2018-10-02:
Yes, I experienced the same issue @Orbif

Orbif, 2018-10-03:
Looks like this problem is related to the AppleLogo component. 
If I remove @Blefish patch, the apple logo show up normally when iPhone 6s is started.
But without this patch, restore fail in iPhone8/8P.

Blefish, 2018-10-03:
I submitted two patches, maybe https://github.com/libimobiledevice/idevicerestore/commit/5b0eafc634c294f9c85c7aa756d8f2e2970484c9 is causing this? Or is this needed for iPhone 8?

Orbif, 2018-10-03:
Yes, without 5b0eafc patch (if I comment the two lines), iPhone 6s start normally. But restore of iPhone 8 is broken again. I think this is needed for iPhone 8/8P anyway.

taanders, 2018-10-03:
@Blefish and @Orbif I tried quickly to replace text AppleLogo from commit https://github.com/libimobiledevice/idevicerestore/commit/5b0eafc634c294f9c85c7aa756d8f2e2970484c9 with text RestoreLogo and it seems to work with iPhone 6 and iPhone 8.

Blefish, 2018-10-03:
For me I was able to restore iPhone 8 without the "AppleLogo" patch just fine, for some reason my devices are different :P. Is RestoreLogo part of the components list?

K3ph4s, 2018-10-15:
Hi community,
i am trying  to restore my iphone 8 plus, but the massage is:
ERROR: Unable to discover device type
i suppose i dont have tha latest idevicerestore.
can anyone tell me where to find it?

Thanks

K3ph4s, 2018-10-15:
> Tested with iPhone 8 and iPhone 8+, It worked !
> Thank you @jduranc and @Blefish !

hi guys, could you help me with this? i have Iphone 8p. my error is "unable to discover device type" is the version of libimobiledevice the problem??
hope you can help me

K3ph4s, 2018-10-24:
All,
can someone help me?

nikias, 2019-01-26:
This issue is resolved with latest code. @K3ph4s make sure you are using latest libirecovery.

closed 2019-01-26 by nikias

K3ph4s, 2019-01-27:
@nikias where i can find the latest libirecovery code? i cant fint it.
Regards

nikias, 2019-01-27:
It's part of libimobiledevice project, not really hard to find...
https://github.com/libimobiledevice/libirecovery

K3ph4s, 2019-01-27:
@nikias can we speak in private¿?